espressif/arduino-esp32 #6673: GET following  secureClient.setInsecure(); results in SSL - An invalid SSL record was received

SheetLightning, 2022-04-30:
### Board

ESP32 WROOM

### Device Description

ESP32 Lolin

### Hardware Configuration

No

### Version

v1.0.6

### IDE Name

Arduino IDE version 1.8.19

### Operating System

Linux Mint

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

New issue to reference issue 4992 which is still not fixed and has been closed.

https://github.com/espressif/arduino-esp32/issues/4992

I am still having the problem with the latest version of the ESP32 library for the Arduino IDE.

### Sketch

```cpp
None provided at this stage
```

### Debug Message

```plain
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():287]: (-29184) SSL - An invalid SSL record was received
```

### Other Steps to Reproduce

Set secureClient.setInsecure(); then perform a get request.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-04-30:
You must provide a certificate authority cert (CAcert) for the site if you want to connect to a protected site.  You can set setInsecure before the connection, which tells WiFiClientSecure to accept whatever certificate is presented by the server.  This is not actually secure, since there is no authority backing up the server's claim of authenticity.

podaen, 2022-05-01:
I think your site you want to connect to doesn't support insecure requests. Some sites doesn't simply approve that kind of connection.

SuGlider, 2022-05-02:
Not being authenticated is the reason for receiving the message `SSL - An invalid SSL record was receive`
As said by @lbernstone and @podaen - it most likely is an error of certificates.

Not an issue of Arduino. Please try https://github.com/espressif/arduino-esp32/issues/4992#issuecomment-811088088 in order to bypass it, but be aware of the warning in the commentary below:

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/WiFiClientSecure.h#L68

podaen, 2022-05-02:
Put your verbose log on and you would be more informed about what happens.  Did you set it to insecure yet?

podaen, 2022-05-02:
> I appreciate that using setInsecure() means that the connection is not SSL encrypted and therefore not secure.

You will keep running the ssl library and get probably an x.509 certificate. That's normal, but you want to connect to google. That is something else. I tried it too a while a go but it  didn't work for me ether. I don't think it possible with the current tcp on the esp, because google is base64 encoded and so on and so on, if I am remembering it good.

me-no-dev, 2022-05-09:
> I appreciate that using setInsecure() means that the connection is not SSL encrypted and therefore not secure.

This is not what it means :) Your connection IS encrypted, BUT the certificate is not validated if authentic. This allows you to connect to places where you do not have certificates for (or have not added proper ROOT certificates), but it also allows you to connect to malicious site, pretending to be someone else.

SheetLightning, 2022-05-09:
Well now that is interesting. So its basically ignoring the certificate chain and validation. Thanks for explaining that. However, that must put the ball squarely in Esspressif's court. If my sketch works flawlessly on an ESP8266, then why shouldn't it work using the equivalent web client libraries (with identical syntax) on the ESP32?

This is the project I am following:
https://github.com/SensorsIot/Reminder-with-Google-Calender

me-no-dev, 2022-05-09:
@SheetLightning previous behavior is very insecure by default. Now it is not. ESP32 is used in many projects that do require proper security. You can add the following snippet before `client.connect();` and it will work on both platforms
```cpp
#ifdef ARDUINO_ARCH_ESP32
client.setInsecure();
#endif
client.connect();
```

lbernstone, 2022-05-12:
The links you are using for the Boards Manager URL are incorrect.  Please verify they match the documentation: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

VojtechBartoska, 2022-06-14:
any updates on this?

VojtechBartoska, 2022-07-27:
Please, are you able to provide more info?

VojtechBartoska, 2022-08-23:
I'm closing this issue as expired due to no answer. If it's needed, please reopen it. Thanks.

closed 2022-08-23 by VojtechBartoska