espressif/arduino-esp32 #6680: Using Default arduino Libs(BLE, WiFiClientSecure) giving error

CodeOn-ArK, 2022-05-03:
### Board

ESP32 Dev Module

### Device Description

ESP-WROOM 32 DevKitC

### Hardware Configuration

NULL

### Version

v2.0.0

### IDE Name

IDF 

### Operating System

Linux

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I am trying to use arduino libraries such as WIFIClientSecure.h in the IDF project. I am able to link IDF components with ease, but am unable to use arduino libraries such as BLE and WIFiClientSecure. It gives undefined reference error. Is there something else that needs to be done to use default arduino libs?

### Sketch

```cpp
The include header :

#include <WiFiClientSecure.h>
```
The variables :
```
WiFiClientSecure client;
MQTTPubSubClient mqtt;
String old,prev_state;
```

Just the inclusion of these give error as mentioned below
```

### Debug Message

```plain
[ 41%] Built target __idf_esp_phy
Consolidate compiler generated dependencies of target __idf_espcoredump
[ 41%] Built target __idf_espcoredump
Consolidate compiler generated dependencies of target __idf_esp_gdbstub
[ 42%] Built target __idf_esp_gdbstub
Consolidate compiler generated dependencies of target __idf_pthread
[ 43%] Built target __idf_pthread
Consolidate compiler generated dependencies of target __idf_nvs_flash
[ 44%] Built target __idf_nvs_flash
Consolidate compiler generated dependencies of target __idf_spi_flash
[... 92 lines cut ...]
/home/ubuntu/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj): in function `WiFiClientSecure::read(unsigned char*, unsigned int)':
/home/ubuntu/hello/components/arduino/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:223: undefined reference to `get_ssl_receive(sslclient_context*, unsigned char*, int)'
/home/ubuntu/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj): in function `WiFiClientSecure::available()':
/home/ubuntu/hello/components/arduino/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:237: undefined reference to `data_to_read(sslclient_context*)'
/home/ubuntu/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj): in function `WiFiClientSecure::WiFiClientSecure()':
/home/ubuntu/hello/components/arduino/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:36: undefined reference to `ssl_init(sslclient_context*)'
/home/ubuntu/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj): in function `WiFiClientSecure::connect(char const*, unsigned short, char const*, char const*, char const*)':
/home/ubuntu/hello/components/arduino/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:134: undefined reference to `start_ssl_client(sslclient_context*, char const*, unsigned int, int, char const*, bool, char const*, char const*, char const*, char const*, bool, char const**)'
/home/ubuntu/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj): in function `WiFiClientSecure::connect(char const*, unsigned short, char const*, char const*)':
/home/ubuntu/hello/components/arduino/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:154: undefined reference to `start_ssl_client(sslclient_context*, char const*, unsigned int, int, char const*, bool, char const*, char const*, char const*, char const*, bool, char const**)'
collect2: error: ld returned 1 exit status
CMakeFiles/hello.elf.dir/build.make:503: recipe for target 'hello.elf' failed
make[2]: *** [hello.elf] Error 1
CMakeFiles/Makefile2:2131: recipe for target 'CMakeFiles/hello.elf.dir/all' failed
make[1]: *** [CMakeFiles/hello.elf.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2
make failed with exit code 2

```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-05-03:
@CodeOn-ArK - Is this an IDF project? 

Are you using Arduino as a Component?

SuGlider, 2022-05-03:
@CodeOn-ArK - Please take a look at
https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

CodeOn-ArK, 2022-05-03:
> @CodeOn-ArK - Is this an IDF project? 
> 
> Are you using Arduino as a Component?

Yes and yes

CodeOn-ArK, 2022-05-03:
> @CodeOn-ArK - Please take a look at
> https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

What does it do actually?

CodeOn-ArK, 2022-05-04:
why did this issue suddenly closed?

SuGlider, 2022-05-04:
> why did this issue suddenly closed?

This issue has not closed. It is still open. I think that you are seeing the "closed label" from issue #1142

SuGlider, 2022-05-04:
> > @CodeOn-ArK - Please take a look at
> > https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html
> 
> What does it do actually? The library builder

Lib Builder is a set of scripts that will actually build all Arduino Libraries used folder:
https://github.com/espressif/arduino-esp32/tree/master/tools/sdk

Arduino uses IDF precompiled libraries. 
Lib builder scripts will git clone all necessary resources and externals repositories.
It will compile Arduino as a Component into the tools/sdk, compile all together and create this repository.
It is used when IDF has changed and it is necessary to update all IDF libraries used by Arduino (precompiled IDF libs).

VojtechBartoska, 2022-05-18:
@CodeOn-ArK Do you still need help?

CodeOn-ArK, 2022-05-18:
No, got the thing resolved.

closed 2022-05-18 by VojtechBartoska

CodeOn-ArK, 2023-04-28:
I really don't remember rn, but using the master works ig, and it's basic installation and setup mistake

AndruKrug, 2023-04-28:
Solution:
It is necessary to change the parameters in the menuconfig according to the warning:
.../.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:24:4: warning: #warning "Please call `idf.py menuconfig` then go to Component config -> mbedTLS 
-> TLS Key Exchange Methods -> Enable pre-shared-key ciphersuites and then check `Enable PSK based cyphersuite modes`. Save and Quit." [-Wcpp]
 #  warning "Please call `idf.py menuconfig` then go to Component config -> mbedTLS -> TLS Key Exchange Methods -> Enable pre-shared-key ciphersuites and then check `Enable PSK based cyphersuite modes`. 
Save and Quit."