espressif/arduino-esp32 #6683: Testing Stream Break

podaen, 2022-05-04:
### Board

ESP32 devkit v4

### Device Description

No

### Hardware Configuration

No

### Version

latest development Release Candidate (RC-X)

### IDE Name

19

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I have a lead on the stream break... The size of the buffer is 1436, and I read 512 in one read. So when I do WiFiClientRxBuffer read... it have to run in the equation below two times before it proceeded to the last line, but when the stream breaks it just do it once.  The occasion bellow does not enter. You can find it in WiFiClient.cpp -> WiFiClientRxBuffer -> read

`if(len <= a || ((len - a) <= (_size - _fill) && fillBuffer() >= (len - a))){`     (if this is true I get len: 512)

Here is my provisional log.

> WiFiClientRxBuffer ended
> len: 512
> len: 512
> WiFiClientRxBuffer ended

Here breaks the stream, equation 1time entered.

> len: 512
> WiFiClientRxBuffer ended

it returns to the normal way, but doesn't work anymore. I believe there is an memory issue now, because... memcpy(dst, _buffer + _pos, len); has not been done.

> len: 512
> len: 512
> WiFiClientRxBuffer ended

This is what I know for now... I am going to check all parameters now.

### Sketch

```cpp
No
```

### Debug Message

```plain
No
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

podaen, 2022-05-04:
No problem my friend. Never heard of milestone... I am new here because I had a problem with my stream. I m happy to help. It's not an easy to reproduce and it has been here for years. I was thinking of memory issue or a timout. The timout I checked, it's not that anymore. So it had to be a memory issue.

WiFiClientRxBuffer ended

First of all I m not sure anymore that my serial monitor did stop on the right moment because he couldn't keep up with all the plots. But again I have one time entering in the equation.

> res read wificlient: 512
> 512
> _pos: 352
> len: 512
> a: 1084
> _size: 1436
> _fill: 1436
> fillBuffer(): 0
> len: 512
> 
> res read wificlient: 512
> 512
> _pos: 864
> len: 512
> a: 572
> _size: 1436
> _fill: 1436
> fillBuffer(): 0
> len: 512
> 
> res read wificlient: 512
> 512
> _pos: 1376
> len: 512
> a: 60
> _size: 1436
> _fill: 1436
> fillBuffer(): 0
> WiFiClientRxBuffer ended
> 
> 
> res read wificlient: 512
> 512
> _pos: 452
> len: 512
> a: 984
> _size: 1436
> _fill: 1436
> fillBuffer(): 0
> len: 512
> 
> res read wificlient: 512
> 512
> _pos: 964
> len: 512
> a: 472
> _size: 1436
> _fill: 1436
> fillBuffer(): 0
> WiFiClientRxBuffer ended
> 
> 
> res read wificlient: 512
> 512
> _pos: 40
> len: 512
> a: 1396
> _size: 1436
> _fill: 1436
> fillBuffer(): 0
> len: 512
> 
> res read wificlient: 512
> 512
[... 269 lines cut ...]
        size_t left = len; //512
        size_t toRead = a; //72
        uint8_t * buf = dst;
        memcpy(buf, _buffer + _pos, toRead);
        _pos += toRead;
        left -= toRead; //left = 512 -74 = 438
        buf += toRead;
        while(left){
            if(!fillBuffer()){
                return len - left;
            }
            a = _fill - _pos;
            toRead = (a > left)?left:a;
            memcpy(buf, _buffer + _pos, toRead);
            _pos += toRead;
            left -= toRead;
            buf += toRead;
        }
        Serial.println("WiFiClientRxBuffer ended");
        Serial.println();
        Serial.println();
        return len;
```

The  if(!fillBuffer()){   condition has never been done. So not sure why it returns 200...

SuGlider, 2022-05-04:
@podaen - It is hard to follow your issue.

Usually the issues are composed of one clear statement, an example sketch that allows us to reproduce and then fix the issue and an output log of the error plus some comment about what is expected as correct output.

Please follow this process in order to help us to help you.
I have the impression that you use this space as your draft notebook about your thinking process and discoveries.
Doing it, doesn't help us much.

podaen, 2022-05-04:
Oke. I am searching for it almost a year now. And I'm close to read every byte that has been received. How far can it still be? 

Why is it bad to understand what I am doing?

Reading byte is a hard thing to do! It is like reading a book each character at the time and make a calculation with it. If I don't say what I am doing you will never understand.

mrengineer7777, 2022-05-04:
@podaen Suggestion: create a sample project demonstrating the issue and host it as a public repository on Github, then link here.  Here's an example of mine: https://github.com/mrengineer7777/MA_Test

podaen, 2022-05-04:
@mrengineer7777 Maybe you could explane me

 why this function

```
            if(!fillBuffer()){
                return len - left;
            }
```
 is not like this

```
            if(fillBuffer()!=true){
                Serial.println("return len - left");
                return len - left;
            }
```

mrengineer7777, 2022-05-04:
> @mrengineer7777 That is an example my friend where you could be proud up! I remembered when I started with ble, that this was the first thing I did.

Thank you

> But now I am doing in practice with a random stream. I don't say the problem is in here, but I would like to know if I received the data well... I hope you understand.

Understood.  Unfortunately we can't do anything to investigate unless the problem can be replicated.  Can you provide a sample project that demonstrates the issue?

podaen, 2022-05-05:
Hi, Can't find this...

recv(_fd, _buffer + _fill, _size - _fill, MSG_DONTWAIT)

I want to understand how it works. Can somebody show me?

lbernstone, 2022-05-05:
This site is not for training, or for issues with <b>your</b> code.  Please keep in mind that the time the devs spend reading and answering your questions is time they are not working on improving the code for everybody else.

podaen, 2022-05-05:
I don't know it yet...

mrengineer7777, 2022-05-05:
@podaen recv() is in sockets.h, which is in the esp-idf base project.
`static inline ssize_t recv(int s,void *mem,size_t len,int flags)`

It has something to do with reading from the socket.  That's all I know.

SuGlider, 2022-05-05:
ESP32 UDP and TCP/IP layers are based on an open project called LWIP.

LWIP implements all necessary Networking Stack, including Sockets, etc.
Hardware will fill the buffers with data from the radios and these data will be processed by LWIP layer.

TCP/IP layer will receive data and process it, eventually passing the payload to HTTP layer and so on up to the Application layer.

It follows OSI/ISO model.

podaen, 2022-05-05:
Thanks! I could have know.

> RETURN VALUE 
>        These calls return the number of bytes received, or -1 if an
>        error occurred.  In the event of an error, [errno](https://man7.org/linux/man-pages/man3/errno.3.html) is set to
>        indicate the error.
> 
>        When a stream socket peer has performed an orderly shutdown, the
>        return value will be 0 (the traditional "end-of-file" return).
> 
>        Datagram sockets in various domains (e.g., the UNIX and Internet
>        domains) permit zero-length datagrams.  When such a datagram is
>        received, the return value is 0.
> 
>        The value 0 may also be returned if the requested number of bytes
>        to receive from a stream socket was 0.

podaen, 2022-05-07:
I want to add to the LWIP

- TEMP_FAILURE_RETRY.
- update errno codes
- errno description
-...

I'm not experienced enough to do that, so I am wondering if someone can add this?

podaen, 2022-05-07:
hmmm, I see this can not be done on esp socket...

SuGlider, 2022-05-07:
As this is not an issue, I'll close it.
@podaen - Please open a new discussion about this subject instead of an issue.
You can do it at https://github.com/espressif/arduino-esp32/discussions

closed 2022-05-07 by SuGlider