espressif/arduino-esp32 #6687: [2.0.2] Big binary size increase

handmade0octopus, 2022-05-04:
### Board

All boards

### Device Description

It's not board specific.

### Hardware Configuration

Not board specific.

### Version

v2.0.2

### IDE Name

PlatformIO 4.2.0

### Operating System

Windows

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I updated PlatformIO from version 3.5.0 to 4.2.0 and flash changed by A LOT compared to what I used before.

From on 3.5.0
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.0% (used 55620 bytes from 327680 bytes)
Flash: [========  ]  77.9% (used 1532256 bytes from 1966080 bytes)
```

To on 4.2.0
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Error: The program size (1966705 bytes) is greater than maximum allowed (1966080 bytes)
RAM:   [==        ]  18.3% (used 60076 bytes from 327680 bytes)
*** [checkprogsize] Explicit exit, status 1
Flash: [==========]  100.0% (used 1966705 bytes from 1966080 bytes)
```

I cannot flash my program anymore and has to revert back to 3.5.0.

Any idea what could cause this? Any workarounds?

Libraries that I use in sketch below.

### Sketch

```cpp
// Libraries that I use:

#include <SPIFFS.h>
#include <SD.h>
#include <SPI.h>
#include <Wire.h>
#include <Update.h>
#include <BluetoothSerial.h>

#include <HTTPClient.h>
#include <WiFi.h>
#include "time.h"
#include <ESPAsyncWebServer.h>
#include <DNSServer.h>
#include <ESPmDNS.h>

#include <StreamUtils.h>
#include <ArduinoJson.h>

#include <TinyGPS++.h>

// Adafruit
#include <Adafruit_I2CDevice.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#include <Adafruit_Sensor.h>
#include <Adafruit_ADXL345_U.h>
```

### Debug Message

```plain
[checkprogsize] Explicit exit, status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-05-04:
It's weird. you are the first one reporting this. It could be probably something with your setup.

SuGlider, 2022-05-04:
Closing it here. 
@handmade0octopus - please fell free to open it again or open a new issue related to this one, if necessary.

closed 2022-05-04 by SuGlider