espressif/arduino-esp32 #6691: Linking issue

stefann63, 2022-05-04:
### Board

nodemcu-32

### Device Description

plain module

### Hardware Configuration

no

### Version

v2.0.2

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

Old project that was compiled/linked properly a year ago, after update doesn't link properly.

### Sketch

```cpp
AsyncWebSocketResponse::AsyncWebSocketResponse(const String& key, AsyncWebSocket *server){
  _server = server;
  _code = 101;
  _sendContentLength = false;

  uint8_t * hash = (uint8_t*)malloc(20);
  if(hash == NULL){
    _state = RESPONSE_FAILED;
    return;
  }
  char * buffer = (char *) malloc(33);
  if(buffer == NULL){
    free(hash);
    _state = RESPONSE_FAILED;
    return;
  }
#ifdef ESP8266
  sha1(key + WS_STR_UUID, hash);
#else
  (String&)key += WS_STR_UUID;
  SHA1_CTX ctx;
  SHA1Init(&ctx);
  SHA1Update(&ctx, (const unsigned char*)key.c_str(), key.length());
  SHA1Final(hash, &ctx);
#endif
  base64_encodestate _state;
  base64_init_encodestate(&_state);
[... 144 lines cut ...]
Compiling .pio\build\nodemcu-32s\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\nodemcu-32s\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\nodemcu-32s\libFrameworkArduino.a
Linking .pio\build\nodemcu-32s\firmware.elf
c:/users/stefan/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\nodemcu-32s\lib4d1\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x10): undefined reference to `SHA1Init'
c:/users/stefan/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\nodemcu-32s\lib4d1\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x14): undefined reference to `SHA1Update'
c:/users/stefan/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\nodemcu-32s\lib4d1\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x18): undefined reference to `SHA1Final'
c:/users/stefan/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\nodemcu-32s\lib4d1\libESP Async WebServer.a(AsyncWebSocket.cpp.o): in function `AsyncWebSocketResponse::AsyncWebSocketResponse(String const&, AsyncWebSocket*)':
O:\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:1269: undefined reference to `SHA1Init'
c:/users/stefan/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: O:\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:1270: undefined reference to `SHA1Update'
c:/users/stefan/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: O:\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:1271: undefined reference to `SHA1Final'
c:/users/stefan/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\nodemcu-32s\lib4d1\libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x4): undefined reference to `mbedtls_md5_starts'
c:/users/stefan/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\nodemcu-32s\lib4d1\libESP Async WebServer.a(WebAuthentication.cpp.o): in function `getMD5(unsigned char*, unsigned short, char*)':
O:\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nodemcu-32s\firmware.elf] Error 1
========================= [FAILED] Took 28.62 seconds =========================
"platformio -f -c eclipse run" terminated with exit code 1. Build might be incomplete.

14:34:05 Build Failed. 10 errors, 4 warnings. (took 29s.43ms)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-05-04:
Make sure ESPAsyncWebServer is updated to latest version.

stefann63, 2022-05-05:
Yeah, the problem come after upgrade, update  in PlatformIO.
I think the the line "version": "1.2.3"  in library.json was not updated.

stefann63, 2022-05-05:
Yes, after updating ESPAsyncWebServer manually.
The updated library has same version 1.2.3 as the old one.

closed 2022-05-05 by me-no-dev

stefann63, 2022-05-05:
@me-no-dev Ще почерпя!

me-no-dev, 2022-05-06:
> @me-no-dev Ще почерпя!

Свободното време ми е малко несъществуващо в момента (май от две години насам поне). Има инициатива в компанията да имаме възможност да отделяме време на подобни странични проекти. Стискам палци да стане реалност ;) Първо да стегнем още малко Ардуиното