espressif/arduino-esp32 #6692: Max services or characteristics with BLE?

paulhamsh, 2022-05-04:
### Board

M5Stack Core 2

### Device Description

ESP32
Board manager M5Stack v2.0.2

### Hardware Configuration

Nothing, vanilla M5Stack Core 2

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

Unknown

### PSRAM enabled

yes

### Upload speed

921600

### Description

There seems to be an error in creating many services and characteristics in the BLE library. 
I have trimmed down this sketch and it gives multiple errors, and when BLE is explored using LightBlue some services and characteristics are missing and some characteristics are under the wrong service.
I amended it to use the createService(BLEUUID(x), nnn) constructor hoping that would fix it, and whilst it allowed more characteristics it still creates errors.

For example, service FE59 doesn't appear.
Services 25a22330-820f-11e3-baa7-0800200c9a66, 03b80e5a-ede8-4b33-a751-6ce34ec4c700 don't appear.
Some characteristics, like 7772e5db-3868-4112-a1a9-f2669d106bf3, appear under the wrong service.

The code works with NimBLE (with the appropriate NimBLE changes ie removing the BLE2902 descriptors, changing the characteristic capability nomenclature and changing the #include)

### Sketch

```cpp
#include <M5Core2.h>

#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

static BLEServer* pServer;

void setup() {
    M5.begin();
    Serial.println("Starting BLE Server");
   
    BLEDevice::init("BLESERVER");
    pServer = BLEDevice::createServer();

#define SA  "180a" 
#define CA1 "2a29"
[... 173 lines cut ...]
[  2906][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 733ee9cc-174d-452e-b45c-0402aa75ff75, but didnt!
[  2919][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 733ee9cc-174d-452e-b45c-0402aa75ff75, but didnt!
[  2932][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 733ee9cc-174d-452e-b45c-0402aa75ff75, but didnt!
[  2946][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 362f71a0-6c96-11e3-981f-0800200c9a66, but didnt!
[  2958][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 362f71a0-6c96-11e3-981f-0800200c9a66, but didnt!
[  2971][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 362f71a0-6c96-11e3-981f-0800200c9a66, but didnt!
[  2985][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: bd066da4-f9ec-4f0d-a53c-1cd99147a641, but didnt!
[  2997][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: bd066da4-f9ec-4f0d-a53c-1cd99147a641, but didnt!
[  3010][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: bd066da4-f9ec-4f0d-a53c-1cd99147a641, but didnt!
[  3024][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 85da4f4b-a2ca-4c7c-8c7d-fcd9e2daad56, but didnt!
[  3036][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 85da4f4b-a2ca-4c7c-8c7d-fcd9e2daad56, but didnt!
[  3049][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 85da4f4b-a2ca-4c7c-8c7d-fcd9e2daad56, but didnt!
[  3064][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 8ec90003-f315-4f60-9fb8-838830daea50, but didnt!
[  3075][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 8ec90003-f315-4f60-9fb8-838830daea50, but didnt!
[  3090][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: ba7cc552-cc2c-404e-bf75-8778f023787d, but didnt!
Advertising
```

### Other Steps to Reproduce

I get the same results using Light Blue on ios and android, and also with nRF Connect.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

paulhamsh, 2022-05-07:
Update: it works with 7 services, but no more - after the seventh, any new characteristics get added to the seventh service.

paulhamsh, 2022-05-10:
Hi
I have added a dump() to each service. The dump shows them as being ok, but LightBlue, nRF connect and my own code in Pythonista on the iphone show all the final characteristics attached to the sixth service (7bdb8dc0-6c95-11e3-981f-0800200c9a66), as though the calls to the underlying library have failed.
I get the same result with Lightblue and nRF Connect on Android. (That shows 1801, 1800, 180A, 180F, 344..., 6fac..., 5cb6...,7bdb... - so, 8 services)
I can't find the source for that on github so not really able to investigate any further on my own.

```
[  2360][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[  2362][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[  2369][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=00002a29-0000-1000-8000-00805f9b34fb to service: UUID: 0000180a-0000-1000-8000-00805f9b34fb, handle: 0x0028
[  2385][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=00002a24-0000-1000-8000-00805f9b34fb to service: UUID: 0000180a-0000-1000-8000-00805f9b34fb, handle: 0x0028
[  2401][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=00002a27-0000-1000-8000-00805f9b34fb to service: UUID: 0000180a-0000-1000-8000-00805f9b34fb, handle: 0x0028
[  2418][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=00002a26-0000-1000-8000-00805f9b34fb to service: UUID: 0000180a-0000-1000-8000-00805f9b34fb, handle: 0x0028
[  2434][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=00002a25-0000-1000-8000-00805f9b34fb to service: UUID: 0000180a-0000-1000-8000-00805f9b34fb, handle: 0x0028
[  2451][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=00002a23-0000-1000-8000-00805f9b34fb to service: UUID: 0000180a-0000-1000-8000-00805f9b34fb, handle: 0x0028
[  2467][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=00002a50-0000-1000-8000-00805f9b34fb to service: UUID: 0000180a-0000-1000-8000-00805f9b34fb, handle: 0x0028
[  2485][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[  2493][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=00002a19-0000-1000-8000-00805f9b34fb to service: UUID: 0000180f-0000-1000-8000-00805f9b34fb, handle: 0x0050
[  2511][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[  2520][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=cad0c949-7dce-4a04-9d80-e767c796b392 to service: UUID: 34452f38-9e44-46ab-b171-0cc578feb928, handle: 0x0078
[  2536][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=76e7df30-e3ad-41b1-a05a-279c80fc7fb4 to service: UUID: 34452f38-9e44-46ab-b171-0cc578feb928, handle: 0x0078
[  2552][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=cdb3a16b-02e1-4bc0-843c-665cbeb378b7 to service: UUID: 34452f38-9e44-46ab-b171-0cc578feb928, handle: 0x0078
[  2569][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=53eea35f-27f1-46e4-a790-34becd28b701 to service: UUID: 34452f38-9e44-46ab-b171-0cc578feb928, handle: 0x0078
[  2586][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=54eea35f-27f1-46e4-a790-34becd28b701 to service: UUID: 34452f38-9e44-46ab-b171-0cc578feb928, handle: 0x0078
[  2603][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[  2612][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=90d9a098-9cd8-4a7a-b176-91ffe80909f2 to service: UUID: 6facfe71-a4c3-4e80-ba5c-533928830727, handle: 0x00b4
[  2629][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[  2638][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=407eda40-6774-11e4-9803-0800200c9a66 to service: UUID: 5cb68410-6774-11e4-9803-0800200c9a66, handle: 0x00c8
[  2655][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[  2664][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=362f71a0-6c96-11e3-981f-0800200c9a66 to service: UUID: 7bdb8dc0-6c95-11e3-981f-0800200c9a66, handle: 0x00dc
[  2680][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=bd066da4-f9ec-4f0d-a53c-1cd99147a641 to service: UUID: 7bdb8dc0-6c95-11e3-981f-0800200c9a66, handle: 0x00dc
[  2697][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=85da4f4b-a2ca-4c7c-8c7d-fcd9e2daad56 to service: UUID: 7bdb8dc0-6c95-11e3-981f2] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 133] ... Unknown
[  2724][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=8ec90003-f315-4f60-9fb8-838830daea50 to service: UUID: 0000fe59-0000-1000-8000-00805f9b34fb, handle: 0x00dc
[  2740][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 133] ... Unknown
[  2750][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=ba7cc552-cc2c-404e-bf75-8778f023787d to service: UUID: 25a22330-820f-11e3-baa7-0800200c9a66, handle: 0x00dc
[  2766][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 133] ... Unknown
[... cut ...]
[  5399][D][BLEAdvertising.cpp:206] start(): - no services advertised
[  5404][D][BLEDevice.cpp:579] getAdvertising(): get advertising
Ready[  5411][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 4]

[  5427][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[  5427][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 1]
[  5439][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[  5440][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 6]

```

paulhamsh, 2022-05-13:
I think it goes wrong here:
``
[  3545][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 362f71a0-6c96-11e3-981f-0800200c9a66, but didnt!
```
After that the handles are all messed up.
Seems to be a bug somewhere to me - and NimBLE doesn't have the same problem.

```
[  3516][D][BLECharacteristic.cpp:90] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: 362f71a0-6c96-11e3-981f-0800200c9a66, service: UUID: 7bdb8dc0-6c95-11e3-981f-0800200c9a66, handle: 0x00dc
[  3535][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[  3545][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 362f71a0-6c96-11e3-981f-0800200c9a66, but didnt!
[  3558][D][BLEService.cpp:109] dump(): Service: uuid:0000fe59-0000-1000-8000-00805f9b34fb, handle: 0xdc
[  3567][D][BLEService.cpp:110] dump(): Characteristics:
handle: 0xffff, uuid: 8ec90003-f315-4f60-9fb8-838830daea50
[  3577][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 362f71a0-6c96-11e3-981f-0800200c9a66, but didnt!
[  3590][D][BLEService.cpp:109] dump(): Service: uuid:25a22330-820f-11e3-baa7-0800200c9a66, handle: 0xdc
[  3599][D][BLEService.cpp:110] dump(): Characteristics:
```

SuGlider, 2022-05-15:
@paulhamsh - is it related to #6739?

paulhamsh, 2022-05-15:
Yes, after investigation it is the same issue!
Shame they can’t just increase the 8 on the standard build - how much more memory would it take?
 Thanks

VojtechBartoska, 2022-12-15:
I'm closing this one as expired. If needed, you can reopen it. Thanks for understanding.

closed 2022-12-15 by VojtechBartoska