espressif/arduino-esp32 #6694: The latest package @4.2.0 doesn't support ARM64 system（RaspberryPi+Vscode+platformio）？

techHappyWater, 2022-05-05:
### Board

ESP32 DEV Module

### Device Description

ESP32 PICO D4

### Hardware Configuration

nan

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Ubuntu

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

When I update the esp32 pakage to V4.2.0, it reminds me that there is no version suitable for this system（arm64）.
Will there be a package for arm64 later?

### Sketch

```cpp
nan
```

### Debug Message

```plain
nan
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-05-05:
I did build the toolchains 8.4.0 [2021r2-patch3](https://github.com/Jason2866/crosstool-NG/releases) for Linux aarch64 and Apple Silicon (arm64). See [https://github.com/espressif/arduino-esp32/discussions/6591](https://github.com/espressif/arduino-esp32/discussions/6591)

techHappyWater, 2022-05-05:
> I did build the toolchains 8.4.0 [2021r2-patch3](https://github.com/Jason2866/crosstool-NG/releases) for Linux aarch64 and Apple Silicon (arm64). See #6591

Thanks for your reply， but I am a green hand， so I didn't really understand how to do this. 
I hope the official can be adapted.

me-no-dev, 2022-05-05:
we do support arm64 with our latest toolchains. They are in master, 2.0.3 and PIO registry. https://github.com/espressif/arduino-esp32/blob/master/package/package_esp32_index.template.json#L92

techHappyWater, 2022-05-06:
> we do support arm64 with our latest toolchains. They are in master, 2.0.3 and PIO registry. https://github.com/espressif/arduino-esp32/blob/master/package/package_esp32_index.template.json#L92

Thanks for your reply, if I wanna update esp32 package or add the source what you said to Platformio(Raspberry), what should I do? Download the .rar directly and copy it to the "/.platformio/platforms" folder?
I am sorry to trouble you but I really didn't know how to do. Thank you very much again.

me-no-dev, 2022-05-06:
PIO Team is working on support for latest Arduino. I can not comment when it will be done (it might already be). Through the Arduino IDE you get it from the Board Manager (2.0.3 is the latest version)

techHappyWater, 2022-05-06:
> PIO Team is working on support for latest Arduino. I can not comment when it will be done (it might already be). Through the Arduino IDE you get it from the Board Manager (2.0.3 is the latest version)

closed 2022-05-06 by techHappyWater

techHappyWater, 2022-05-06:
> PIO Team is working on support for latest Arduino. I can not comment when it will be done (it might already be). Through the Arduino IDE you get it from the Board Manager (2.0.3 is the latest version)

OKOK, Thanks very much! In my computer(X86), PIO can get the latest version is V4.2.0, but in Raspberry(arm64), when I update the version from 3.5.0 to 4.2.0, it reminds me that there is no firmware package for arm64, I think it will happen soon.
Thanks again.

Jason2866, 2022-05-06:
@techHappyWater 
Maybe this is working (a hack tested and working with Ubuntu aarch64) for you.
```
platform = https://github.com/Jason2866/platform-espressif32.git#203/aarch64
```

techHappyWater, 2022-05-09:
> https://github.com/Jason2866/platform-espressif32.git#203/aarch64

It's amazing!! I used your method to successfully solve the problem of error reporting updates!
Thank you very much!!!

techHappyWater, 2022-05-09:
> @techHappyWater Maybe this is working (a hack tested and working with a Ubuntu aarch64 VM) for you.
> 
> ```
> platform = https://github.com/Jason2866/platform-espressif32.git#203/aarch64
> ```

I have s small question: By your ways, I successfully solve the problem of error reporting updates.
But when I try the ways provided with platformio/platform-espressif32 like this:
platform = https://github.com/platformio/platform-espressif32.git

It also remind me that:
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3' requirements for your system 'linux_aarch64'

I am puzzledthat if your project has been modified on the official supply ability, so that it can support arm64?
Thank you.

Jason2866, 2022-05-09:
If you mean Apple Silicon (arm64) this is working.
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip
```

techHappyWater, 2022-05-09:
> If you mean Apple Silicon (arm64) this is working.
> 
> ```
> platform = https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip
> ```

Esp32 official github provided package can not be used for Raspberry + Ubuntu + vscode + PIO, I mean how to generate my own package according to esp32 official github package can be used for Raspberry, but in this part I am not very good at working, so decided to use the project you provided directly. 
Thank you very much again.

techHappyWater, 2022-05-17:
> @techHappyWater Maybe this is working (a hack tested and working with a Ubuntu aarch64 VM) for you.
> 
> ```
> platform = https://github.com/Jason2866/platform-espressif32.git#203/aarch64
> ```

Hello, emm, I meet a quention again(crying), I used your method, 
platform = https://github.com/Jason2866/platform-espressif32.git#203/aarch64
It works well before.
But I built the same project yesterday, it prompt me the following message in the picture. And I tried to solve it but failed. 
[image]

Is this because of a version update issue?

Jason2866, 2022-05-17:
Idk why it is not working anymore for you. Delete the hidden folder `.platformio`
it does solve often strange issues. Platformio will rebuild this folder.

techHappyWater, 2022-05-18:
> Idk why it is not working anymore for you. Delete the hidden folder `.platformio` it does solve often strange issues. Platformio will rebuild this folder.

I tried what you said, and it can work well now.
What a staange problem!
Tkanks!!!!