espressif/arduino-esp32 #6696: ESP32 TTgo T-Display SD-Card weird behaviour

atumra1975, 2022-05-05:
### Board

ESP32 TTGo 

### Device Description

TTgo T-Display (https://github.com/Xinyuan-LilyGO/TTGO-T-Display)

### Hardware Configuration

GPIO 33 --> CS
GPIO 25 --> SCLK     
GPIO 27 --> MISO     
GPIO 26 --> MOSI 
(as described in Factory Test)

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Good evening
I'm facing a little bit weird behaviour about SD-Card shield and ESP-32 Dev board.
I got a standard microSD card shield like this (https://www.amazon.it/Storage-Memory-Shield-modulo-Arduino/dp/B01INGOJJU)
connected to GPIO
GPIO 33 --> CS
GPIO 25 --> SCLK     
GPIO 27 --> MISO     
GPIO 26 --> MOSI 
(as described in Factory Test)
When the code is uploaded and the sketch is running, the microSD card seems to be correctly mounted, readable, ecc.
For the first 3 times (3 reboot), evertything works perfectly... but onto the last reboot the card will broke
sdcard_mount(): f_mount failed: (3) The physical drive cannot work
The microSD card is readable in any pc but in the shield
I've tryed tons of microSD card... results are the same
I cannot find the solution :/ :/ :/
Thanks

### Sketch

```cpp
/*
 * Definizione SDCard
 */
#define ENABLE_SPI_SDCARD
#ifdef ENABLE_SPI_SDCARD

#include "FS.h"
#include "SD.h"

SPIClass SDSPI(HSPI);

#define MY_CS       33
#define MY_SCLK     25
#define MY_MISO     27
#define MY_MOSI     26
bool setupSDCard()
{
  SDSPI.begin(MY_SCLK, MY_MISO, MY_MOSI, MY_CS);
    if (!SD.begin(MY_CS, SDSPI)) {
      return false;
    }
    else
    {
      return true;
    }
}
#else
#define setupSDCard()
#endif
```

### Debug Message

```plain
[W][sd_diskio.cpp:535] ff_sd_initialize(): APP_OP_COND failed: 1
[E][sd_diskio.cpp:775] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
Errore SDCard...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atumra1975, 2022-05-14:
Indeed...i finally found the problem... The uSD Card i used to for were incopatible
Then i reverted to the latest lib and everything works!
Thanks

VojtechBartoska, 2022-05-16:
Closing as solved, thanks for your contribution and good luck with all your projects!

closed 2022-05-16 by VojtechBartoska