espressif/arduino-esp32 #6697: HardwareSerial::availableForWrite still return 128 after HardwareSerial::setTxBufferSize

imwhocodes, 2022-05-05:
### Board

ESP32 Dev Module

### Device Description

Custom PCB

### Hardware Configuration

A lot of stuff

### Version

latest master (checkout manually)

### IDE Name

Platformio

### Operating System

Macos

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm trying to use the new feature of **v2.0.3**

After setting _TX buffer size_ with `HardwareSerial::setTxBufferSize` the function `HardwareSerial::availableForWrite` still always return `128` keeping track only of the free space in the _TxFIFO_  without accounting of the newly allocated buffer for transmission

### Sketch

```cpp
#define NO_GLOBAL_SERIAL
#include <Arduino.h>

HardwareSerial SerialTest{1};

void setup(){
  SerialTest.setRxBufferSize(4096);
  SerialTest.setTxBufferSize(4096);
  SerialTest.begin(9600);
}

void loop() {
  log_e("availableForWrite:\t%d\n", SerialTest.availableForWrite());
}
```

### Debug Message

```plain
availableForWrite: 128
availableForWrite: 128
availableForWrite: 128
availableForWrite: 128
availableForWrite: 128
availableForWrite: 128
....
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-05-05:
@imwhocodes - Issue confirmed. Along the changes this fuction was left behind.
 `availableForWrite()` only looks into the FIFO space. It has to be updated/fixed.

Thanks for reporting. I'll set a PR to fix it.

SuGlider, 2022-05-06:
I just looked into the code.... there is a big problem.
IDF has no function to get the left space in TX Ring Buffer associated to  `setTxBufferSize(size)`.
The Ring Buffer is created by IDF, but it can't be read, neither accessed.

I'll investigate it further, but it looks like there is no way to fix it at this time.

imwhocodes, 2022-05-10:
I opened an issue / feature request on esp-idf, let see what they say

https://github.com/espressif/esp-idf/issues/8932

imwhocodes, 2022-07-12:
> I opened an issue / feature request on esp-idf, let see what they say
> 
> [espressif/esp-idf#8932](https://github.com/espressif/esp-idf/issues/8932)

@SuGlider issue was fixed on esp-idf repo, we need to update Arduino internal code

SuGlider, 2022-07-12:
I think it may be availble for a Arduino Core 2.0.5 (next release).
I'll add the fixes in the Arduino API and post a PR ready for when we have this IDF function in IDF 4.4.

SuGlider, 2022-07-16:
It has been backported to 4.4: https://github.com/espressif/esp-idf/commit/dfb75aa230354e684510ce7689da69424b30340f

SuGlider, 2022-07-16:
@imwhocodes - PR #6998 shall solve the issue. Thanks for the help!

PR #6998 (2022-08-08): Fixes HardwareSerial::availableForWrite + setTxBufferSize
## Description of Change
Fixes an issue with `HardwareSerial::availableForWrite()` that only returns TX FIFO available space even when there is a TX RingBuffer in use through `HardwareSerial::setTxBufferSize()`.
This isse was fixed by IDF 4.4 https://github.com/espressif/esp-idf/commit/dfb75aa230354e684510ce7689da69424b30340f

new IDF UART API `esp_err_t uart_get_tx_buffer_free_size(uart_port_t uart_num, size_t *size)` has been incorporated to the code in [`uint32_t uartAvailableForWrite(uart_t* uart)`](https://github.com/espressif/ardu [... cut]