espressif/arduino-esp32 #6703: ESP32-S2 TCPIPV6

asvith, 2022-05-06:
### Board

ESP32S2MINI1U

### Device Description

ESP32-S2-DevkitM-1 v1.
[esp32-s2-mini-1_esp32-s2-mini-1u_datasheet_en.pdf](https://github.com/espressif/arduino-esp32/files/8638830/esp32-s2-mini-1_esp32-s2-mini-1u_datasheet_en.pdf)
0

### Hardware Configuration

ESP32-S2-DevkitM-1
v1.0

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

started working on ESP32S2MINI1U module, we found that  IPV6 TCP sample code is not available for testing, whereas it is found only IPV6 UDP sample code is available and it is working.. 
Please send the sample code for ipv6tcp for esp32s2mini1u arduino based libraries.

### Sketch

```cpp
we don't have a code . Please send us a sample code.
```

### Debug Message

```plain
no debug information.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-05-18:
Hello ESP32-DevKitM-1 (Singlecore) is not supported.

Related issue where this is tracked: https://github.com/espressif/arduino-esp32/issues/5183

VojtechBartoska, 2022-06-14:
PR was merged. I'm closing this, if needed, please reopen the issue.

VojtechBartoska, 2022-07-26:
@asvith Supposed to be solved now. Are you able to retest it under v2.0.4? Thanks!

VojtechBartoska, 2022-08-23:
Closing as solved, if you are still facing the issue please reopen it.

closed 2022-08-23 by VojtechBartoska