espressif/arduino-esp32 #6704: ESP32-S2 Webserver Library

asvith, 2022-05-06:
### Board

ESP32S2MINI1U

### Device Description

[esp32-s2-mini-1_esp32-s2-mini-1u_datasheet_en.pdf](https://github.com/espressif/arduino-esp32/files/8638842/esp32-s2-mini-1_esp32-s2-mini-1u_datasheet_en.pdf)

### Hardware Configuration

ESP32-S2-DevkitM-1
v1.0

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

We are facing one more issue with web server library that is as we are sending request in "jquery.js" format with more than 60kb web server library is not supporting for esp32s2 , where as with esp8266 web server library there is no limitation with size.  we also tried to include esp8266 web server library to  esp32s2 its not supporting .we also tried webserver with "spiffs" library but it is supporting only "esptool.py" tool we have library with big sur tool so we are unable to proceed further.
Kindly support to resolve this issue to progress further.
[image]
 

### Sketch

```cpp
webserver.h library is used for web server but it is limited upto 60kb ("jquery.js" file uploaded more than 60 kb) so we are facing issue.
```

### Debug Message

```plain
This compiling and running but "jquery.js" is not accepting more than 60kb.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-05-06:
@asvith - 
I need more information about it.

Where are the web files in the S2? In SPIFFS folders?
What limitation is that about? SPIFSS?
Could you describe what you need to achieve with more details, including how is it setup in the ESP8266?
Is there any github with an example that we can try to reproduce and debug the issue?

SuGlider, 2022-05-06:
@lbernstone - In that case the JS files goes with the compiled blob, right?
Is it possible that the issue that @asvith has is related to the default Partition model, which may not have enough space?

@asvith - Is there  any error message in the console of you Arduino IDE?

lbernstone, 2022-05-06:
His issue may certainly be about partitioning.  More info needed.

My design philosophy is that if you are using a library (and javascipt like jquery is certainly a library in this case), it should be embedded in your code rather than a separate file.  This means you know exactly what version is being delivered, and it gets replaced/updated with whatever code is calling the script.  The file system should be reserved for use by user or sensor data that will be generated on the device.  The file upload features may be handy for testing, but they get messy in any sort of long-term support situation where you want consistency in your end product.

asvith, 2022-05-09:
@asvith - Is there any error message in the console of you Arduino IDE?
@lbernstone - no error message in console

asvith, 2022-05-09:
we are not using any files, directly in code we are using char array.

lbernstone, 2022-05-09:
You are using a <b>const</b> char array?

asvith, 2022-05-09:
yes

lbernstone, 2022-05-09:
Please provide a minimal example that reproduces your error. Remove a large section of the 60k data like as in my example.

asvith, 2022-05-09:
// Example code more than 60kb data....
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>

const char* ssid = "YOUR SSID";
const char* password = "YOUR PASSWORD";

const char test[]PROGMEM = "Lorem ipsum dolor sit amet, consectetur adipiscing elit. Vivamus condimentum sagittis lacus, laoreet luctus ligula laoreet ut. Vestibulum ullamcorper accumsan velit vel vehicula. Proin tempor lacus arcu. Nunc at elit condimentum, semper nisi et, condimentum mi. In venenatis blandit nibh at sollicitudin. Vestibulum dapibus mauris at orci maximus pellentesque. Nullam id elementum ipsum. Suspendisse cursus lobortis viverra. Proin et erat at mauris tincidunt porttitor vitae ac dui.Donec vulputate lorem tortor, nec fermentum nibh bibendum vel. Lorem ipsum dolor sit amet, consectetur adipiscing elit. Praesent dictum luctus massa, non euismod lacus. Pellentesque condimentum dolor est, ut dapibus lectus luctus ac. Ut sagittis commodo arcu. Integer nisi nulla, facilisis sit amet nulla quis, eleifend suscipit purus. Class aptent taciti sociosqu ad litora torquent per conubia nostra, per inceptos himenaeos. Aliquam euismod ultrices lorem, sit amet imperdiet est tincidunt vel. Phasellus dictum justo sit amet ligula varius aliquet auctor et metus. Fusce vitae tortor et nisi pulvinar vestibulum eget in risus. Donec ante ex, placerat a lorem eget, ultricies bibendum purus. Nam sit amet neque non ante laoreet rutrum. Nullam aliquet commodo urna, sed ullamcorper odio feugiat id. Mauris nisi sapien, porttitor in condimentum nec, venenatis eu urna. Pellentesque feugiat diam est, at rhoncus orci porttitor non.Nulla luctus sem sit amet nisi consequat, id ornare ipsum dignissim. Sed elementum elit nibh, eu condimentum orci viverra quis. Aenean suscipit vitae felis non suscipit. Suspendisse pharetra turpis non eros semper dictum. Etiam tincidunt venenatis venenatis. Praesent eget gravida lorem, ut congue diam. Etiam facilisis elit at porttitor egestas. Praesent consequat, velit non vulputate convallis, ligula diam sagittis urna, in venenatis nisi justo ut mauris. Vestibulum posuere sollicitudin mi, et vulputate nisl fringilla non. Nulla ornare pretium velit a euismod. Nunc sagittis venenatis vestibulum. Nunc sodales libero a est ornare ultricies. Sed sed leo sed orci pellentesque ultrices. Mauris sollicitudin, sem quis placerat ornare, velit arcu convallis ligula, pretium finibus nisl sapien vel sem. Vivamus sit amet tortor id lorem consequat hendrerit. Nullam at dui risus.Vestibulum ante ipsum primis in faucibus orci luctus et ultrices posuere cubilia Curae; Lorem ipsum dolor sit amet, consectetur adipiscing elit. Sed feugiat semper velit consequat facilisis. Etiam facilisis justo non iaculis dictum. Fusce turpis neque, pharetra ut odio eu, hendrerit rhoncus lacus. Nunc orci felis, imperdiet vel interdum quis, porta eu ipsum. Pellentesque dictum sem lacinia, auctor dui in, malesuada nunc. Maecenas sit amet mollis eros. Proin fringilla viverra ligula, sollicitudin viverra ante sollicitudin congue. Donec mollis felis eu libero malesuada, et lacinia risus interdum.Etiam vitae accumsan augue. Ut urna orci, malesuada ut nisi a, condimentum gravida magna. Nulla bibendum ex in vulputate sagittis. Nulla facilisi. Nullam faucibus et metus ac consequat. Quisque tempor eros velit, id mattis nibh aliquet a. Aenean tempor elit ut finibus auctor. Sed at imperdiet mauris. Vestibulum pharetra non lacus sed pulvinar. Sed pellentesque magna a eros volutpat ullamcorper. In hac habitasse platea dictumst. Donec ipsum mi, feugiat in eros sed, varius lacinia turpis. Donec vulputate tincidunt dui ac laoreet. Sed in eros dui. Pellentesque placerat tristique ligula eu finibus. Proin nec faucibus felis, eu commodo ipsum.Integer eu hendrerit diam, sed consectetur nunc. Aliquam a sem vitae leo fermentum faucibus quis at sem. Etiam blandit, quam quis fermentum varius, ante urna ultricies lectus, vel pellentesque ligula arcu nec elit. Donec placerat ante in enim scelerisque pretium. Donec et rhoncus erat. Aenean tempor nisi vitae augue tincidunt luctus. Nam condimentum dictum ante, et laoreet neque pellentesque id. Curabitur consectetur cursus neque aliquam porta. Ut interdum nunc nec nibh vestibulum, in sagittis metus facilisis. Pellentesque feugiat condimentum metus. Etiam venenatis quam at ante rhoncus vestibulum. Maecenas suscipit congue pellentesque. Vestibulum suscipit scelerisque fermentum. Nulla iaculis risus ac vulputate porttitor.Mauris nec metus vel dolor blandit faucibus et vel magna. Ut tincidunt ipsum non nunc dapibus, sed blandit mi condimentum. Quisque pharetra interdum quam nec feugiat. Sed pellentesque nulla et turpis blandit interdum. Curabitur at metus vitae augue elementum viverra. Sed mattis lorem non enim fermentum finibus. Sed at dui in magna dignissim accumsan. Proin tincidunt ultricies cursus. Maecenas tincidunt magna at urna faucibus lacinia.Quisque venenatis justo sit amet tortor condimentum, nec tincidunt tellus viverra. Morbi risus ipsum, consequat convallis malesuada non, fermentum non velit. Nulla facilisis orci eget ligula mattis fermentum. Aliquam vel velit ultricies, sollicitudin nibh eu, congue velit. Donec nulla lorem, euismod id cursus at, sollicitudin et arcu. Proin vitae tincidunt ipsum. Vivamus elementum eleifend justo, placerat interdum nulla rutrum id.Phasellus fringilla luctus magna, a finibus justo dapibus a. Nam risus felis, rhoncus eget diam sit amet, congue facilisis nibh. Inter [... cut ...]
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp32")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
  delay(2);//allow the cpu to switch to other tasks
}

check the output....

SuGlider, 2022-05-09:
@asvith - it didn't compile....
I had to change it here:

``` cpp
server.on("/inline", {
server.send(200, "text/plain", "this works as well");
});
```

I changed it to use an external function:
```
void handlePlain() {
  server.send(200, "text/plain", "this works as well");
}

...

  server.on("/inline", handlePlain);
```

### Reseults:

I was able to retrive 68,262 bytes from the WebServer...
It worked fine in a ESP32 WROVER board.

guestisp, 2022-05-09:
> I serve javascript files with esp32 without any issues. Here's some tricks. Use `xxd -i <filename>` to convert the file into a char array. Add that into your code directly, and then deliver it using send_P. By making it into a const char*, it will be delivered directly from flash and not use up precious RAM (like PROGMEM in esp8266). You can also gzip it first to make the file even smaller (in your code and on the wire). Here's an example:
> 
> ```
> // Where you define your WebServer config
>   server.on("/files/jquery.min.js", []() {
>     server.sendHeader("Content-Encoding", "gzip");
>     server.send_P(200, "application/javascript", jquery_min_js_gz, jquery_min_js_gz_len); });
> 
> // Anywhere in your code
> // jquery.js 3.6.0
> const unsigned int jquery_min_js_gz_len = 30761;
> const char jquery_min_js_gz[] = {
>   0x1f, 0x8b, 0x08, 0x08, 0x8c, 0x8f, 0x58, 0x62, 0x02, 0x03, 0x6a, 0x71,
>   0x75, 0x65, 0x72, 0x79, 0x2e, 0x6d, 0x69, 0x6e, 0x2e, 0x6a, 0x73, 0x00,
> // The other 30k of data...
>   0xbb, 0xc5, 0xe5, 0x64, 0xf2, 0xff, 0xfc, 0x1f, 0x84, 0x5b, 0x9a, 0x60,
>   0x8f, 0x53, 0x01, 0x00
> };
> ```

Trying this. Works for small files, but bigger files trigger an `net::ERR CONTENT LENGTH MISMATCH` error on the browser because the trnasfer is interrupted (i had the same even serving files from SPIFFS).

A workaround for SPIFF  was to send chunked data, but how to send chunked data from const char array ?

lbernstone, 2022-05-09:
server.send and server.send_P are different functions.  send_P requires you to include the string length, because it does not assume you are sending \0x0 terminated strings.  send() is not appropriate for binary text, and for long strings like this, I would recommend using send_P and knowing the length of your strings.  Here's my example derived from yours.  Note that strlen only works for zero terminated strings- you need to know the length of your strings in C code!.
```
#include <WebServer.h>

const char* test1 = R"literal(
Lorem ipsum dolor sit amet, consectetur adipiscing elit. 
Vivamus condimentum sagittis lacus, laoreet luctus ligula laoreet ut. 
Vestibulum ullamcorper accumsan velit vel vehicula. 
Proin tempor lacus arcu. 
Nunc at elit condimentum, semper nisi et, condimentum mi. 
In venenatis blandit nibh at sollicitudin. 
Vestibulum dapibus mauris at orci maximus pellentesque. 
Nullam id elementum ipsum. 
Suspendisse cursus lobortis viverra. 
Proin et erat at mauris tincidunt porttitor vitae ac dui.
... //repeat many times
Donec vulputate lorem tortor, nec fermentum nibh bibendum vel. 
)literal";
const char* test2 = "this is a short test";

WebServer server(80);

void setup(void) {
  Serial.begin(115200);
  WiFi.begin("ssid", "password");
  WiFi.waitForConnectResult();
  
  Serial.printf("Connected to %s\nIP address: ",ssid);
  Serial.println(WiFi.localIP());
  Serial.printf("test data is %d bytes\n", strlen(test1));
  server.on("/", [](){server.send_P(200, "text/plain", test1, strlen(test1));});
  server.on("/test2", [](){server.send(200, "text/plain", test2);});
  server.onNotFound([](){server.send(404,"text/plain", "File not found");});
  server.begin();
}

void loop(void) {
  server.handleClient();
  delay(150);//allow the cpu to switch to other tasks
}

output:
07:30:41.397 -> Connected to TP-Link_1404
07:30:41.397 -> IP address: 192.168.0.120
07:30:41.397 -> test data is 80065 bytes

lbernstone@ideapad:/~$ curl http://192.168.0.120/ | wc -c
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 80065  100 80065    0     0  27164      0  0:00:02  0:00:02 --:--:-- 27168
80065
```

closed 2022-05-11 by me-no-dev