espressif/arduino-esp32 #6706: WiFi softAP extremely slow

rtrbt, 2022-05-06:
### Board

ESP32-Ethernet-Kit V1.2, Tinkerforge ESP32 Brick (all flashed as ESP32 Dev Module)

### Device Description

ESP32 Ethernet Kit: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html#get-started-esp32-ethernet-kit-v1-2

ESP32 Brick: https://www.tinkerforge.com/en/doc/Hardware/Bricks/ESP32_Brick.html
ESP32 Ethernet Brick: https://www.tinkerforge.com/en/doc/Hardware/Bricks/ESP32_Ethernet_Brick.html

### Hardware Configuration

Nothing else is attached to the boards.

### Version

other

### IDE Name

Arduino IDE

### Operating System

Arch Linux

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

The WiFi performance of clients connected to an ESP running as softAP is extremely bad in my tests. I'm using the following test setup:
- An ESP32-Ethernet-Kit V1.2 running the arduino-esp32 WiFiAccessPoint example (stripped down because I don't need the HTTP server code. See attached sketch)
- Two Tinkerforge ESP32 Ethernet Bricks, both running the ESP-IDF (v4.4.1) Iperf example: https://github.com/espressif/esp-idf/tree/master/examples/wifi/iperf

Both clients are connected to the access point, one runs as Iperf server, one as client with the following results:
(Server)
```
iperf> sta yourAP yourPassword
I (71521) cmd_wifi: sta connecting to 'yourAP'
I (73029) esp_netif_handlers: sta ip: 192.168.4.2, mask: 255.255.255.0, gw: 192.168.4.1
iperf> iperf -s -i 3
I (112400) cmd_wifi: mode=tcp-server sip=192.168.4.2:5001, dip=0.0.0.0:5001, interval=3, time=30
I (112402) iperf: Socket created
iperf> W (123592) wifi:<ba-add>idx:0 (ifx:0, c4:dd:57:b8:16:e9), tid:0, ssn:0, winSize:64
I (123606) iperf: accept: 192.168.4.3,54260

        Interval Bandwidth
   0-   3 sec       0.03 Mbits/sec
   3-   6 sec       0.02 Mbits/sec
   6-   9 sec       0.02 Mbits/sec
   9-  12 sec       0.02 Mbits/sec
  12-  15 sec       0.01 Mbits/sec
  15-  18 sec       0.01 Mbits/sec
  18-  21 sec       0.01 Mbits/sec
  21-  24 sec       0.02 Mbits/sec
  24-  27 sec       0.01 Mbits/sec
  27-  30 sec       0.01 Mbits/sec
   0-  30 sec       0.02 Mbits/sec
I (153688) iperf: TCP Socket server is closed.
[... 182 lines cut ...]

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
[⸮⸮mum⸮⸮⸮⸮2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

Configuring access point...
[    29][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   114][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 10 - AP_START
AP IP address: 192.168.4.1
Server started
[ 13349][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[ 13422][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[ 44244][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[ 44286][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

rtrbt, 2022-05-09:
Some additional tests with two ESPs as clients running the ESP-IDF 4.4.1 IPerf example as before and another ESP running the arduino-esp32 softAP example. 

```
arduino-esp32 1.0.6

        Interval Bandwidth
   0-   3 sec       2.37 Mbits/sec
   3-   6 sec       0.11 Mbits/sec
   6-   9 sec       0.34 Mbits/sec
   9-  12 sec       0.37 Mbits/sec
  12-  15 sec       0.96 Mbits/sec
  15-  18 sec       0.40 Mbits/sec
  18-  21 sec       0.55 Mbits/sec
  21-  24 sec       1.02 Mbits/sec
  24-  27 sec       0.11 Mbits/sec
  27-  30 sec       0.89 Mbits/sec
   0-  30 sec       0.71 Mbits/sec

arduino-esp32 2.0.0 RC1

        Interval Bandwidth
   0-   3 sec       0.33 Mbits/sec
   3-   6 sec       0.19 Mbits/sec
   6-   9 sec       0.03 Mbits/sec
   9-  12 sec       0.82 Mbits/sec
  12-  15 sec       0.09 Mbits/sec
  15-  18 sec       0.03 Mbits/sec
  18-  21 sec       0.08 Mbits/sec
  21-  24 sec       0.28 Mbits/sec
  24-  27 sec       0.06 Mbits/sec
  27-  30 sec       0.64 Mbits/sec
   0-  30 sec       0.25 Mbits/sec

arduino-esp32 2.0.0

        Interval Bandwidth
   0-   3 sec       0.74 Mbits/sec
   3-   6 sec       0.05 Mbits/sec
   6-   9 sec       0.04 Mbits/sec
   9-  12 sec       0.42 Mbits/sec
  12-  15 sec       0.05 Mbits/sec
  15-  18 sec       0.47 Mbits/sec
  18-  21 sec       0.01 Mbits/sec
  21-  24 sec       0.84 Mbits/sec
  24-  27 sec       0.03 Mbits/sec
  27-  30 sec       0.02 Mbits/sec
   0-  30 sec       0.27 Mbits/sec

arduino-esp32 2.0.1

        Interval Bandwidth
   0-   3 sec       0.05 Mbits/sec
   3-   6 sec       0.09 Mbits/sec
   6-   9 sec       0.05 Mbits/sec
   9-  12 sec       0.03 Mbits/sec
  12-  15 sec       0.04 Mbits/sec
  15-  18 sec       0.06 Mbits/sec
  18-  21 sec       0.04 Mbits/sec
  21-  24 sec       0.08 Mbits/sec
  24-  27 sec       0.03 Mbits/sec
  27-  30 sec       0.08 Mbits/sec
   0-  30 sec       0.06 Mbits/sec
```

For comparision: with a "proper" WiFi router as AP:
```
        Interval Bandwidth
   0-   3 sec       11.91 Mbits/sec
   3-   6 sec       14.11 Mbits/sec
   6-   9 sec       14.50 Mbits/sec
   9-  12 sec       15.12 Mbits/sec
  12-  15 sec       15.38 Mbits/sec
  15-  18 sec       15.18 Mbits/sec
  18-  21 sec       15.35 Mbits/sec
  21-  24 sec       15.11 Mbits/sec
  24-  27 sec       15.76 Mbits/sec
  27-  30 sec       15.42 Mbits/sec
   0-  30 sec       14.78 Mbits/sec
```

guestisp, 2022-05-09:
Saw something similiar right now.
When ESP32 is running in AP mode, like for a captive portal, transferring files from the ESP32 to the wifi client is very very slow (the captive portar load after 20-30 seconds, if not timeout). The exact same sketch but with ESP32 connected to a wifi network and the same client reaching the esp32 via the wifi network , is very fast, the page loads almost in no time.

SonnyZheng, 2023-02-01:
Facing the same problem, any solution? Thanks!

guestisp, 2023-02-02:
i was able to solve only using ESP Arduino 1.x

SuGlider, 2023-02-05:
This will require some time. But it is in the list for investigation and fixing.

ttlappalainen, 2023-04-08:
I have the same problem. 

I use AsyncTCP and ESPAsyncWebServer with PlatformIO. I use currently Espressif 32 3.5.0. With it SoftAP is pretty fast and usable. I have been trying to move to Espressif 32 6.1.0, but there SoftAP can not be used.

Saffarofski, 2023-04-24:
I'm facing the same issue. 
I'm using esp32 wrover module with custome board and the speed rate is very low on AP mode but It's good in station mode.
Another thing to mention is that this problem disappears whenever I touch the antenna ! (doesn't matter if It's external or internal)
 is it related to software or hardware?

ttlappalainen, 2023-04-28:
As it works well with 3.5.0, sounds strange that touching antenna has anything to do with it.

jasongaunt, 2023-05-26:
I too am having this same issue. Multiple different sketches I've written over the last 12-18 months for ESP32's using PlatformIO and espressif32 all have one thing in common; _Intermittent / random_ massive slowdowns in WiFi performance. 

Sometime's they work fine, sometimes they drop packets heavily, sometimes a reboot or a power cycle fixes it, sometimes it does not.

I never got to the bottom of this. I see the same performance issues depending on whether I'm connecting to it (access point) or if it's connected to my WiFi network.

I don't know if this information will be of any use but it may help narrowing down the cause;

**Platform:** M5Stack Stamp Pico (ESP32-PICO-D4)
**Codebase:** Espressif 32 (5.0.0) with the following dependency graph...
```
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.220619 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3

Dependency Graph
|-- FastLED @ 3.5.0
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- AsyncTCP @ 1.1.1+sha.ca8ac5f
|-- ArduinoJson @ 6.21.2
|-- LittleFS @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- DNSServer @ 2.0.0
|-- WiFi @ 2.0.0
```

**platformio.ini**
```ini
[env]
platform = espressif32
framework = arduino
monitor_speed = 115200
build_flags = -Wno-write-strings
lib_deps =
	fastled/FastLED
	https://github.com/me-no-dev/ESPAsyncWebServer
	https://github.com/me-no-dev/AsyncTCP
	bblanchon/ArduinoJson
extra_scripts =
	pre:scripts/prebuild.py

[env:m5stack-pico]
board = pico32
board_build.partitions = scripts/stamp_partition_layout.csv
monitor_port = COM1
upload_speed = 921600
upload_port = COM1
monitor_filters = esp32_exception_decoder

[env:m5stamp-ota]
board = pico32
board_build.partitions = scripts/stamp_partition_layout.csv
upload_protocol = espota
upload_port = pico32.lan
```

**Init routine**
```c++
// Initialises WiFi as an AP and sets up captive portal DNS
void initWiFi() {
  Serial.print("Starting WiFi SoftAP... ");
  WiFi.useStaticBuffers(true);
  WiFi.setHostname(WIFI_HOSTNAME);
  WiFi.hostname(WIFI_HOSTNAME);
  WiFi.mode(WIFI_MODE_APSTA);
  IPAddress SOFTAP_IP(169, 254, 200, 1);
  IPAddress SOFTAP_NM(255, 255, 255, 0);
  IPAddress SOFTAP_GW(169, 254, 200, 1);
  WiFi.softAP(WIFI_AP_SSID, WIFI_AP_PASS);
  delay(200);
  WiFi.softAPConfig(SOFTAP_IP, SOFTAP_GW, SOFTAP_NM);
  dnsServer.start(DNS_PORT, "*", SOFTAP_IP);
  WiFi.onEvent(WiFiStationGotIP, ARDUINO_EVENT_WIFI_STA_GOT_IP);
  Serial.printf("done.\nYou can connect with the following details:\nSSID: %s\nPass: %s\n", WIFI_AP_SSID, WIFI_AP_PASS);
}

void setup()
{
  initWiFi();
  // Usual ArduinoOTA init code here..
  // My setup code here..
}

void loop() {
  ArduinoOTA.handle();
  dnsServer.processNextRequest();
  // My loop code here..
}
```

closed 2025-01-16 by Parsaabasi