espressif/arduino-esp32 #6708: function setBatteryLevel()  in BLEHIDDevice.cpp is invalid

BG2CRW, 2022-05-06:
### Board

ESP32 DevModule

### Device Description

a ble keyboard use ESP32 DevModule

### Hardware Configuration

BLE Server

### Version

v2.0.2

### IDE Name

arduino ide

### Operating System

win10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

BLEHIDDevice hid;
hid->setBatteryLevel(level);    

esp32实现的蓝牙键盘(server 模式)，用这种方法进行电量设置，在win10系统下显示键盘电量。
在初次连接时，电量显示正常，电量也可变化。但是断开连接后重新连接，电量始终显示为初次连接时的电量，且更改变量赋值后，win10系统显示变量仍然无变化。

I want to use this way to make my BLE keyboard change battery level in win10. but it is invalid.
When I connect my keyboard first time, it is work, meanwhile, win10 can show my real-time battery level.
but after disconnect and reconnect, win10 can only show its last battery level, and can't show my realt-ime battery level, when I change my battery level variable.

### Sketch

```cpp
BLEHIDDevice hid;
hid->setBatteryLevel(level);
```

### Debug Message

```plain
no debug message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

BG2CRW, 2022-05-06:
I find this page https://github.com/espressif/arduino-esp32/pull/4517, it seems to deal with this bug, but I think it is not work

BG2CRW, 2022-05-06:
oh, I fix this bug, in BLEHIDDevice.cpp #45

this is origon:
	/*
	 * Mandatory battery level characteristic with notification and presence descriptor
	 */
	BLE2904* batteryLevelDescriptor = new BLE2904();
	batteryLevelDescriptor->setFormat(BLE2904::FORMAT_UINT8);
	batteryLevelDescriptor->setNamespace(1);
	batteryLevelDescriptor->setUnit(0x27ad);

	m_batteryLevelCharacteristic = m_batteryService->createCharacteristic((uint16_t) 0x2a19, BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_NOTIFY);
	m_batteryLevelCharacteristic->addDescriptor(batteryLevelDescriptor);
	m_batteryLevelCharacteristic->addDescriptor(new BLE2902());

this is my fix:

	/*
	 * Mandatory battery level characteristic with notification and presence descriptor
	 */
	BLE2904* batteryLevelDescriptor = new BLE2904();
	batteryLevelDescriptor->setFormat(BLE2904::FORMAT_UINT8);
	batteryLevelDescriptor->setNamespace(1);
	batteryLevelDescriptor->setUnit(0x27ad);

	m_batteryLevelCharacteristic = m_batteryService->createCharacteristic((uint16_t) 0x2a19, BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_NOTIFY);
	m_batteryLevelCharacteristic->addDescriptor(batteryLevelDescriptor);
	m_batteryLevelCharacteristic->addDescriptor(new BLE2904());

mrengineer7777, 2022-05-06:
https://github.com/espressif/arduino-esp32/blob/0b3f1a9fa9bb5cc4df9169fe9130f324cbcba9db/libraries/BLE/src/BLEHIDDevice.cpp#L45

Looks like you've found a possible bug.  Changing from `new BLE2902()` to `new BLE2904()` causes it to work?

BG2CRW, 2022-05-07:
> https://github.com/espressif/arduino-esp32/blob/0b3f1a9fa9bb5cc4df9169fe9130f324cbcba9db/libraries/BLE/src/BLEHIDDevice.cpp#L45
> 
> Looks like you've found a possible bug. Changing from `new BLE2902()` to `new BLE2904()` causes it to work?

yes,  it works.  I don't know why. I only think it should same as "BLE2904* batteryLevelDescriptor = new BLE2904()"

SuGlider, 2022-05-07:
@BG2CRW Thank you for reporting and investigating it!

I think it is neessary to investigate the issue.

Descriptor 0x2904 aka "Characteristic Presentation Format" is already added by Line 44.

Line 45 adds another Descriptor 0x2902, which is the notify descriptor - necessary for notifying Win10 that the value has changed.

BG2CRW, 2022-05-07:
I’m looking forward to the right code.

BG2CRW, 2022-05-09:
if line 45 is 2902, it won't change battery level since the second connect.
if line 45 is 2904, it will change battery level, but if I delete the bond on my PC and reconnect it to my PC, the battery icon won't appear.
I want help!

SuGlider, 2022-05-09:
Based on a fast research, Battery Level Service uses 2901 and 2902, instead of 2904 and 2902.

I need to confirm it and then create the code for the 2901, which is different from 2904.

BG2CRW, 2022-05-09:
@SuGlider I copy BLE2904.h and BLE2904.cpp as BLE2901.h and BLE2901.h, and replace 2901 as 2904. Then change 2904 in BLEHIDDevice.cpp.  
It can compile and the result is same as 2904. So it not work.

SuGlider, 2022-05-23:
After some investigation, it seems that Windows only shows the battery once connected.
Win10 and Win11 don't have a "real-time" update feature for the battery indicator in the Bluetooth Setting window.

Some keyboards will offer a separated application that monitors the battery and display it as an icon in the Taskbar.
Some other will also pop an alert when the battery goes down, under a certain level.

But it sounds like Windows itself doesn't monitor the battery from BLE/Bluetooth devices.

BG2CRW, 2022-05-23:
But, if you test it like me. first, burn it when line 45 is "2902", and connect with your WIN11. Second, change  line 45 as 2904 and burn it. Then, the battery level can change real-time whenever connect it to your PC.

BG2CRW, 2022-05-23:
Meanwhile, I use the NRF library to code keyboard, it can change battery level real-time. I trust that, esp32  also can change bettery level real-time on win11.

SuGlider, 2022-05-23:
@BG2CRW - where in Windows you see the battery level changing? In the Bluetooth settings?
Could you send a screenshot of the window where you see the battery indicator?

BG2CRW, 2022-05-23:
[image]

BG2CRW, 2022-05-23:
you can use this test .https://github.com/T-vK/ESP32-BLE-Keyboard

and change the main function as:

#include "BleKeyboard.h"

BleKeyboard bleKeyboard;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleKeyboard.begin();
}

void loop() {
  if(bleKeyboard.isConnected()) {
    bleKeyboard.setBatteryLevel(0);
    delay(1000);
    bleKeyboard.setBatteryLevel(25);
    delay(1000);
    bleKeyboard.setBatteryLevel(50);
    delay(1000);
    bleKeyboard.setBatteryLevel(75);
    delay(1000);
    bleKeyboard.setBatteryLevel(100);
    delay(1000);
  }

  delay(5000);
}

SuGlider, 2022-05-23:
OK, thanks.
Question: Do you see the number (%) changing in the same window from the screenshot?

BG2CRW, 2022-05-23:
step1: if line 45 is "m_batteryLevelCharacteristic->addDescriptor(new BLE2902());", you can find that the battery level change  once connect, but don't change after reconnect.

step1: ”//“ line 45, and reconnect to your PC, you can find that it can change.

step3: if you delete the connection and reconnect, you can find that the battery icon disappear.

step4: you return the step1.

BG2CRW, 2022-05-23:
whether  you have  reproduced my phenomenon？

BG2CRW, 2022-05-23:
@SuGlider 
That depends on what line 45 is.

SuGlider, 2022-05-24:
OK... I just got it to work... which is actually weird.
No change to the library...

I just compiled and uploaded the code... I can see the INPUT device name ESP32 Keyboard, connected and its battery changes from 0, 25, 50, 75, 100% with the batery icon changing as well (empty to full). When it is 0%, a little widow shows up in the bottom right corner saying the the keybord has no battery.

In other words... it works and I just can't reproduce the issue.
It took a while to connect to show up and connect to the Windows 11...

### Important detail:

If I reset the ESP32 board, it says it has reconneced, but the **Battery indicator stops updating**.
I have to remove it from the paired devices and then scan it again, connect it again and then it works fine.

------------------------------------
This is the final sketch (few changes to the one you sent me):

``` cpp
#include "BleKeyboard.h"

BleKeyboard bleKeyboard;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleKeyboard.begin();
}

void loop() {
  if (bleKeyboard.isConnected()) {
    Serial.println("Setting Level to 0%");
    bleKeyboard.setBatteryLevel(0);
    delay(2000);
    Serial.println("Setting Level to 25%");
    bleKeyboard.setBatteryLevel(25);
    delay(2000);
    Serial.println("Setting Level to 50%");
    bleKeyboard.setBatteryLevel(50);
    delay(2000);
    Serial.println("Setting Level to 75%");
    bleKeyboard.setBatteryLevel(75);
    delay(2000);
    Serial.println("Setting Level to 100%");
    bleKeyboard.setBatteryLevel(100);
    delay(2000);
  } else {
    Serial.println("Not connected yet...");
  }
  Serial.println("Pause of 5 seconds...");
  delay(5000);
}

```

SuGlider, 2022-05-24:
This is in Portuguese... Google Lens can translate it from the image, if necessary. I used it to read the details in your screenshot. :-)

[image]

SuGlider, 2022-05-24:
This is in Portuguese... Google Lens can translate it from the image, if necessary. 
I used it to read the details in your screenshot.  ;-)

[image]

SuGlider, 2022-05-24:
### In summary:

It seems it is necessary to remove the ESP32 Keyboard from the list of paired devices and then scan/add it again in order to display and to update the Battery Indicator. Not sure if this is a ESP32 issue or a Windows issue...

BG2CRW, 2022-05-24:
Yes, I want to say is that need to scan/add it again in order to display and to update the Battery Indicator. Only reset ESP32,the  Battery Indicator won't update.

BG2CRW, 2022-05-24:
BUT, this time, you can try to do the step2: ”//“ line 45 in BLEDevice.cpp, and reconnect to your PC, you can find that it can update.

BG2CRW, 2022-05-24:
I'm sure it is the problem in ESP32. Because my NRF52840 works well.

SuGlider, 2022-05-24:
OK. I'll work in this issue and try to find out what is going on.
Can you post the NFR HID source code link that you have used?

SuGlider, 2022-05-24:
I think that the issue may be in the BLE bonding process. 
It may be missing in the ESP32 protocol layer.
I have enabled some debug messages and there are some request that the ESP32 doesn't know how to answer.
It is possible that Windows demands strong bonding when pairing by exchanging encrypted keys.

If this is the case... it may take a big effort and time.

SuGlider, 2022-05-24:
BLE bonding has to do with reconnection and running BLE as it has never disconnected before in a seamless way.
Everything points to it as it only works on the first Pairing attempt and it fails when reconnecting.

Changing or commenting out Line 45 doesn't seem to the the way to solve it defenetively.
Battery Level Service needs the 2902 characteristics. It just doesn't need the 2904.

BG2CRW, 2022-05-24:
Yes, I know it is not a right way. Now, my project use this feature to recognize if this is a new connect and comment or not line 45 to solve this problem for my product users. 
This is a huge bug for my product users, but I can't take back and change their motherboard.

BG2CRW, 2022-05-26:
this is ble_bas.c in NRF sdk.

/**
 * Copyright (c) 2012 - 2021, Nordic Semiconductor ASA
 *
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without modification,
 * are permitted provided that the following conditions are met:
 *
 * 1. Redistributions of source code must retain the above copyright notice, this
 *    list of conditions and the following disclaimer.
 *
 * 2. Redistributions in binary form, except as embedded into a Nordic
 *    Semiconductor ASA integrated circuit in a product or a software update for
 *    such product, must reproduce the above copyright notice, this list of
 *    conditions and the following disclaimer in the documentation and/or other
 *    materials provided with the distribution.
 *
 * 3. Neither the name of Nordic Semiconductor ASA nor the names of its
 *    contributors may be used to endorse or promote products derived from this
 *    software without specific prior written permission.
 *
 * 4. This software, with or without modification, must only be used with a
 *    Nordic Semiconductor ASA integrated circuit.
 *
 * 5. Any software provided in binary form under this license must not be reverse
 *    engineered, decompiled, modified and/or disassembled.
 *
 * THIS SOFTWARE IS PROVIDED BY NORDIC SEMICONDUCTOR ASA "AS IS" AND ANY EXPRESS
 * OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES
 * OF MERCHANTABILITY, NONINFRINGEMENT, AND FITNESS FOR A PARTICULAR PURPOSE ARE
 * DISCLAIMED. IN NO EVENT SHALL NORDIC SEMICONDUCTOR ASA OR CONTRIBUTORS BE
 * LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
 * CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE
 * GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
 * HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
 * LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT
 * OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 *
 */
/* Attention!
 * To maintain compliance with Nordic Semiconductor ASA's Bluetooth profile
 * qualification listings, this section of source code must not be modified.
 */
#include "sdk_common.h"
#if NRF_MODULE_ENABLED(BLE_BAS)
#include "ble_bas.h"
#include <string.h>
#include "ble_srv_common.h"
#include "ble_conn_state.h"

#define NRF_LOG_MODULE_NAME ble_bas
#if BLE_BAS_CONFIG_LOG_ENABLED
#define NRF_LOG_LEVEL       BLE_BAS_CONFIG_LOG_LEVEL
#define NRF_LOG_INFO_COLOR  BLE_BAS_CONFIG_INFO_COLOR
#define NRF_LOG_DEBUG_COLOR BLE_BAS_CONFIG_DEBUG_COLOR
#else // BLE_BAS_CONFIG_LOG_ENABLED
#define NRF_LOG_LEVEL       0
#endif // BLE_BAS_CONFIG_LOG_ENABLED
#include "nrf_log.h"
NRF_LOG_MODULE_REGISTER();

#define INVALID_BATTERY_LEVEL 255

/**@brief Function for handling the Write event.
 *
 * @param[in]   p_bas       Battery Service structure.
 * @param[in]   p_ble_evt   Event received from the BLE stack.
 */
[... 263 lines cut ...]

    ret_code_t err_code;

    if (p_bas->is_notification_supported)
    {
        ble_gatts_hvx_params_t hvx_params;
        uint16_t               len = sizeof(uint8_t);

        memset(&hvx_params, 0, sizeof(hvx_params));

        hvx_params.handle = p_bas->battery_level_handles.value_handle;
        hvx_params.type   = BLE_GATT_HVX_NOTIFICATION;
        hvx_params.offset = 0;
        hvx_params.p_len  = &len;
        hvx_params.p_data = &p_bas->battery_level_last;

        err_code = battery_notification_send(&hvx_params, conn_handle);

        return err_code;
    }

    return NRF_ERROR_INVALID_STATE;
}

#endif // NRF_MODULE_ENABLED(BLE_BAS)

BG2CRW, 2022-05-26:
BLE_UUID_REPORT_REF_DESCR      is 0x2908

SuGlider, 2022-06-14:
@BG2CRW,  0x2908 is related to HID only, not to Battery Level. 
Thanks for sending all this information!

It is now fixed by setting the notification to 1 by default on the Battery Level Service creating in the ESP32 BLE Arduino Library.
Check #6864 for details.

BG2CRW, 2022-06-14:
Thanks a lot!

PR #6864 (2022-06-16): fixes BLE HID Battery Level Indicator
## Description of Change
BLE HID uses GATT Service to send Battery Level Indication.
It works fine on first time pairing and bonding, but on reconnect it doesn't update the Battery Level characteristic anymore. 
It is fixed by setting the Notification to be ON, by default at the Service Creation. 

## Tests scenarios
Tested with ESP32 using Bluedroid and NimBLE.

## Related links
Fixes #6708