espressif/arduino-esp32 #6709: Using ADC causes boot loop

w4ilun, 2022-05-07:
### Board

ESP32-PICO-D4

### Device Description

Watchy

### Hardware Configuration

ADC on GPIO 34

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

926100

### Description

Battery connected to voltage divider to ADC (GPIO 34) causes the ESP32 to be stuck in a boot loop. When no battery is connected it is fine.

This issue occurs in 2.0.3, when downgraded to 2.0.2 there are no issues.

### Sketch

```cpp
int value = 0;

void setup() {
  Serial.begin(115200);
}

void loop() {

  value = analogRead(34);

  Serial.print("value = ");
  Serial.println(value);

  delay(1000);
}
```

### Debug Message

```plain
NA
```

### Other Steps to Reproduce

Board: ESP32 Dev Module

Issue occurs in 2.0.3, not in 2.0.2

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-05-07:
@P-R-O-C-H-Y, PTAL. Thanks!

P-R-O-C-H-Y, 2022-05-09:
Hi @w4ilun,

I'm unable to reproduce this. Don't have any battery with divider. When I connect to GPIO 34 (3v3 or GND) it always boot correctly and the analogRead is working.

P-R-O-C-H-Y, 2022-05-20:
Closing due to no response. Fell free to reopen if needed :)

closed 2022-05-20 by P-R-O-C-H-Y