espressif/arduino-esp32 #6710: ESP32-C3 Serial.available   BIG PROBLEM

0hotpotman0, 2022-05-07:
### Board

NanoESP32-C3

### Device Description

NanoESP32-C3,  the chip is ESP32C3

### Hardware Configuration

use a ComTool to connect with the board.

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

WINDOWS 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

Hi   

I try this code using Serial to read data from other dev, but the received data is wrong. could someone help?
for example, I sent 1, it received 49, etc.   I attach the code as below:

```
int incomedate = 0;
void setup() {

  Serial.begin(115200); 

  Serial.println("Hello world.");// "Hello world."
}

void loop() {

  if (Serial.available() > 0)
  {
    incomedate = Serial.read();

    Serial.println(incomedate);

  }
}
``` 

I using the NanoESP32-C3 board.       

### Sketch

```cpp
int incomedate = 0;
void setup() {

  Serial.begin(115200); 

  Serial.println("Hello world.");// "Hello world."
}

void loop() {

  if (Serial.available() > 0)
  {
    incomedate = Serial.read();

    Serial.println(incomedate);

  }
}
```
```

### Debug Message

```plain
I try this code using Serial to read data from other dev, but the received data is wrong. could someone help?
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-05-07:
There is probably no driver for the USB serial chip APM32F103 installed.
Details how to use https://github.com/wuxx/nanoESP32-C3#esplink

SuGlider, 2022-05-07:
@Jason2866 - Nice catch! Indeed there is a APM32F103 in the board... That's quite different. 
I guess this APM chip needs the right firmware to work as UART-USB bridge.

[image]

SuGlider, 2022-05-07:
@0hotpotman0 - I propose an experiment to confirm that the issue is actually in the APM32 chip.

Using an external UART-USB converter board (such as FTDI, CH340, CP2102 etc), connect it to the +3.3V, GND, RX/TX pins of the NanoESP32-C3 board and then the converter to the PC/Linux.

I'm almost sure that your sketch will work fine.

0hotpotman0, 2022-05-09:
> 

I don't think that is APM32 chip problem,   Coz I did in ESP32C3 M5 C3U board as well. Some problem.    you could try on any ESP32 C3 board.

me-no-dev, 2022-05-09:
how is the data wrong? Your sketch printlanes every byte it receives back. If you expect the same data to come out, then maybe use `Serial.write(Serial.read())` instead

0hotpotman0, 2022-05-09:
I did it. same result.  the data is wrong.

SuGlider, 2022-05-09:
OK... I got it...

Here is the output of your sketch when seding "ESP32-C3" to the Serial Monitor:

``` 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: a9753a4fc647c6545c1b919ef08db429130a48592727edca270f1e5a3da0d0a9
Expected: 3bf6ef2cf3b9eefcd4b3c70cc5d1ce5138292d101a5cb1d5db6fbebf081b0a19
Attempting to boot anyway...
entry 0x403ce000
Hello world.
69
83
80
51
50
45
67
51
13
10
```

>I try this code using Serial to read data from other dev, but the received data is wrong. could someone help?
for example, I sent 1, it received 49, etc. I attach the code as below:

ASCII of '1' is 49...

SuGlider, 2022-05-09:
Try this:

``` cpp
char incomedate = 0;
void setup() {

  Serial.begin(115200); 

  Serial.println("Hello world.");// "Hello world."
}

void loop() {

  if (Serial.available() > 0)
  {
    incomedate = Serial.read();

    Serial.write(incomedate);

  }
}
```

This is the output with this new sketch when sending "ESP32-C3" to the Serial Monitor:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: a9753a4fc647c6545c1b919ef08db429130a48592727edca270f1e5a3da0d0a9
Expected: 3bf6ef2cf3b9eefcd4b3c70cc5d1ce5138292d101a5cb1d5db6fbebf081b0a19
Attempting to boot anyway...
entry 0x403ce000
Hello world.
ESP32-C3

```

0hotpotman0, 2022-05-10:
it is quite weird, in normal Arduino, ''Serial.read()'' is read the data function, after receive data then we should use "Serial.print()" or "Serial.println()" to print the data on the monitor, "Serial.write()" function should from ESP32C3 sent data to other dev.

0hotpotman0, 2022-05-10:
> Try this:
> 
> ```c++
> char incomedate = 0;
> void setup() {
> 
>   Serial.begin(115200); 
> 
>   Serial.println("Hello world.");// "Hello world."
> }
> 
> void loop() {
> 
>   if (Serial.available() > 0)
>   {
>     incomedate = Serial.read();
> 
>     Serial.write(incomedate);
> 
>   }
> }
> ```
> 
> This is the output with this new sketch when sending "ESP32-C3" to the Serial Monitor:
> 
> ```
> ESP-ROM:esp32c3-api1-20210207
> Build:Feb  7 2021
> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fcd6100,len:0x438
> load:0x403ce000,len:0x90c
> load:0x403d0000,len:0x2358
> SHA-256 comparison failed:
> Calculated: a9753a4fc647c6545c1b919ef08db429130a48592727edca270f1e5a3da0d0a9
> Expected: 3bf6ef2cf3b9eefcd4b3c70cc5d1ce5138292d101a5cb1d5db6fbebf081b0a19
> Attempting to boot anyway...
> entry 0x403ce000
> Hello world.
> ESP32-C3
> ```

your code is work anyway.

0hotpotman0, 2022-05-10:
I think "Serial.read()" function only gets the ASCII data, it does not transfer to normal value, and when I print any value, it will add "\n" ASCII in the end automatically.

SuGlider, 2022-05-10:
> I think "Serial.read()" function only gets the ASCII data, it does not transfer to normal value, and when I print any value, it will add "\n" ASCII in the end automatically.        

`Serial.read()` reads a single byte from UART.

Arduino Serial Monitor is the one that converts whatever the user types into ASCII.

`Serial.print()` has different behaviors, depending on its arguments. It also depends on the type of the argument passed to it.

SuGlider, 2022-05-17:
@0hotpotman0 - Can we close this issue?

SuGlider, 2022-05-22:
Closing it for now.
Please feel free to open it again if necessary.

closed 2022-05-22 by SuGlider