espressif/arduino-esp32 #6713: Enable log in LWIPOPTS

podaen, 2022-05-08:
### Board

ESP32 devkit v4

### Device Description

no

### Hardware Configuration

no

### Version

latest development Release Candidate (RC-X)

### IDE Name

1.8.19

### Operating System

Windows 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm looking how to enable the LWIP debug log in lwipopts.h. 

When I config for example #define CONFIG_LWIP_ETHARP_DEBUG I don't see anything shown up in my monitor. 

How can I enable it? 

### Sketch

```cpp
no
```

### Debug Message

```plain
no
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-05-08:
All the CONFIG_* values are compiled into the esp-idf library archives (.a files) that are included in arduino-esp32.  To change them, you need to compile your own libraries with [lib-builder](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html).

podaen, 2022-05-09:
Can it be done with visual studio instead of visual code?

SuGlider, 2022-05-09:
As @lbernstone already said, ESP32 Arduino uses precompiled IDF Libraries - which includes LWIP. 
The header files and its setup (CONFIG) are fixed for an specific Arduino Core Version. 
There is no way to change them in Arduino and it won't be created/added.

In order to create your own version of Libraries and Settings for LWIP, it is necessary to build the libraries using Lib Builder.
Lib builder is a set of tools found at https://github.com/espressif/esp32-arduino-lib-builder

You can read the main script in order to check the options:
https://github.com/espressif/esp32-arduino-lib-builder/blob/master/build.sh#L19-L31

The LWIP CONFIG can be modified using IDF Menuconfig. 
For example for ESP32, first run menuconfig and then build the libraries for it:
`$> ./build.sh -t esp32 -b menuconfig`
`$> ./build.sh -t esp32`

podaen, 2022-05-09:
Still not understand it how to use it on windows. Download it and use it as a library to setup the LWIP libary to be used in an Arduino IDE?

SuGlider, 2022-05-09:
Humm... it is better to use Linux to run LibBuilder.
I have never tried it on Windows. Maybe using Ubuntu WSL ?!
 
Anyway, it is a linux shell script. 
Nothing that you can run on a VS Code or Arduino IDE.

podaen, 2022-05-09:
A while a go, I did try to install ubuntu on my desktop... but it failed, maybe because my primary ssd's is not big enough... Don't know it anymore, long time ago. 

Now I have visual code installed, but don't know to take advantage concerning this topic. 

I still have a note book to experimented with it.

SuGlider, 2022-05-09:
Visual Code for Windows would only help on editing the sdkonfig files, if it is desired. But it doen't help on your quest.

Even with Linux, VSCode won't build the libraries used in Arduino Core, like as PlatformIO does for an Arduino Sketch.
You will need to use Linux and got to the Terminal and run the script manually.

podaen, 2022-05-09:
Debian seem intresting... for my windows desktop I could program in visual code and use espressif idf... Will be able to use arduino libraries?

podaen, 2022-05-09:
Debian seem interesting for me, but I don't think I have engough room anymore on my windowsdesktop. That would maybe an option for mac. I'm confused. What I see in VisualStudio is...

Platform IO -> platform: Arduino or IDF

If I understand it right, IDF is an linux kenel based. So that would work to addapt the LWIP but I will not be able to use Arduino library's or is [this ](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html)documentation about that.

mrengineer7777, 2022-05-10:
To modify LWIP you have three options:
1) Use [Arduino lib-builder](https://github.com/espressif/esp32-arduino-lib-builder) to compile your own binaries, then replace your arduino-esp32 folder.  Note you'll need a Linux box to compile the library.  IDF is not linux based, but the builder scripts use Linux commands to compile.  Haven't tried this as I don't have a Linux box, and didn't want to mess with my Mac.

2) Base your project on IDF.  You can create sub-folders and bring in specific arduino-esp32 libraries.  Note these libraries are highly interlinked so you'll need to bring in about half of them and edit slightly to get them to compile.  I tried this path.  Had some success but wasted a couple months.  Eventually switched back to Arduino target platform (VS Code with PIO).

3) Use [Arduino as ESP-IDF component](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html).  This had multiple conflicting defines that caused compile to fail.  I was never able to resolve the issues.  Some fixes were added recently.  Haven't tried again.

Compiling under IDF is also extremely slow as it compiles EVERY file.  The Arduino target platform has the advantage of the IDF binaries being pre-compiled.

SuGlider, 2022-05-10:
> Platform IO -> SDK: Arduino or IDF Espressif IDF -> SDK IDF
Yes, but actually it is:  Arduino Core 2.0.0+ -> IDF 4.4+ -> SDK IDF
In the IDE world: PlatformIO -> VSCode

PIO is just a skin for VSCode with some rules for building Arduino Sketches using ESP Arduino Cores.

> If I understand it right, IDF platform is an linux kenel based. 
No, IDF is a IoT Development Framework that builds ESP32 SoCs firmwares, written in C.
It's possible to use Tool Chains for Windows, Linux and MacOS.

>So that would work to addapt the LWIP but I will not be able to use Arduino library's... only by using the [arduino component](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html) on the IDF platform in Espressif IDF or in PlatformIO. Correct? And could I complile other arduino libraries as wel?

LWIP is an open TCP/IP stack in C. It is used by IDF to provide Networking APIs.
Arduino uses IDF and its LWIP Component, but Arduino also wraps IDF/LWIP API with its own API Layer.
You can build Arduino firmware using IDEs (such as Arduino IDE, PlatformIO IDE or even other IDE). But you can also just create your own Makefile or CMake.txt to build it. There is no relationship of IDF to the IDE/Build Method.
 
> And than there are languages. I want to stay close to C++ or C and avoid as many as possible the scripting thing. I see they IDF and PlatformIO run both on C and C++, so that's not a problem.

IDF is C based. 
Arduino is C++ based. File.ino is actually a C++ program (like File.cpp) and compiled with gcc.
Arduino Core is a C++ layer on top of IDF C API.

podaen, 2022-05-11:
Thanks for the wel explaned explanation. Now I understand. For me the best think on a linux computer adapted the sdk but the when I m installing Debian to the mac, I am afread that I will loos my keyboard functionality and I don't have complete acces to it so I am a bit locked whit that. I'm looking to install ubuntu to my windows desktop as a dual boot.

me-no-dev, 2022-05-11:
@podaen scripts work on Mac just fine. AMF I use Mac to develop them and CI runs them on Linux. Could be a problem only with Windows, since bash is not native to it (but can be installed in various ways)