espressif/arduino-esp32 #6714: temperatureRead() linked error. esp32s3

xiexingfeng, 2022-05-08:
### Board

esp32 s3

### Device Description

esp32 s3

### Hardware Configuration

无关系

### Version

latest master (checkout manually)

### IDE Name

platformIO

### Operating System

all

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

Linking .pio/build/esp32s3/firmware.elf
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/libFrameworkArduino.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x4): undefined reference to `temp_sensor_set_config'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/libFrameworkArduino.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x8): undefined reference to `temp_sensor_start'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/libFrameworkArduino.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0xc): undefined reference to `temp_sensor_read_celsius'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/libFrameworkArduino.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x10): undefined reference to `temp_sensor_stop'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/libFrameworkArduino.a(esp32-hal-misc.c.o): in function `temperatureRead':
/Users/jiayingpeng/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:71: undefined reference to `temp_sensor_set_config'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/jiayingpeng/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:72: undefined reference to `temp_sensor_start'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/jiayingpeng/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:73: undefined reference to `temp_sensor_read_celsius'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/jiayingpeng/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:74: undefined reference to `temp_sensor_stop'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32s3/firmware.elf] Error 1

### Sketch

```cpp
"CPU Temperature: "
temperatureRead()
```

### Debug Message

```plain
Linking .pio/build/esp32s3/firmware.elf
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/libFrameworkArduino.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x4): undefined reference to `temp_sensor_set_config'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/libFrameworkArduino.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x8): undefined reference to `temp_sensor_start'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/libFrameworkArduino.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0xc): undefined reference to `temp_sensor_read_celsius'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/libFrameworkArduino.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x10): undefined reference to `temp_sensor_stop'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32s3/libFrameworkArduino.a(esp32-hal-misc.c.o): in function `temperatureRead':
/Users/jiayingpeng/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:71: undefined reference to `temp_sensor_set_config'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/jiayingpeng/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:72: undefined reference to `temp_sensor_start'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/jiayingpeng/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:73: undefined reference to `temp_sensor_read_celsius'
/Users/jiayingpeng/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/jiayingpeng/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:74: undefined reference to `temp_sensor_stop'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32s3/firmware.elf] Error 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-05-08:
See https://github.com/espressif/arduino-esp32/discussions/6197#discussioncomment-2691060

SuGlider, 2022-05-09:
@xiexingfeng - I think that the temperature reading function is not supported on the Arduino Core at all.
@VojtechBartoska - Can you confirm it?

Jason2866, 2022-05-09:
@SuGlider it does work for esp32, s2 and c3.

SuGlider, 2022-05-09:
@xiexingfeng 
from https://github.com/espressif/arduino-esp32/discussions/6197#discussioncomment-2691060

```
Hello,

S3 temperature support will be fully supported in IDF 5.0, it's part of driver refactoring and there are breaking changes.

Workaround solution for temp sensor for S3 will be backported to IDF 4.4 but it's not ready yet.
```

So, it sounds like it won't work for the S3 for a while.

VojtechBartoska, 2022-05-09:
Not supported just on S3 now, will be added later.

SuGlider, 2022-05-22:
Closing it for now. 
Please feel free to open it again if necessary.

closed 2022-05-22 by SuGlider

beachmiles, 2022-08-10:
Working on esp32-c3 with legacy temp sensor driver

```
#include "driver/temp_sensor.h"    //legacy esp32 temp sensor driver. https://github.com/espressif/esp-idf/blob/master/components/driver/test_apps/legacy_rtc_temp_driver/main/test_rtc_temp_driver.c

void initTempSensor(){
    temp_sensor_config_t temp_sensor = TSENS_CONFIG_DEFAULT();
    temp_sensor.dac_offset = TSENS_DAC_L2;  //TSENS_DAC_L2 is default   L4(-40℃ ~ 20℃), L2(-10℃ ~ 80℃) L1(20℃ ~ 100℃) L0(50℃ ~ 125℃)
    temp_sensor_set_config(temp_sensor);
    temp_sensor_start();
}

//Run the initTempSensor once then get temps in main like this
float result = 0;
temp_sensor_read_celsius(&result); 
```

VojtechBartoska, 2022-08-11:
Temperature sensor on ESP32-S3 is already supported in the master branch. Will be part of 2.0.5.