espressif/arduino-esp32 #6715: ESP32 devkit V1 not working!

Maxime66410, 2022-05-09:
### Board

ESP32 DevKitV1

### Device Description

esp32-wroom-32

### Hardware Configuration

Just plugged in pc with USB.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

My ESP 32 Devkit V1 don't work.

It's been two weeks since I touched them, it's new and in very good condition.

When I tried to go back to it it didn't work at all.

So when I turn it on no led, no detection on the pc, no communication port on.

If I keep pressing EN a red led stays on until I stop pressing.

### Sketch

```cpp
Don't have any sketch.
```

### Debug Message

```plain
Don't have any message.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-05-09:
It sounds like a hardware or an USB cable problem. 
You should see, at least, the Red ON indicator LED when the board is connected to the PC using an USB cable.

Can you test it with a different USB Cable?
Do you have another board to test it?

SuGlider, 2022-05-09:
[image]

Maxime66410, 2022-05-09:
Hello, I have already tested with other cables but also on several different connections.
Also the led does not turn on.
It activates only if I press EN and when I release nothing.

SuGlider, 2022-05-09:
> Hello, I have already tested with other cables but also on several different connections. Also the led does not turn on. It activates only if I press EN and when I release nothing.

It sounds like a hardware/board problem to me...
Is your board the same as in the pictures above?

me-no-dev, 2022-05-09:
could also be that USB-UART drivers are not installed

SuGlider, 2022-05-09:
My concern is that @Maxime66410 says that not even the ON LED turns on. It only turns on when EN is pressed.
Is it right @Maxime66410?

SuGlider, 2022-05-09:
@Maxime66410 - The default behavior of the board is to turn on the ON LED as soon as it is connected to the PC, using the USB Cable. Pressing EN shouldn't affect this LED.

As I said, I think this ESP32 board may have some problem. 
In that case there is nothing we can do to help you...

Maxime66410, 2022-05-11:
> @Maxime66410 - The default behavior of the board is to turn on the ON LED as soon as it is connected to the PC, using the USB Cable. Pressing EN shouldn't affect this LED.
> 
> As I said, I think this ESP32 board may have some problem. In that case there is nothing we can do to help you...

In case or as you had asked me I have pictures, I also think it's over for this ESP32..

When I feed it there is nothing.

[image]

But if I press the EN button the red LED stays on and as soon as I stop pressing it goes out.

[image]

SuGlider, 2022-05-11:
Humm... yes, this is not the ESP32 Board expected behavior...
When you press EN, it shall reset and halt the ESP32 MCU. Thus, ti won't work at all.
Final thoughts:

1- Feed the board with external +5V power source directly to the 5V pin. Check if the LED ON turns on. This may indicate that the LDO or some electronic component of the power regulator is burned.
2- If (1) works, try to see the Serial Boot Output in the Serial Monitor of the Arduino IDE or in any other Serial Communication Software.
3- if (2) works, try uploading a sketch, like the Blink example.

Maxime66410, 2022-05-12:
[1- Feed the board with external +5V power source directly to the 5V pin. Check if the LED ON turns on. This may indicate that the LDO or some electronic component of the power regulator is burned. Another action is to feed +3.3v directly to the 3.3 pin, bypassing the LDO...]

I haven't tried yet.
I do it whenever I can.

[2- If (1) works, try to see the Serial Boot Output in the Serial Monitor of the Arduino IDE or in any other Serial Communication Software.]

Already try, there is no communication between ESP32 and the Arduino IDE, no communication port is open.

[3- if (2) works, try uploading a sketch, like the Blink example.]

No need to try if the first step didn't work.

SuGlider, 2022-05-17:
@Maxime66410 - Can we close this issue?

Maxime66410, 2022-05-18:
Yep !

closed 2022-05-18 by Maxime66410