espressif/arduino-esp32 #6723: Uncompleted packages

podaen, 2022-05-10:
### Board

Esp32 dev kit

### Device Description

No

### Hardware Configuration

No

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino Ide

### Operating System

Windows 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

Iq there any solution to the WiFi client for uncomplete receive of packages?

### Sketch

```cpp
No
```

### Debug Message

```plain
Described in previous issues.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-05-10:
I'm guessing you mean incomplete packets on WiFiClient.  What is your application?

If you're interested in going a slightly different direction... :
For us we have found the AsyncWiFiServer to be very helpful.  It includes async versions of WiFi client and server.
Checkout https://github.com/esphome/esphome-webserver
It is based on the fantastic work by me-no-dev: https://github.com/me-no-dev/ESPAsyncWebServer

podaen, 2022-05-10:
Before starting to a server the client need to be fixed.

me-no-dev, 2022-05-10:
based on the info you gave, there is nothing we can do. 2.0.3 stable was released, update to that first, then if you still have the issue, submit a new "Issue" with all information to reproduce the error and we can have a look then.

closed 2022-05-10 by me-no-dev

mrengineer7777, 2022-05-10:
@me-no-dev Please add "2.0.3" as an option in new issue creation.

me-no-dev, 2022-05-10:
> Please add "2.0.3" as an option in new issue creation.

Woops... done :)

podaen, 2022-05-11:
look at my log... If you want to reproduce it. I can't give you a simple shetch right now. Don't you have a simple shetch to collect an audio stream. (not output it, not needed) 

> left: 88
> 2
> _fill: 0
> _size: 1336
> received size: 1336
> New_fill size: 1336
> toRead: 88
> _pos: 0
> 
> res read wificlient: 512
> 3
> res read wificlient: 512
> 3
> res read wificlient: 512
> 3
> toRead: 224
> _pos: 1112
> left: 288
> 2
> _fill: 0
> _size: 1336
> received size: 336
> New_fill size: 336
> toRead: 288
> _pos: 0
> 
> res read wificlient: 512
> 3
> res read wificlient: 48
> 2
> _fill: 0
> _size: 1336
> received size: 1336
> New_fill size: 1336
> res read wificlient: 512
> 3
> res read wificlient: 512
> 3
> toRead: 312
> _pos: 1024
> left: 84
> 2
> _fill: 0
> _size: 1336
> received size: 84
> New_fill size: 84
> toRead: 84
> _pos: 0
> run again left until completed size
> 
> res read wificlient: 396
> 2
> _fill: 0
> _size: 1336
> received size: 1336
> New_fill size: 1336
> res read wificlient: 512
> 3
> res read wificlient: 512
> 3
> toRead: 312
> _pos: 1024
> left: 200
> 2
> _fill: 0
> _size: 1336
> received size: 1336
> New_fill size: 1336
[... 508 lines cut ...]
> 
>        EINTR  The receive was interrupted by delivery of a signal before
>               any data was available; see signal(7).
> 
>        EINVAL Invalid argument passed.
> 
>        ENOMEM Could not allocate memory for recvmsg().
> 
>        ENOTCONN
>               The socket is associated with a connection-oriented
>               protocol and has not been connected (see connect(2) and
>               accept(2)).
> 
>        ENOTSOCK
>               The file descriptor sockfd does not refer to a socket.
> CONFORMING TO         top
>        POSIX.1-2001, POSIX.1-2008, 4.4BSD (these interfaces first
>        appeared in 4.2BSD).
> 
>        POSIX.1 describes only the MSG_OOB, MSG_PEEK, and MSG_WAITALL
>        flags.
> 
> 

the errors applys only to MSG_OOB, MSG_PEEK, and MSG_WAITALL

podaen, 2022-05-11:
I can give you the code where I generate the log with. But the whole thing... you would be buzzy for quit some time. It only happens with stream, what I think with a give volume of data.

podaen, 2022-05-11:
I just had news there is maybe a way to give you an example to test it. I will try it first

podaen, 2022-05-11:
I tried, It's connecting but after that I receive only one byte...

```
#include <WiFi.h>
#include <WiFiClient.h>
#include <HTTPClient.h>
#include <WiFiClientSecure.h>
#include <esp_bt_main.h>//power saving bluetooth
#include <esp_bt.h>//power saving bluetooth

const char* ssid = "ssid";
const char* password = "pass";

// the setup function runs once when you press reset or power the board
void setup() {
	Serial.begin(500000);
	while (!Serial);

	WiFi.mode(WIFI_MODE_STA);
	WiFi.begin(ssid, password);
	while (WiFi.status() != WL_CONNECTED) {
		delay(1000);
		// Serial.println("Connecting to WiFi..");
	}
    Serial.println("Connected");
	WiFi.setSleep(false);

	esp_bluedroid_disable();
	esp_bluedroid_deinit();
	esp_bt_controller_disable();
	esp_bt_controller_deinit();
    Serial.println("end setup");

}

// the loop function runs over and over again until power down or reset
void loop() {
	if (WiFi.status() == WL_CONNECTED) {
		HTTPClient http;
		http.begin("http://22183.live.streamtheworld.com/JOE.mp3");
		Serial.println("[HTTP] begin...\n");

        http.addHeader("Host", "22183.live.streamtheworld.com");
        http.addHeader("Connection", "close");
        http.addHeader("Accept-Encoding", "");
        http.addHeader("Accept", "*/*");
        http.setUserAgent("Mozilla/5.0 (compatible; Googlebot/2.1; +http://www.google.com/bot.html)");
        
        int httpCode = http.GET();
        if (httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            Serial.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if (httpCode == HTTP_CODE_OK) {
                Serial.print("HTTP_CODE_OK: ");
                Serial.println(HTTP_CODE_OK);
                // get length of document (is -1 when Server sends no Content-Length header)
                int len = http.getSize();
                Serial.print("len: ");
                Serial.println(len);

                uint8_t buff[512] = { 0 };
                WiFiClient* stream = http.getStreamPtr();

                while (http.connected() && (len > 0 || len == -1)) {
                    // get available data size
                    size_t size = stream->available();
                    Serial.print("size: ");
                    Serial.println(size);
                    if (size) {
                        int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));
                        //// write it to Serial
                        //Serial.write(buff, c);
                        if (len > 0) {
                            len -= c;
                        }
                    }
                    delay(1);
                }

                Serial.println();
                Serial.print("[HTTP] connection closed or file end.\n");
            }
        }
        else {
            Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
    }

    delay(10000);

}
```

podaen, 2022-05-11:
ooo sorry it was not meant for you guys. I will come back to when it working for this link.

podaen, 2022-05-11:
Ok. I will do