espressif/arduino-esp32 #6725: ESP32 in STA+AP with SNTP

JokerGTS, 2022-05-10:
### Board

ESP32 Dev Module

### Device Description

[image]
(with CH9102X)

### Hardware Configuration

nothing

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2 (rc6) and Arduino IDE 1.8.18

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Can't get time via SNTP using STA + AP mode. The serial port monitor displays the day of the week, hour, and minute. But I get always 4 days of the week, 3 hours and 0 minutes (changes over time). As far as I understand, it is not possible to connect to the NTP server (my tests with other libraries showed this). In STA mode everything works fine. Thanks in advance for your help.

### Sketch

```cpp
#include <WiFi.h>
#include "time.h"
#include "sntp.h"

const char* wifi_network_ssid = "ROUTER";
const char* wifi_network_password = "ROUTER";
//AP network credentials
const char* ssid = "ESP32";
const char* password = "ESPPASS";

struct tm timeinfo;

//Timezone is GMT+3 
#define GMT_sec 3 * 3600
//uncomment to set summer offset
#define daylightOffset_sec 0

// Callback function (get's called when time adjusts via NTP)
void timeavailable(struct timeval *t) {
  Serial.println("Got time adjustment from NTP");
  getLocalTime(&timeinfo);
}

void setup()
{
  Serial.begin(115200);

  const char* ntpServer1 = "pool.ntp.org";
  const char* ntpServer2 = "time.nist.gov";
  // set notification call-back function
  sntp_set_time_sync_notification_cb(timeavailable);
  sntp_servermode_dhcp(1);    // (optional)
  configTime(GMT_sec, daylightOffset_sec, ntpServer1, ntpServer2);

  //connect to WiFi
  WiFi.mode(WIFI_MODE_APSTA);

  /*-------------------ESP32 AP setup-------------------*/
  WiFi.softAP(ssid, password);  //ssid, pass, channel, hideSSID. maxClients
  WiFi.softAPConfig(IPAddress(8, 8, 4, 4), IPAddress(192, 168, 4, 1), IPAddress(255, 255, 255, 0)); //Captive Portal

  /*-------------------ESP32 STA setup-------------------*/  
  Serial.print("Connecting");
  // Set your Static IP address
  IPAddress local_IP(192, 168, 0, 24);
  IPAddress gateway(192, 168, 0, 1);
  IPAddress subnet(255, 255, 255, 0);
  WiFi.config(local_IP, gateway, subnet);
  // Connect to Wi-Fi network with SSID and password
  WiFi.begin(wifi_network_ssid, wifi_network_password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.print(".");
  }
  Serial.println("Done");
}

void loop()
{
  delay(3000);
  getLocalTime(&timeinfo);
  Serial.println(timeinfo.tm_wday); //days since Sunday  0-6
  Serial.println(timeinfo.tm_hour);
  Serial.println(timeinfo.tm_min);
 
}
```

### Debug Message

```plain
Connecting.Done
4
3
0
(a minute later:)
4
3
1
```

### Other Steps to Reproduce

Just switch ESP to STA + AP mode

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-05-10:
You need to provide DNS server as fourth argument to WiFi.config, then it will work. The ESP has no idea how to resolve the NTP server address without DNS.

closed 2022-05-10 by me-no-dev

JokerGTS, 2022-05-11:
> You need to provide DNS server as fourth argument to WiFi.config, then it will work. The ESP has no idea how to resolve the NTP server address without DNS.

Thanks for the quick reply, it really solved my problem!