espressif/arduino-esp32 #6727: Problem with receiving data from Serial to update ESP32 OTA

primus192, 2022-05-11:
### Board

ESP32 Dev Module, custom board

### Device Description

ESP32 Dev Module, custom board

### Hardware Configuration

Quectel 2G modem attached to pin 26 and 27 of the esp32

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Using earlier versions of arduino framework (pre 2.0 update) i was able to receive all bytes from my GSM 2G modem and push it through the Serial1 connected to pins 26 and 27 to Esp32 to update it OTA with using Serial1.setRXBufferSize(1024) before initializing communication. Baud rate is 115200.

Now with newest Arduino 2.0.3 or 2.0.4 or master branch progress of flashing just stops at 85%, modem sent all the data but esp32 missed some during receiving and after some time prints error number 6 which means timeout. Setting different RX buffer sizes before Serial1.begin() did not help.  Maybe setting rx buffer size is broken? Or other things changed which i do not think about? Thanks in advance.

### Sketch

```cpp
Related parts, it is hard for me to paste whole code:
What works in previous frameworks:
fragment of main part:
//////////////////////////////////
        Serial1.setRxBufferSize(1024);
        Serial1.flush();
        btStop();
        if(espUpdate.updateGsm(modem.ftpDownload((String)argv[6].c_str()))) {
            modem.mqttPublish(topic, "Update success");
        } else {
            modem.mqttPublish(topic, "Update fail, try again later");
        }
/////////////////////////////////////
espUpdate part:
////////////////////////////////////

bool ESPUpdate::updateGsm(long filesize) {
    _contentLength = filesize;
    if(filesize < 1) {
        Serial.println("No file or wrong size.");
        return 0;
    }
    log_i("Firmware size: %s bytes", String(_contentLength));
    Update.onProgress([](unsigned int progress, unsigned int total) {
        Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
        });

    while(!Serial1.available()) {
    };

    if (Update.begin(_contentLength)) {
        size_t written = Update.writeStream(Serial1);
        if (written == _contentLength) {
            log_i("Written: %s successfully.", String(written));
        }
        else {
            log_i("Written: %s/%s.", String(written), String(_contentLength));
        }
        if (Update.end()) {
            log_i("Update done.");
            if (Update.isFinished()) {
                Serial.println("Update firmware success.");
                return 1;
                //ESP.restart();
            }
            else {
                _contentLength = 0;
                log_i("Update not finished? Something went wrong!");
                return 0;
            }
        }
        else {
            _contentLength = 0;
            log_e("Error Occurred. Error #: %d", Update.getError());
            return 0;
        }
    }
    else {
        log_e("Something is wrong. OTA Update failed.");
        return 0;
    }
    Serial.println("Update firmware fail. Try again later.");
        return 0;
}

////////////////////////////

What i tried in newer arduino framework which did not help:

 Serial1.end();
        Serial1.setRxBufferSize(1024);
        Serial1.begin(115200,SERIAL_8N1,26,27);
        Serial1.flush();
        btStop();
        if(espUpdate.updateGsm(modem.ftpDownload((String)argv[6].c_str()))) {
            modem.mqttPublish(topic, "Update success");
        } else {
            modem.mqttPublish(topic, "Update fail, try again later");
        }
        device.restart(true);
```

### Debug Message

```plain
2.0.3 and up:
Flashing firmware_1.2.1.bin
[ 52534][I][ESPUpdate.cpp:64] updateGsm(): Firmware size: 1500912 bytes
[ 52841][D][Updater.cpp:133] begin(): OTA Partition: app1
[562038][I][ESPUpdate.cpp:78] updateGsm(): Written: 1282730/1500912.
[562038][E][ESPUpdate.cpp:95] updateGsm(): Error Occurred. Error #: 6

before 2.0.0 update:

Flashing firmware_1.2.1.bin
[I][ESPUpdate.cpp:64] updateGsm(): Firmware size: 1500912 bytes
[D][Updater.cpp:132] begin(): OTA Partition: app1
[I][ESPUpdate.cpp:75] updateGsm(): Written: 1500912 successfully.
[I][ESPUpdate.cpp:81] updateGsm(): Update done.
Update firmware success.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

primus192, 2022-05-11:
I must add, the key point for getting it to work on 1.0.6 release was setting the rx buffer to 1024. Without that the buffer will overflow and the bytes are missing. However setting the buffer in arduino > 2.0.0 do not help

SuGlider, 2022-05-11:
Similar to #6644

SuGlider, 2022-05-11:
What about setting a bigger RXBuffer, such as 4096? Would it make any difference?

SuGlider, 2022-05-11:
Please try this and let me know:

```
  Serial1.setRxBufferSize(4096);
  Serial1.setTimeout(0);
  Serial1.begin(115200,SERIAL_8N1,26,27);
  ...
```

primus192, 2022-05-11:
> ```
> Related parts, it is hard for me to paste whole code:
> What works in previous frameworks:
> fragment of main part:
> //////////////////////////////////
>         Serial1.setRxBufferSize(1024);
>         Serial1.flush();
>         btStop();
>         if(espUpdate.updateGsm(modem.ftpDownload((String)argv[6].c_str()))) {
>             modem.mqttPublish(topic, "Update success");
>         } else {
>             modem.mqttPublish(topic, "Update fail, try again later");
>         }
> /////////////////////////////////////
> ```
> 
> You are sure that `Serial1.setRxBufferSize()` is being executed before `Serial1.begin()`, right?

Yes, while moving to new framework i moved setRxBufferSize(1024) before begin of the serial1 because it was throwing me error that i cant set buffer when serial is already running (not a problem in 1.0.6 tho). Unfortunately setting buffer size to 4096 and even 10000 didnt fix the problem and the progress is not increasing , still about 1282730/1500912 bytes

SuGlider, 2022-05-11:
I don't have a modem to test it, but maybe I can try to emulate it with a Python script.
In order to do it, I would like to make it simulate the amount of data received along the time, including, maby, some data bursts. 

Can you provide me with some sort of output that has something like this:
[time_in_milliseconds], [number_of_bytes_received_from_Quectel 2G modem]

example (time_ms, #bytes):
100, 2345
205, 120
220, 8450
221, 45
327, 9887 

Thanks!

primus192, 2022-05-11:
> I don't have a modem to test it, but maybe I can try to emulate it with a Python script. In order to do it, I would like to make it simulate the amount of data received along the time, including, maybe, some data bursts.
> 
> Can you provide me with some sort of output that has something like this: [time_in_milliseconds], [number_of_bytes_received_from_Quectel 2G modem]
> 
> example (time_ms, #bytes): 100, 2345 205, 120 220, 8450 221, 45 327, 9887
> 
> Thanks!

I will Try to prepare something tomorrow

SuGlider, 2022-05-11:
Just to verify... you say that with Arduino Core 2.0.0, you get the same issue, right.
Meaning that the problem was introduced when moved from version 1.0.6 to the 2.0.0?

primus192, 2022-05-11:
It is working for me in arduino 1.0.6 with setting rx buffer to 1024. Without setting buffer it is not. In arduino 2.0.0 and later it is not working regardless of setting serial buffer size or not

primus192, 2022-05-12:
Did not have time today to measure the timings and bytes. I did check Reading data from the stream directly and check how many bytes did i read. 1500912 which is exactly my binary file size. So the esp32 can handle data from the modem but when i pass Stream directly to the writeStream() function, it looks like esp32 cannot keep up with Reading and writing data to flash at the same time.

mudgalp, 2022-05-13:
> Did not have time today to measure the timings and bytes. I did check Reading data from the stream directly and check how many bytes did i read. 1500912 which is exactly my binary file size. So the [...]

i also think so in the same kind of problem i'm facing in my issue #6644

primus192, 2022-05-13:
> i also think so in the same kind of problem i'm facing in my issue #6644

Yes. Did you Try my solution in arduino 1.0.6?

mudgalp, 2022-05-15:
> Yes. Did you Try my solution in arduino 1.0.6?

Yes, i tried that. didn't work.

SuGlider, 2022-05-15:
I have seen some people that uses OTA from a LTE/GPRS modem in a different way:
First the application copies the firmaware to the SPIFFS and after finishing, it flashes the firmware from this SPIFFS partition...

Not sure if this way would work for you.

SuGlider, 2022-05-17:
Try this code:

https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/132#issuecomment-841683522

primus192, 2022-05-18:
Unfortunately my binary size exceeds 1.5MB so i wont load it to spiffs. I just wonder why it works on 1.0.6 and not on > 2.0.0. I will stick to 1.0.6 for now

mudgalp, 2022-05-20:
> Unfortunately my binary size exceeds 1.5MB so i wont load it to spiffs. I just wonder why it works on 1.0.6 and not on > 2.0.0. I will stick to 1.0.6 for now

i will also face this problem in future because i'm testing the OTA with only the OTA functions not with all the working functions on the device. after adding the complete code it will be about 1.7MB in size.

SuGlider, 2022-06-06:
There is a possible work around that may fix this issue.

Pease try this code/example when starting Serial1 (used to receive data from the Modem with OTA firmware):
```
#include "driver/uart.h"
void setup(){
        Serial1.setRxBufferSize(1024);
        Serial1.begin(115200,SERIAL_8N1,26,27);
        Serial1.setRxTimeout(1);
        uart_set_always_rx_timeout(1, true); // 1 is for Serial1, in this case
        Serial1.flush();
}
```

Please let me know if this solves the issue.

VojtechBartoska, 2022-08-23:
any updates @primus192?

primus192, 2022-08-23:
Hi. I i am still on Old framework. Can not get it working on >2.0

Pragalf, 2022-09-15:
> I must add, the key point for getting it to work on 1.0.6 release was setting the rx buffer to 1024. Without that the buffer will overflow and the bytes are missing. However setting the buffer in arduino > 2.0.0 do not help

Hi,
 I am also having the same issue. Ota update working in ESP version 1.0.6 but not in ESP 2.0.3. Data is received in the modem but didn't receive in ESP. Tried to print the data and it was much less than my defined packet size. Rx buffer is set to match the packet size, but I think it isn't working.

SuGlider, 2022-09-19:
We have done a few updates to UART in the new Arduino Core 2.0.5 version.
There are new functions that may help your OTA applications:

- `void onReceive(OnReceiveCb function, bool onlyOnTimeout = false);`
    `onReceive()` will setup a callback that will be called whenever an UART interruption occurs (UART_INTR_RXFIFO_FULL or 
    UART_INTR_RXFIFO_TOUT). This means that it is possible to set up a callback that will be executed as soon as data is 
    received.
    UART_INTR_RXFIFO_FULL interrupt triggers at UART_FULL_THRESH_DEFAULT bytes received (defined as 120 bytes by 
    default in IDF). This default value of 120 can be changed using `setRxFIFOFull()` with a value from 1 to 127.
    UART_INTR_RXFIFO_TOUT interrupt triggers at UART_TOUT_THRESH_DEFAULT symbols passed without any reception 
    (defined as 2 symbols by default). This can also be changed with `setRxTimeout()`
    
   `onlyOnTimeout` parameter will define how onReceive will behave:
                 `true` -- The callback will only be called when RX Timeout happens. 
                              Whole stream of bytes will be ready for being read on the callback function at once.
                              This option may lead to Rx Overflow depending on the Rx Buffer Size and number of bytes received
                              in the streaming
   **_Default:_** `false` -- The callback will be called when FIFO reaches `RXFIFO_FULL` bytes and also on RX Timeout.
                              The stream of incommig bytes will be "split" into blocks of minimum `RXFIFO_FULL` bytes on each callback.
                              This option avoids any sort of Rx Overflow, but leaves the UART packet reassembling work 
                              to the Application.

- `void onReceiveError(OnReceiveErrorCb function);`
   `onReceiveError()` will be called on error events (see hardwareSerial_error_t) and it will also call the `onReceive()` callback in 
   case some was defined. By this, it is possible to read data using `onReceive()` callback when a BREAK is sent to UART, for 
   instance.

-  `void setRxTimeout(uint8_t symbols_timeout);`
    `setRxTimeout()` sets the timeout after which onReceive callback will be called (after receiving data, it waits for this time of UART rx inactivity to call the callback fnc)
    param symbols_timeout defines a timeout threshold in uart symbol periods. 
    Setting 0 (zero) symbol timeout disables the callback call by timeout.
    Maximum timeout setting is calculacted automatically by IDF. If set above the maximum, it is ignored and an error is printed on Serial0 (check console).
    For example symbols_timeout=1 defines a timeout equal to transmission time of one symbol (~11 bit) on current baudrate. 
    For a baudrate of 9600, SERIAL_8N1 (10 bit symbol) and symbols_timeout = 3, the timeout would be 3 / (9600 / 10) = 3.125 ms
    
-  `void setRxFIFOFull(uint8_t fifoBytes);`
    `setRxFIFOFull()` will set the number of bytes that will trigger UART_INTR_RXFIFO_FULL interrupt and fill up RxRingBuffer
    This affects some functions such as Serial::available() and Serial.read() because, in a UART flow of receiving data, 
    Serial internal RxRingBuffer will be filled only after these number of bytes arrive or a RX Timeout happens.
    This parameter can be set to  a low value, such as 1 in order to receive byte by byte, but it will also consume more 
    CPU time as the ISR will be activates often.

primus192, 2022-09-19:
> Using Arduino Core 2.0.5, please try to setup the Serial used for OTA like
> this:
>
> void setup(){
>         Serial1.setRxBufferSize(1024);
>         Serial1.begin(115200,SERIAL_8N1,26,27);
>         Serial1.setRxTimeout(1);
>         Serial1.setRxFIFOFull(1);
>         Serial1.flush();
> }
>
> This will force that whenever a single byte arrives to the Serial Port, it
> will be made immediately available for Serial.read().

I will Try that tomorrow and will report back.

Pragalf, 2022-09-29:
> Using Arduino Core 2.0.5, please try to setup the Serial used for OTA like this:
> 
> ```c++
> void setup(){
>         Serial1.setRxBufferSize(1024);
>         Serial1.begin(115200,SERIAL_8N1,26,27);
>         Serial1.setRxTimeout(1);
>         Serial1.setRxFIFOFull(1);
>         Serial1.flush();
> }
> ```
> 
> This will force that whenever a single byte arrives to the Serial Port, it will be made immediately available for `Serial.read()`.

Let me check it and comeback.

primus192, 2022-11-10:
Unfortunately it does not work. Updating still hangs at 97% while on arduino 1.0.6 it works like a charm.

SuGlider, 2022-12-04:
@primus192  @mudgalp 

Please try using lastest master branch from Arduino ESP32 Github to build you application.
There is a new PR that may fix the issue: https://github.com/espressif/arduino-esp32/pull/7525

This PR fixes `Stream::readBytes()` to make it faster using IDF in the new Core 2.0.x

I think that it may solve the timeout issue.

Please use this setup:
```
void setup(){
        Serial.setRxBufferSize(2048);
        Serial.begin(115200,SERIAL_8N1,26,27);  // RxPin = 26 & TxPin = 27
        Serial.setRxTimeout(1);
        Serial.setRxFIFOFull(120);
        Serial.flush();
}
```

primus192, 2022-12-04:
Thank you. I will  check and report back tomorrow.

primus192, 2022-12-07:
I checked that, unfortunately, no success.

primus192, 2022-12-13:
Not completed by any means ;)

Dave-Osborne, 2023-06-26:
I've just updated from 1.0.6 to 2.0.9 and am facing the same issue. Is there an ETA on resolving this or a workaround?

SuGlider, 2023-06-26:
> I've just updated from 1.0.6 to 2.0.9 and am facing the same issue. Is there an ETA on resolving this or a workaround?
> 
> UPDATE: After setting a cb for HardwareSerial::onReceiveError I'm now seeing additional logging
> 
> _[W] UART1 FIFO Overflow. Consider adding Hardware Flow Control to your Application._
> 
> Adding hardware flow control isn't really an option as our product has been working fine up until this software upgrade.

@Dave-Osborne - FIFO Overflow means that the internal ESP32 UART1 FIFO got full before the data could be read by the Arduino Sketch.
One way to overcome it would be to add a bigger RX Buffer by executing `Serial1.setRxBufferSize(4096);`, for instance, prior to executing `Serial1.begin(baudrate);`. What is the baud rate used in the project?

Dave-Osborne, 2023-06-26:
@SuGlider - Hi, the rx buffer size is already set to 4096 before the serial begin (I've also tried setting it larger).

Baud rate is 115200

SuGlider, 2023-06-26:
What is the sketch used for OTA?
Is it connected to a LTE Modem or something like that?

Dave-Osborne, 2023-06-27:
It is part of OTA (via LTE-M Modem), I download smaller gzipped sections of the firmware and store on a LittleFS partition and then later stitch them together and perform an OTA update.

Normally I use 300k sections, but have dropped to about 50k whilst looking into this issue. 

I've changed the code to only read the number of bytes available during the loop, as you can see below using this method 109 bytes in consistently read which is nowhere near the 4096 buffer limit, it looks like perhaps the Serial1.setRxBufferSize(4096); call isn't actually increasing the FIFO buffer?

```
[2649] FileStream: Estimated completion: 7 %, bytes written: 3597
[ 79172][I][commsGsm.cpp:678] doGet(): Read 109, remaining 47605 of 51202
[ 79172][I][commsGsm.cpp:678] doGet(): Read 109, remaining 47496 of 51202
[ 79174][I][commsGsm.cpp:678] doGet(): Read 109, remaining 47387 of 51202
[ 79180][I][commsGsm.cpp:678] doGet(): Read 109, remaining 47278 of 51202
[ 79187][I][commsGsm.cpp:678] doGet(): Read 109, remaining 47169 of 51202
[ 79238][W][HardwareSerial.cpp:298] _uartEventTask(): UART1 FIFO Overflow. Consider adding Hardware Flow Control to your Application.
[2765] FileStream: Estimated completion: 8 %, bytes written: 4142
 79277][I][commsGsm.cpp:678] doGet(): Read 109, remaining 47060 of 51202
[ 79277][I][commsGsm.cpp:678] doGet(): Read 109, remaining 46951 of 51202
[ 79279][I][commsGsm.cpp:678] doGet(): Read 109, remaining 46842 of 51202
[ 79285][I][commsGsm.cpp:678] doGet(): Read 109, remaining 46733 of 51202
[ 79292][I][commsGsm.cpp:678] doGet(): Read 109, remaining 46624 of 51202
```

SuGlider, 2023-06-27:
`Stream::readBytes()` may be related to the issue... 

Please try using `Serial1.read(uint8_t *buffer, size_t size)` instead. 
Let me know if it changes anything about the issue.

Dave-Osborne, 2023-06-27:
Thanks for @SuGlider, I'm still using readBytes() but have just found that if I don't write my buffer to LittleFs there is no issue reading serial data, so perhaps there is an underlying issue reading serial and then writing to flash?

As I'm downloading the firmware in sections I'm going to transfer to PSRam, check by CRC and then append to a larger file on LittleFs before performing the update.

primus192, 2023-06-27:
As far as I remember I checked regular  read before and it did not fix my issue either

Dave-Osborne, 2023-06-27:
I can confirm that if I no longer write to flash directly I no longer get FIFO Overflows.

As a quick workaround I'm now writing to a PS RAM filesystem first https://github.com/tobozo/ESP32-PsRamFS

I download in 300k sections and then append each to a LittleFS file, but if you have enough PSRam you could probably update directly from the PS RAM filesystem?

I'll perhaps refactor to use a PSRam stream instead at some point, but this workaround is working for me at the moment.

primus192, 2023-06-28:
> I can confirm that if I no longer write to flash directly I no longer get FIFO Overflows.
>
> As a quick workaround I'm now writing to a PS RAM filesystem first https://github.com/tobozo/ESP32-PsRamFS [...]

Unfortunately PSRAM is not the option for me :(

SuGlider, 2023-06-29:
@primus192 - Can you please try this:

``` cpp
        Serial1.setRxBufferSize(4096);
        Serial1.begin(115200,SERIAL_8N1,26,27);
        Serial1.setTimeout(0);   // ====> This may affect the process of time out and help with OTA
        Serial1.flush();
```

primus192, 2023-06-29:
> @primus192 - Can you please try this:
> 
> ```c++
>         Serial1.setRxBufferSize(4096);
>         Serial1.begin(115200,SERIAL_8N1,26,27);
>         Serial1.setTimeout(0);   // ====> This may affect the process of time out and help with OTA
>         Serial1.flush();
> ```

Tomorrow i will try that but i think i have Tried that before with No success

techNiv1, 2024-02-28:
Hi @primus192 did you resolved this issue?
I am also facing the same issue. If you have resolved it, please give us some pointers.
Thank you in adv.

TheMasterofBlubb, 2024-06-17:
@primus192 Just a small idea:
I guess you were writing to flash when doing the OTA, which would mean that while you write, you can not read from flash. So consequently you can not run code at that time as it would be read from flash.

Did you try to put the functions reading Serial and writing Flash into IRAM? ( using `IRAM_ATTR` you can tell the toolchain to do that)

Putting the functions into IRAM would make them independent of the flash and allow for execution while flash is written.

primus192, 2024-06-17:
Well that is clever idea, i will try to do that. Anyway i am using hardware
flow control now which fixed issue for me  and improved stability of my
communication with modem :)

Parsaabasi, 2025-01-09:
Hello, 

This issue seems resolved. Therefore I close it. 

In case it occurred again, please reopen it. 

Thanks.