espressif/arduino-esp32 #6733: SOLUTION to stream start to falter /  failed / breaks / doesn't return not false

podaen, 2022-05-12:
### Board

ESP32 devkit

### Device Description

No

### Hardware Configuration

No

### Version

v2.0.3

### IDE Name

1.8.19

### Operating System

windows 10

### Flash frequency

40 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

When you do a read and when there is no RXbuffer avalaible the stream ends up in strange behavior. I have seen this behavior in many library that uses stream. And I have found the solution for this by checking if there is buffer availible before reading. 

Just use this shetch it will get clear to you. I don't have the courage anymore to find the exact reason to why the stream stops. But I have still some advise to the develops to improve there code.

In the     int read(uint8_t * dst, size_t len)   int the WiFiClient you need to put the fillbuffer on top, because if one of the conditions in the if is completed in an OR the next one condition will not exicude with the C++11 compiler (The fillbuffer). 

```
int frb = fillBuffer();
if(!dst || !len || ( !frb && _pos == _fill)){
```

Than you need to do is too.

`if(len <= a || ((len - a) <= (_size - _fill) && frb >= (len - a))){`

### Sketch

```cpp
#include <WiFi.h>
#include <WiFiClient.h>
#include <HTTPClient.h>
#include <WiFiClientSecure.h>
#include <esp_bt_main.h>//power saving bluetooth
#include <esp_bt.h>//power saving bluetooth

const char* ssid = "SSID";
const char* password = "PASS";
#define BUFFER_SIZE 512

// the setup function runs once when you press reset or power the board
void setup() {
	Serial.begin(500000);
	while (!Serial);

	WiFi.mode(WIFI_MODE_STA);
	WiFi.begin(ssid, password);
	while (WiFi.status() != WL_CONNECTED) {
		delay(1000);
		// Serial.println("Connecting to WiFi..");
	}
    Serial.println("Connected");
	WiFi.setSleep(false);

	esp_bluedroid_disable();
	esp_bluedroid_deinit();
	esp_bt_controller_disable();
	esp_bt_controller_deinit();
    Serial.println("end setup");

}

// the loop function runs over and over again until power down or reset
void loop() {
	if (WiFi.status() == WL_CONNECTED) {
        HTTPClient http;
        http.addHeader("Host", "22183.live.streamtheworld.com");
        http.addHeader("Connection", "close");
        http.addHeader("Accept-Encoding", "");
        http.addHeader("Accept", "*/*");

        http.setTimeout(30000);
        int httpCode = http.GET();
        if (httpCode > 0) {
            Serial.printf("[HTTP] GET... code: %d\n", httpCode);

            if (httpCode == HTTP_CODE_OK) {
                uint8_t buffer[512] = { 0 };
                int buffer_size = 512;

                WiFiClient* stream = http.getStreamPtr();
                delay(5000);
                size_t result = stream->available();
                //int result = stream->available();
                if (result > 0) {
                    while (http.connected()) {
                        result = stream->available();
                        if (result > 2 * buffer_size) {
                            size_t size = min(result, static_cast<size_t>(buffer_size));
                            size_t read = stream->read((uint8_t*)buffer, size);
                            if (read > 0) {
                                Serial.print("read: ");
                                Serial.println(read);
                                Serial.println("");
                            }
                        }
                        else
                        {

                        }

                    }
                }

                Serial.println();
                Serial.print("[HTTP] connection closed or file end.\n");
            }
        }
        else {
            Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
    }
    delay(10000);
}
```

### Debug Message

```plain
read is allways full...

received size: 1336
New_fill size: 1336

read: 512

read: 512

received size: 1087
New_fill size: 1087

read: 512

received size: 249
New_fill size: 1336
read: 512

read: 512

received size: 1336
New_fill size: 1336

read: 512

read: 512

read: 512
```

### Other Steps to Reproduce

Use the shetch, Once you don that delete this line if (result > 2 * buffer_size) {

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-05-12:
Possible signed/unsigned issues:
`size_t result = stream->available();`  //WiFiClient::available() returns int, but internally converts size_t (unsigned int) to int.
`size_t read = stream->read();`   //WiFiClient.read(() returns an int, which is -1 on failure.

podaen, 2022-05-12:
After some time the stream still failed, when it doesn't receive the whole package or the buffer received is not complety emtry.  What would it be?

> readed bytes: 512
> received size: 259
> New_fill size: 1336
> readed bytes: 512
> received size: 1336
> New_fill size: 1336

mrengineer7777, 2022-05-12:
> After some time the stream still failed, when it doesn't receive the whole package or the buffer received is not complety emtry. What would it be?

No idea, I don't see those debug statements in your example.
Did you change `size_t read = stream->read(` to `int read =`?
Keep in mind I'm a fellow developer, not the maintainer of the project.  Without an example project in a Github repo, I won't be able to run your code.

> When the received size is 1336 it stops sometimes but it continues.
That is oddly specific

> You beter should try it.... Just add this in the client.cpp fillbuffer
> 
> ```
>             Serial.print("received size: ");
>             Serial.println(res);
>             Serial.print("New_fill size: ");
>             Serial.println(_fill);
> ```

I have no idea what this is.
 
> I know it have to be logging, but I searching to get it right. so this is what i do. It can take some time before it start to wigle. And will keep data out, but what data... Data that maybe has a pointer in that doesn't belown there... a frame that is cuttet in half. who knows... But I am sure the delay of receiving is causing this issue. It can be realy big. And you could say it is the stream, but I serrously doubd it, because on a pc it runs good and I know it has more resources but still what is the problem?
> 
> The timout keeps coming back!

I doubt it's the internet connection. I strongly suspect there's something wrong in your code.  Without an example project we really can't help you.

podaen, 2022-05-12:
I m glad you want to look in to it. Thanks for that. You need more resources than that to test the whole thing, but I m looking in to decoded to pcm and output it with the internal dac. For the moment I don't have low power spreaker to test it. Beside of that I can still see it in the data.

Yes I did changed it to an int but keep looking...

podaen, 2022-05-13:
Here you can find my [PR](https://github.com/podaen/URL-Player-Simple). Feel free to comment bellow.

podaen, 2022-05-15:
I have made a library to evaluate streams, [Click](https://github.com/podaen/Arduino-UrlStream_Analyser) the link to findout.

closed 2022-11-06 by podaen