espressif/arduino-esp32 #6736: PSRAM for Stack and Heap - Arduino

gpashokgp, 2022-05-13:
### Related area

PSRAM

### Hardware specification

ESP32 Wrover E

### Is your feature request related to a problem?

, however, we 

We are unable to fit Heap and Stack within internal RAM and would like to use PSRAM as part of the main memory so that we can have a large Stack and Heap. However, we are able to use PSRAM for Malloc and buffers.

Your input to fix this issue is very much appreciated. Please let me know if you need any further information. 

### Describe the solution you'd like

Is there any configuration, which will help to fix this issue?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2022-05-13:
Hi @gpashokgp I did ask you on the forum to fill out ALL information.
We have no idea what version of this repository you are running, what operating system, what have you tried, what did not work.

gpashokgp, 2022-05-13:
Hi @me-no-dev , I will add more details to that.

me-no-dev, 2022-05-13:
You need to update your PIO installation. Newer versions of our repository map PSRAM to malloc (ps_malloc is not needed)

gpashokgp, 2022-05-13:
I am using the following platformio release

lbernstone, 2022-05-13:
@me-no-dev Is stack always allocated from dram?

me-no-dev, 2022-05-13:
i'm not 100% on this, but my logic is that if something is initialized before PSRAM is initialized (do_global_ctors should run before), then it will end up in DRAM. Things that otherwise get their memory allocation afterwards, should be in PSRAM if their size is above 4K. Things on the heap (above 4K) will go onto PSRAM too

lbernstone, 2022-05-13:
I would recommend using @jason2866 framework in your platformio.ini, as this uses more recent code than the included platformio arduino-esp32 framework:
`platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
`

Jason2866, 2022-05-13:
Please use latest build based on core 2.0.3
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip
```

zekageri, 2022-05-16:
So if we try to allocate more than 4k in the heap, it will use psram instead. But what if we allocate a bunch of things below 4k, can it crash the chip or will it allocate from psram if internal ram is not sufficent?

For example if i have 100k internal ram and i create 101 class with new where every class allocates 1k space, will the 101th class go to the psram?

zekageri, 2022-05-16:
Should there be an option to allocate every single possible variable in psram from the user app and use the internal ram for background purposes only, like task swtiching, wifi etc..

lbernstone, 2022-05-16:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#provide-external-ram-via-malloc
"If a suitable block of preferred internal/external memory is not available, the allocator will try the other type of memory."

zekageri, 2022-05-16:
So this makes all my `ps_malloc()` and all my ArduinoJSON custom `SpiRamAllocator` pointless. hmm...

Thanks for the response!

Jason2866, 2022-06-23:
We are using PSRAM in Project Tasmota. There is no general memory leak.
Please provide a minimal code example for this issue.

zekageri, 2022-06-24:
Sorry I dont have a memory leak, i just use big arrays and classes but every allocation happens via malloc but not ps_malloc. If i use ps_malloc directly, it will work and allocates memory from my ext ram, but never automatically.

lbernstone, 2022-06-25:
If you want more control over how your memory is allocated, you will need to learn how to [compile your own libraries](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html#esp32-arduino-lib-builder).  arduino-esp32 is set up to provide a simple, common base for supporting arduino projects with known good settings.  It is just a wrapper around esp-idf, and you can build that wrapper for yourself, in the manner you like.  The entry for this is [CONFIG_SPIRAM_MALLOC_ALWAYSINTERAL](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-malloc-alwaysinternal).  There are other menu options that will allow you to more aggressively use the psram, but if you don't understand the limitations, you will find yourself dealing with a lot of system panics.

LusterWong, 2022-06-28:
My problem seems similar. Because the system SRAM is not enough, you want to use PSRAM as the task stack.
I use ESP arduino v2.0.3.
```
static StackType_t* stack = (StackType_t*)ps_malloc(sizeof(StackType_t) * 8192);
static StaticTask_t tcb;

xTaskCreateStatic(TaskHandle, "TaskHandle", 8192, NULL, 20, stack, &tcb);
```
Compile successful but run occurs assert
```
assert failed: xTaskCreateStaticPinnedToCore tasks.c:677 (xPortcheckValidStackMem(pxStackBuffer))
```

VojtechBartoska, 2023-02-23:
I'm closing this issue as expired, if needed, you can reopen it.

and3rson, 2024-03-27:
I'm having the same issue as @LusterWong. ESP32-S3, latest stable arduino-esp32 framework.

```cpp
    soundTaskStack = static_cast<StackType_t*>(ps_malloc(8192 * sizeof(StackType_t)));
    soundTaskHandle = xTaskCreateStaticPinnedToCore(
        soundTask, "soundTask", 8192, sound_task_data, 1, soundTaskStack, &soundTaskBuffer, 0
    );

    // assert failed: xTaskCreateStaticPinnedToCore tasks.c:677 (xPortcheckValidStackMem(pxStackBuffer))
```

RadioDisney, 2024-12-29:
me too,this issue

RadioDisney, 2024-12-29:
> I'm closing this issue as expired, if needed, you can reopen it.

please reopen it

ValdayMl, 2025-04-11:
Arduino 1.8.12
Core 3.1.3
`xStack_loop_15s=(uint8_t*)heap_caps_calloc(1, 5*1024, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT | MALLOC_CAP_32BIT);`
and
`xStack_loop_15s=(StackType_t*)malloc(5*1024);
loop_15s_in_0_core=xTaskCreateStaticPinnedToCore(loop_15s_in_0_core_f,"15 sec loop",5*1024,NULL,1,xStack_loop_15s,&xTaskBuffer_loop_15s,1);
`
error
19:53:17.310 ->  create my static task
19:53:17.310 -> 93700 used bytes
19:53:17.310 -> 
19:53:17.310 -> assert failed: xTaskCreateStaticPinnedToCore freertos_tasks_c_additions.h:298 (xPortcheckValidStackMem(puxStackBuffer))
19:53:17.349 -> 
19:53:17.349 -> 
19:53:17.349 -> Backtrace: 0x40082b9d:0x3ffb1f90 0x400905c1:0x3ffb1fb0 0x40096de6:0x3ffb1fd0 0x40093f3e:0x3ffb2100 0x400ffd7e:0x3ffb2150 0x40128b77:0x3ffb2270 0x4009133e:0x3ffb2290