espressif/arduino-esp32 #6737: SOLUTION to -1 pin assignment causing apps wont start

tsctrl, 2022-05-13:
### Board

ESP32 Dev Module

### Device Description

Devkit 

### Hardware Configuration

Any SPI, I2c MODULES

### Version

v2.0.3

### IDE Name

IDF Component, Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

-1 pin assignment causing apps wont start:
In the Arduino world an I/O pin number of -1 is OK to signify to a function that a pin allocation is not wanted, Lots of existing arduino library written this way. Recent update causing apps wont start with repeated error as below:

```
[715][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
E (285) gpio: gpio_set_level(226): GPIO output gpio_num error
E (285) gpio: gpio_set_level(226): GPIO output gpio_num error
E (285) gpio: gpio_set_level(226): GPIO output gpio_num error
E (285) gpio: gpio_set_level(226): GPIO output gpio_num error
E (285) gpio: gpio_set_level(226): GPIO output gpio_num error
E (285) gpio: gpio_set_level(226): GPIO output gpio_num error
```

Would like to know the solution regarding this issue. This issue has been raised/mentioned in multiple esp arduino library.
Thanks!

### Sketch

```cpp
Assign any arduino library begin method in setup() with -1 pin assignment:

example:
SPI.begin(-1,-1,-1,-1)
```

### Debug Message

```plain
as in description.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-05-13:
And what is the expected result if you call `SPI.begin(-1,-1,-1,-1)` or `Wire.begin(-1, -1)` and which are those libraries? If you do not want to change the default pins, why not just call `SPI.begin()` or `Wire.begin()`?

tsctrl, 2022-05-14:
understood, we have to define and properly check those pin is suitable for assignment before initialize the io. 
initially adding pinMode(PULLUP, -1) is acceptable but now throwing those error. which need additional check before setting those mode. no other option. 

i am ok, close this if there is no other optional such as new NOTUSE pin definition to be use in other library as indication the pin and peripheral are not in use to replace the -1.

thanks!

tsctrl, 2022-05-14:
drawback on this error is,

for custom pin mapping in such way CLASS.begin can have multiple parameters to set. 
such as CLASS.begin(pina,pinb,pinc,pind) or CLASS.begin(pina)

we could not use CLASS.begin(pina,pinb,-1,-1) in case the pin is need to be set to default or not in use.
But since now it is returning error. we have to define CLASS.begin based on if the pin is set as below:

if(pina,pinb,pinc,pind)
CLASS.begin(pina,pinb,pinc,pind)
else if(pinc,pind)
CLASS.begin(pina,pinb)
endif

instead of just:
CLASS.begin(pina,pinb,pinc,pind)

without and conditional check

tsctrl, 2022-05-15:
another valid scenario is when we have to use EN instead of GPIO when we dont want to use the pin. previously we set this pin to -1.
then what pin to set now?
@me-no-dev here is the library:
https://github.com/adafruit/Adafruit-ST7735-Library/blob/c2b3a2c0970988b1c65e8fd0e14a23e96e95b662/Adafruit_ST7735.cpp#L12

me-no-dev, 2022-05-16:
Some pins do accept -1. Maybe only SDA/SCL and SCK/MISO/MOSI do not?

@SuGlider PTAL

P-R-O-C-H-Y, 2022-08-18:
Hi @tsctrl,

from reading this issue I am not sure if I got the point of it.
You can use either:

-  `SPI.begin()` - default pins used
-  `SPI.begin(-1,-1,-1,-1)` - default pins used
-  `SPI.begin(pina,pinb,-1,-1)` - will use pina and pinb + set default pins for rest

The library you sent, there is this option for setting -1 to unused the pin only for some specific pin, not for all.
Can you please explain it a bit more please?
@Bodmer Can you send the lib you have problem with? :)

VojtechBartoska, 2022-08-23:
Any updates on this guys?

SuGlider, 2022-08-24:
I think that PR #7161 addresses this issue.

P-R-O-C-H-Y, 2022-11-14:
@tsctrl Closing this due to no answer for a long time. Feel free to reopen if needed :)