espressif/arduino-esp32 #6738: The I2S capture time doesn't fit the sample time set

rainvalley, 2022-05-13:
### Board

ESP32-CAM

### Device Description

ESP32-CAM with ESP32S, and I have used the PSRAM to store the I2S's DMA data.

### Hardware Configuration

I2S_0 with ADC1_CHANNEL_7, the GPIO pin is GPIO35, the sample rate is 16K with 16bits per sample.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have tried to capture voice data by the I2S_0 port using ADC1_CH7. Because I'm trying to sending data to web API, so I have set the buff size to 160000*2 bytes on psram.

The I2S sample rate is 16K, and the bits per sample are 16bit. So the capture time one frame should be **10s** ? But when I test it on my board, the result is 907446us(0.907s).

When I change the sample rate to 8K, the time is 1814640us(1.814s); when the sample rate is 4K, the time is 3629080us(3.629us).

Maybe I mistake the documentation about the function.

### Sketch

```cpp
//Vars related to the code:
#define len_song 160000
uint16_t *song = NULL;
uint8_t flag_song = 0;
song = (uint16_t *)ps_calloc(len_song, sizeof(uint16_t));

//Below is the i2s initialization:
void i2s_init()
{
  i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),
    .sample_rate =  16000,              // The format of the signal using ADC_BUILT_IN
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, // is fixed at 12bit, stereo, MSB
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S_MSB,
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 4,
    .dma_buf_len = 8,
    .use_apll = false,
    .tx_desc_auto_clear = false,
    .fixed_mclk = 0
  };
  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
  i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_7);
  i2s_adc_enable(I2S_NUM_0);
}

//Below is the test code:
size_t byte_read;
size_t start_time = micros();
i2s_read(I2S_NUM_0, (void*)song, len_song * 2, &byte_read, portMAX_DELAY);
Serial.print("I2S has read bytes:"); Serial.println(byte_read);
Serial.print("I2S has passed:"); Serial.println(micros() - start_time);
```

### Debug Message

```plain
setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[00:39:19.772] [   842][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[00:39:19.830] [   904][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
[00:39:19.940] [  1014][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
[00:39:20.012] [  1078][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
SD_MMC Card Size: 120MB
.
[00:39:20.342] [  1408][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  1408][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.137.213, MASK: 255.255.255.0, GW: 192.168.137.1
[00:39:21.010] ESP Ready! The IP is:192.168.137.213
[00:39:21.438] Connnection Opened
[00:39:21.542] I2S has read bytes:32000
I2S has passed:90576
[00:39:21.654] Start to send song identify data!
[00:39:21.835] Connnection Closed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

rainvalley, 2022-05-13:
The debug message uploaded is made for `I2S has read bytes:32000` , so the true capture time should be 1s?

SuGlider, 2022-05-15:
@PilnyTomas - Can you help on it?

PilnyTomas, 2022-05-16:
Hi @rainvalley, can you please provide a complete minimal sketch? something that I could flash right away and would demonstrate the issue?

VojtechBartoska, 2022-06-14:
Any updates @rainvalley?

VojtechBartoska, 2022-07-15:
Closing issue as expired due to no answer.

If you needed, you can reopen it. Thanks for understanding.

closed 2022-07-15 by VojtechBartoska