espressif/arduino-esp32 #6739: GATTS_ReserveHandles: no free handle blocks - how can I get more BLE services?

paulhamsh, 2022-05-13:
### Board

EPS32-C3

### Device Description

Beetle-ESP32-C3

### Hardware Configuration

n/a

### Version

v2.0.3

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Following on from issue 6692, this is the output of a test program on an ESP32-C3 with v2.0.3 library. Now it shows a real error message - but how can I increase the number of handles? Reducing the number of characteristics doesn't help - it seems to max out on Services.

Code:

```
E (1104) BT_GATT: GATTS_ReserveHandles: no free handle blocks
```

### Sketch

```cpp
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE Server");

  BLEServer* pSCServer;
  BLEDevice::init("SKC50S-4 v3.0.1 9E02");
  pSCServer = BLEDevice::createServer();
      
  BLEService* pS1 = pSCServer->createService(BLEUUID("12345678-1234-0001-0000-000000000001"),10);        
  BLECharacteristic* pCa1 = pS1->createCharacteristic("00000001-1234-1234-1234-000000000001", BLECharacteristic::PROPERTY_READ); 
  BLECharacteristic* pCa2 = pS1->createCharacteristic("00000001-1234-1234-1234-000000000002", BLECharacteristic::PROPERTY_READ); 
 
  BLEService* pS2 = pSCServer->createService(BLEUUID("12345678-1234-0001-0000-000000000002"),10);        
  BLECharacteristic* pCb1 = pS2->createCharacteristic("00000002-1234-1234-1234-000000000001", BLECharacteristic::PROPERTY_READ); 
  BLECharacteristic* pCb2 = pS2->createCharacteristic("00000002-1234-1234-1234-000000000002", BLECharacteristic::PROPERTY_READ); 

[... 94 lines cut ...]
[  1058][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[  1067][D][BLECharacteristic.cpp:90] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: 00000005-1234-1234-1234-000000000002, service: UUID: 12345678-1234-0001-0000-000000000005, handle: 0x0050
[  1087][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[  1096][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[  1106][D][BLECharacteristic.cpp:90] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: 00000006-1234-1234-1234-000000000001, service: UUID: 12345678-1234-0001-0000-000000000006, handle: 0x005a
[  1125][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[  1134][D][BLECharacteristic.cpp:90] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: 00000006-1234-1234-1234-000000000002, service: UUID: 12345678-1234-0001-0000-000000000006, handle: 0x005a
[  1154][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[  1163][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[  1173][D][BLECharacteristic.cpp:90] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: 00000007-1234-1234-1234-000000000001, service: UUID: 12345678-1234-0001-0000-000000000007, handle: 0x0000
E (1561) BT_APPL: service not created
```
```

### Other Steps to Reproduce

The seventh service fails on all platforms - but C3 gives a proper error message, the rest just seem to say something like

```
[  2947][E][BLEService.cpp:284] handleGATTServerEvent(): Expected to find characteristic with UUID: 362f71a0-6c96-11e3-981f-0800200c9a66, but didnt!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-05-13:
CONFIG_BT_GATT_MAX_SR_PROFILES is compiled into the included esp-idf library archives.  You would need to [build your own](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html) using [lib-builder](https://github.com/espressif/esp32-arduino-lib-builder) if you want to increase this.

paulhamsh, 2022-05-13:
Thank you.
It is odd, but the NimBLE library doesn’t have this limit.

paulhamsh, 2022-05-13:
Ah - just realised NimBLE seems to have its own ESP-IDF component so it is probably very different

VojtechBartoska, 2022-05-17:
> Ah - just realised NimBLE seems to have its own ESP-IDF component so it is probably very different

Did you find a solution for your issue?

paulhamsh, 2022-05-17:
Only the one [lbernstone] mentioned - recompiling the IDF. Seems a shame CONFIG_BT_GATT_MAX_SR_PROFILES can't be increased for all builds as standard, or that it isn't dynamic to start with. A strange limitation.

VojtechBartoska, 2022-06-14:
Closing as won't do. If needed, you can reopen this.

closed 2022-06-14 by VojtechBartoska