espressif/arduino-esp32 #6741: Can't attach interrupt to GPIO38 to wake from deep sleep

DustinWatts, 2022-05-13:
### Board

ESP32S3 Dev Module

### Device Description

ESP32-S3-DevKitC-1 V1.0 attached to a ILI9488 TFT screen with FT6202 touch controller over I2C.

### Hardware Configuration

GPIO 17 and 18 are connected to the FT6202. GPIO38 is connected to the touchscreen's IRQ pin.

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Attaching an interrupt to GPIO38 does not wake the ESP32 from deep sleep. Although the documentation says it should:

```
gpio_num: GPIO number used as wakeup source. 
Only GPIOs which are have RTC functionality can be used: 0,2,4,12-15,25-27,32-39. 
```
Source: https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32s3/api-reference/system/sleep_modes.html#_CPPv428esp_sleep_enable_ext0_wakeup10gpio_num_ti

The function `esp_sleep_is_valid_wakeup_gpio()` also confirms that this does not work, because it returns FALSE. 

Used GPIO1 as a test, and that works although the docs says it shouldn't. 

### Sketch

```cpp
pinMode(GPIO_NUM_38, INPUT_PULLUP);

esp_sleep_enable_ext0_wakeup(GPIO_NUM_38, 0);
esp_deep_sleep_start();
```

### Debug Message

```plain
None related to this issue. No GPIO errors.
```

### Other Steps to Reproduce

To reproduce, try to wake up from deep sleep using GPIO38.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

DustinWatts, 2022-05-13:
Update:

Using a simple sketch to check for valid wakeup GPIOs only pins 1 trough 21 are valid:

```
Wake up should work on GPIO_NUM_1: 1
Wake up should work on GPIO_NUM_2: 1
Wake up should work on GPIO_NUM_3: 1
Wake up should work on GPIO_NUM_4: 1
Wake up should work on GPIO_NUM_5: 1
Wake up should work on GPIO_NUM_6: 1
Wake up should work on GPIO_NUM_7: 1
Wake up should work on GPIO_NUM_8: 1
Wake up should work on GPIO_NUM_9: 1
Wake up should work on GPIO_NUM_10: 1
Wake up should work on GPIO_NUM_11: 1
Wake up should work on GPIO_NUM_12: 1
Wake up should work on GPIO_NUM_13: 1
Wake up should work on GPIO_NUM_14: 1
Wake up should work on GPIO_NUM_15: 1
Wake up should work on GPIO_NUM_16: 1
Wake up should work on GPIO_NUM_17: 1
Wake up should work on GPIO_NUM_18: 1
Wake up should work on GPIO_NUM_19: 1
Wake up should work on GPIO_NUM_20: 1
Wake up should work on GPIO_NUM_21: 1
Wake up should work on GPIO_NUM_26: 0
```

From 26 on `esp_sleep_is_valid_wakeup_gpio()` returns FALSE.

lbernstone, 2022-05-13:
Please report at the upstream project. https://github.com/espressif/esp-idf/issues

me-no-dev, 2022-05-14:
this is an issue in the documentation. Probably left over from older chips. On ESP32-S3, IOs 0-21 are RTC capable.

DustinWatts, 2022-05-14:
Yes thanks, figured it out!

closed 2022-05-14 by DustinWatts