espressif/arduino-esp32 #6742: onewire does not work in V2.0.3 while it is OK in 2.0.2

aalucia, 2022-05-13:
### Board

ESP32-s2

### Device Description

ESP 32 s2 basic board

### Hardware Configuration

No

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Suse Linux

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

57600

### Description

After compiling the very basic example below, while OK with 2.0.2 with 2.0.3 gives the error(s).

### Sketch

```cpp
/*********
  Rui Santos
  Complete project details at https://RandomNerdTutorials.com  
*********/

#include <OneWire.h>
#include <DallasTemperature.h>

// GPIO where the DS18B20 is connected to
const int oneWireBus = 4;     

// Setup a oneWire instance to communicate with any OneWire devices
OneWire oneWire(oneWireBus);

// Pass our oneWire reference to Dallas Temperature sensor 
DallasTemperature sensors(&oneWire);

void setup() {
  // Start the Serial Monitor
  Serial.begin(115200);
  // Start the DS18B20 sensor
  sensors.begin();
}

void loop() {
  sensors.requestTemperatures(); 
  float temperatureC = sensors.getTempCByIndex(0);
  float temperatureF = sensors.getTempFByIndex(0);
  Serial.print(temperatureC);
  Serial.println("ºC");
  Serial.print(temperatureF);
  Serial.println("ºF");
  delay(5000);
}
```

### Debug Message

```plain
In file included from /home/alex/.arduino15/packages/esp32/hardware/esp32/2.0.3/cores/esp32/Arduino.h:36,
                 from /home/alex/Arduino/libraries/OneWire/OneWire.cpp:142:
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:191:38: error: 'esp32_gpioMux' was not declared in this scope
         ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;
                                      ^~~~~~~~~~~~~
/home/alex/.arduino15/packages/esp32/hardware/esp32/2.0.3/cores/esp32/esp32-hal.h:73:47: note: in definition of macro 'ESP_REG'
 #define ESP_REG(addr) *((volatile uint32_t *)(addr))
                                               ^~~~
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:232:38: error: 'esp32_gpioMux' was not declared in this scope
         ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;
                                      ^~~~~~~~~~~~~
/home/alex/.arduino15/packages/esp32/hardware/esp32/2.0.3/cores/esp32/esp32-hal.h:73:47: note: in definition of macro 'ESP_REG'
 #define ESP_REG(addr) *((volatile uint32_t *)(addr))
                                               ^~~~
exit status 1
Error compiling for board ESP32S2 Dev Module.
```

### Other Steps to Reproduce

none, just upload the code and use v 2.0.3

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-05-13:
Please search before you post. https://github.com/espressif/arduino-esp32/issues?q=is%3Aissue+onewire+sort%3Aupdated-desc+
Duplicate https://github.com/espressif/arduino-esp32/issues/6559

aalucia, 2022-05-13:
Hi, I've added it because is present also for esp32-s2, I should have had a more clear title for the issue.

lbernstone, 2022-05-13:
And it is a dupe, since OneWire is not part of this repo.

aalucia, 2022-05-14:
Hi, so what I should do, delete the request since is a duplicate ? Cheers, Alex.

lbernstone, 2022-05-14:
You can just close it and subscribe to the other issue.

aalucia, 2022-05-14:
HI, thanks for the instructions. Hope You'll catch the bug soon. Cheers.

closed 2022-05-14 by aalucia

lbernstone, 2022-05-14:
The solution is known.  It just can't be fixed here.