espressif/arduino-esp32 #6744: Wifi setting not return to default

roysG, 2022-05-14:
### Board

Esp32 dev module

### Device Description

Esp32 dev module

### Hardware Configuration

Esp32 dev module

### Version

v2.0.0

### IDE Name

Arduino IDE

### Operating System

macOS 12

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I setted the wifi to work with protocol LR (long range).

Now i need to return the configuration wifi to his default, but all the commands i tried, did not helped.

I have to use the default configuration cause i need that the wifi will connect to my home network.

I added the code i try to run to connect, but with no success. I also tried this code on few esp32 i have. in all of them i have the same problem.

### Sketch

```cpp
/*
  Complete details at https://RandomNerdTutorials.com/esp32-useful-wi-fi-functions-arduino/
*/

#include <esp_wifi.h>
#include <WiFi.h>

// Replace with your network credentials (STATION)
const char* ssid = "roy";
const char* password = "123456789";

void initWiFi() {
    //Serial.println("Get protocl");
    //Serial.println(esp_wifi_get_protocol());  

    esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11G);  
    Serial.println("Setted The Protocol2");

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.print("Connecting to WiFi ..");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(1000);
  }
  Serial.println(WiFi.localIP());
}

void setup() {
  Serial.begin(115200);
  initWiFi();
  Serial.print("RRSI: ");
  Serial.println(WiFi.RSSI());
}

void loop() {
  // put your main code here, to run repeatedly:
  
   Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }

  delay(1000);
}
}
```

### Debug Message

```plain
load:0x40078000,len:13540
load:0x40080400,len:3604
entry 0x400805f0
Setted The Protocol2
Connecting to WiFi .............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-05-14:
I don't believe there is any such persistence for LR mode, but you can wipe the device using `python <path>/esptool_py/3.0.1/esptool.py erase_flash` from the command line.
I don't know the default path on Mac- on linux it is ~/.arduino15/packages/esp32/tools/

me-no-dev, 2022-05-14:
it's `~/Library/Arduino15/packages/esp32/tools/` on a Mac

me-no-dev, 2022-05-14:
BTW, if you select esptool as programmer in the board menu and then hit "Burn Bootloader" it will actually wipe the flash (maybe worth adding to docs @pedrominatel)

lbernstone, 2022-05-14:
Oh yeah, that is much easier than explaining how to run a cmdline tool.

roysG, 2022-05-14:
> After many test, even in a new computer that nothing was insalled before.
> The problem is concern to LR.
>
> Do you have any suggestion to test and send you result.
>
> I am running on this problem whole the day. It ahould be very simple.
>
> When i try to run as hotspot. Just others esp32 can see it in the list,
> but not regular wifi.
>
> I tried also to run the same code of hotspot in new computer that no any
> packages installed before. Same result

Sorry, i meant is NOT concern yo LR protocol

roysG, 2022-05-14:
Update:
I took new esp32 and in the new esp32 is working great.

How can i reset the others esp32 to their wifi defaults factory?

lbernstone, 2022-05-14:
Tools -> Programmer -> Esptool
Tools -> Burn Bootloader
This will wipe the flash memory, which returns your device to factory defaults.

roysG, 2022-05-14:
I get error:
Error while burning the bootloader: Missing programmer

lbernstone, 2022-05-14:
Upgrade to latest version (https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) or follow the instruction above to erase flash from command line.

roysG, 2022-05-14:
Works by command line!

closed 2022-05-14 by roysG

me-no-dev, 2022-05-14:
> Oh yeah, that is much easier than explaining how to run a cmdline tool.

BTW Sketch->Upload Using Programmer will upload just the app bin (if the programmer is selected)

> I get error in arduino 2.0

Which version of this repo are you running? Maybe outdated

roysG, 2022-05-15:
I tried to do the same command for the esp32s2

But this is the error i get:

esptool.py -p /dev/tty.usbmodem01 erase_flash
esptool.py v3.2-dev
Serial port /dev/tty.usbmodem01
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Traceback (most recent call last):
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 374, in detect_chip
    res = struct.unpack("<IBBBBBBBBI", res[:16])  # 4b flags, 1b flash_crypt_cnt, 7*1b key_purposes, 4b chip_id
struct.error: unpack requires a buffer of 16 bytes

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/roysabah/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/serial/serialposix.py", line 575, in read
    buf = os.read(self.fd, size - len(read))
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 465, in command
    p = self.read()
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 410, in read
    return next(self._slip_reader)
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 3302, in slip_reader
    read_bytes = port.read(1 if waiting == 0 else waiting)
  File "/Users/roysabah/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/serial/serialposix.py", line 581, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/roysabah/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/serial/serialposix.py", line 398, in _reconfigure_port
    orig_attr = termios.tcgetattr(self.fd)
termios.error: (6, 'Device not configured')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 4930, in <module>
    _main()
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 4923, in _main
    main()
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 4370, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 118, in get_default_connected_device
    _esp = ESPLoader.detect_chip(each_port, initial_baud, before, trace,
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 385, in detect_chip
    detect_port.connect(connect_mode, connect_attempts, detecting=True, warnings=False)  # Need to connect again
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 660, in connect
    last_error = self._connect_attempt(mode=mode, usb_jtag_serial=usb_jtag_serial, extra_delay=extra_delay)
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 625, in _connect_attempt
    self.sync()
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 516, in sync
    val, _ = self.command(self.ESP_SYNC, b'\x07\x07\x12\x20' + 32 * b'\x55',
  File "/Users/roysabah/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 481, in command
    self._port.timeout = saved_timeout
  File "/Users/roysabah/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/serial/serialutil.py", line 372, in timeout
    self._reconfigure_port()
  File "/Users/roysabah/.espressif/python_env/idf5.0_py3.9_env/lib/python3.9/site-packages/serial/serialposix.py", line 401, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (6, 'Device not configured')

lbernstone, 2022-05-15:
Again, outdated repo.  v2.0.0 was not complete for s2 and c3.

roysG, 2022-10-11:
After many test, even in a new computer that nothing was insalled before.
The problem is concern to LR.

Do you have any suggestion to test and send you result.

I am running on this problem whole the day. It ahould be very simple.

When i try to run as hotspot. Just others esp32 can see it in the list, but
not regular wifi.

I tried also to run the same code of hotspot in new computer that no any
packages installed before. Same result