espressif/arduino-esp32 #6746: Sntp time WiFi fixed IP not Working

atumra1975, 2022-05-14:
### Board

TTGo LilyGo Lora 32-OLED

### Device Description

[TTGo LilyGo](https://github.com/Xinyuan-LilyGO/TTGO-T-Display)

### Hardware Configuration

Default

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

W11

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm using Acetime as timelib (same using standard timelib)
When fixed IP is set on the sketch, it's impossible to retrieve SNTP time.
Otherwise, when DHCP is set, everythig goes fine.
Even if the reservation is on the router, the board cannot reach the srv.
No blocking fw rules r set onto the router...
TIP: i've set an exposed host (Board IP) - nothing changed
I've defined local ip, netmask, primary dns and secondary dns
Changed pri and sec dns from internal (router) to external (8.8.8.8, 8.8.4.4) and also from my isp
No luck...

### Sketch

```cpp
#include <Arduino.h>
#include <AceTime.h>
#include <Arduino.h>
#include <ESP32Ping.h>
...
#include <TimeLib.h>
#include <String.h>
#include <Wire.h>
#include <WiFi.h>
...
using namespace ace_time;
...
#ifdef STATIC_IP
IPAddress local_IP(192, 168, 188, 200);
IPAddress gateway(192, 168, 188, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress primaryDNS(8, 8, 8, 8); 
IPAddress secondaryDNS(8, 8, 4, 4);
#endif
...
const char* ntpServerName = "it.pool.ntp.org";
...
WiFi.mode(WIFI_STA);
#ifndef DYNAMIC_IP
  scriviSeriale("Imposto i parametri statici" , true);
  WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS); //
#endif
WiFi.setHostname(ID_DEVICE);
WiFi.begin(WIFI_SSID,WIFI_PASSWD);
while ( WiFi.status() != WL_CONNECTED && wificount <= wifiretry)
 {
    scriviSeriale("Attesa WiFi.status()" , true);
    delay(1000);
    wificount += 1;
  }
  if (wificount > wifiretry){
    scriviSeriale("Retry > wificount" , true);
    return false;
  }
...
configTime(0 , 0 , ntpServerName);
epochNow = time(nullptr);
if (epochNow >= EPOCH_2000_01_01) 
{
   LocalDateTime ldt = LocalDateTime::forUnixSeconds64(epochNow);
   tzRome = TimeZone::forZoneInfo(&zonedbx::kZoneEurope_Rome,&zoneProcessorRome);
   zdtRome = ZonedDateTime::forUnixSeconds64(epochNow, tzRome);
   scriviSeriale("Done.",true);
   scriviSeriale("Data Rilevata: " + formatLocalDate(zdtRome) + " " + formatLocalTime(zdtRome),true);
   return true;
}
...
```

### Debug Message

```plain
//no errors - only srv unreachable in fixed mode
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-05-14:
Please enable debug to Verbose in the board menu and then post FULL uart log. I tested this exact case two days ago and it worked just fine

atumra1975, 2022-05-16:
DHCP Enabled
[ 12854][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: xxxxxx, BSSID: xx:xx:xx:xx:xx:xx, Reason: 203
[ 12854][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 12862][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 203 - ASSOC_FAIL
[ 12869][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
[ 12877][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 12954][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: xxxxxx, BSSID: xx:xx:xx:xx:xx:xx, Channel: 6, Auth: UNKNOWN
[ 12956][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 12982][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got New IP:192.168.188.52
[ 12983][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 12986][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.188.52, MASK: 255.255.255.0, GW: 192.168.188.1
STATIC IP
[ 12865][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: xxxxx, BSSID: xx:xx:xx:xx:xx:xx, Reason: 203
[ 12865][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 12873][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 203 - ASSOC_FAIL
[ 12880][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
[ 12951][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: xxxxx, BSSID: xx:xx:xx:xx:xx:xx, Channel: 6, Auth: UNKNOWN
[ 12953][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 12955][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got New IP:192.168.188.200
[ 12968][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 12975][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.188.200, MASK: 255.255.255.0, GW: 192.168.188.1

Another sketch directly from examples (wifi.h)
Connecting to NIKKIOWiFi
[    67][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   165][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   164][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
[   171][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
[   180][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 192.168.188.200, MASK: 255.255.255.0, GW: 192.168.188.1
E (573) wifi:Association refused temporarily, comeback time 1024 mSec
....[  2227][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: xxxxx, BSSID: xx:xx:xx:xx:xx:xx, Reason: 203
[  2228][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  2235][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 203 - ASSOC_FAIL
[  2242][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
[  2306][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: xxxxx, BSSID: xx:xx:xx:xx:xx:xx, Channel: 6, Auth: UNKNOWN
[  2307][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  2309][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got New IP:192.168.188.200
[  2322][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  2329][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.188.200, MASK: 255.255.255.0, GW: 192.168.188.1
.
WiFi connected
IP address: 
192.168.188.200
connecting to data.sparkfun.com
[ 21046][E][WiFiGeneric.cpp:1360] hostByName(): DNS Failed for data.sparkfun.com
connection failed
connecting to data.sparkfun.com
[ 40046][E][WiFiGeneric.cpp:1360] hostByName(): DNS Failed for data.sparkfun.com
connection failed
It seems unable to resolve DNS

thanks

lbernstone, 2022-05-22:
What news are you expecting?  Your DNS is failing.  Fix that and your time should resolve correctly.  If you want to confirm that DNS is the problem, then put in an IP address for a time server (ping ntp.pool.org to get one), and your time should sync.

atumra1975, 2022-05-23:
thanks a lot for your answer

closed 2022-05-23 by atumra1975