espressif/arduino-esp32 #6747: spiTransferBytesNL Warning during compile in platformio

jowser7, 2022-05-15:
### Board

Heltec WiFi Kit 32 (V2)

### Device Description

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_kit_32_v2.html
PLATFORM: Espressif 32 (3.5.0) > Heltec WiFi Kit 32 (V2)
HARDWARE: ESP32 240MHz, 320KB RAM, 8MB Flash

### Hardware Configuration

uart3
baud_rate: 2400
tx_pin: GPIO19
rx_pin: GPIO22

### Version

latest master (checkout manually)

### IDE Name

esphome, platformio

### Operating System

Windows 10

### Flash frequency

26 Mhz Crystal 

### PSRAM enabled

no

### Upload speed

450800

### Description

Using ESPhome to build pipsolar uses platformio. ESPhome developers say this is arduino issue with the framework.

In function 'spiTransferBytesNL':
.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^

### Sketch

```cpp
command:
esphonme run esp32example.yaml with 
esp32:  
  board: heltec_wifi_kit_32_v2
  framework: 
    type: arduino
```

### Debug Message

```plain
Compiling .pioenvs\pipsolar\FrameworkArduino\esp32-hal-time.c.o
Compiling .pioenvs\pipsolar\FrameworkArduino\esp32-hal-timer.c.o
.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
C:/Users/aaaaa/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
C:/Users/aaaaa/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-05-15:
You use Arduino core 1.0.6. This core version is outdated and not supported anymore.
Please use and try actual Arduino version 2.0.3

jowser7, 2022-05-15:
> You use Arduino core 1.0.6. This core version is outdated and not supported anymore. Please use and try actual Arduino version 2.0.3
Thanks Jason,  I had run all these platformio updates and thought it was most recent.  Was this the key identifier? 
framework-arduinoespressif32@3.10006.210326   << Is that hiding the 1.0.6 ?

closed 2022-05-15 by jowser7

amjadali56, 2022-11-01:
The issue is in the Espressif@^3.5.0 library esp32-hal-spi.c file where there is an incompatible pointer.
...../platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/esp32-hal-spi.c
The ‘result’ and ‘last_data’ are initialized as uint32_t.

The original code:
uint8_t * last_out8 = &result[c_longs-1];
uint8_t * last_data8 = &last_data;

To this:
uint32_t * last_out8 = &result[c_longs-1];
uint32_t * last_data8 = &last_data;