espressif/arduino-esp32 #6748: undefined reference to `esp_hf_client_outgoing_data_ready'

jokubasver, 2022-05-15:
### Board

ESP32 Dev Module

### Device Description

Generic 36 pin ESP32 dev module

### Hardware Configuration

UDA1334A DAC board, MCP2515 CAN board

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm using some of the HFP Client features in my sketch to use car steering wheel buttons to control my phone via Bluetooth. 
These are the parts that work just fine:
```
esp_hf_client_register_callback
esp_hf_client_init
esp_hf_client_start_voice_recognition
esp_hf_client_answer_call
esp_hf_client_reject_call
```
And I'm able to trigger the voice assistant, answer and reject phone calls using the car's steering wheel buttons, which is great.

Now I'm trying to route phone call audio, and it seems like I need to use `esp_hf_client_outgoing_data_ready`. The code compiles fine, but fails at the linking stage:
```
`Compiling .pio\build\esp32dev\src\main.cpp.o
Linking .pio\build\esp32dev\firmware.elf
c:/users/xxxxxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.cpp.o:(.literal._Z5setupv+0x34): undefined reference to `esp_hf_client_outgoing_data_ready'
c:/users/xxxxxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.cpp.o: in function `setup()':
C:\Users\xxxxxx\Documents\GitHub\xxxxxx/src/main.cpp:149: undefined reference to `esp_hf_client_outgoing_data_ready`
```

I've tried now and old versions of arduino-esp32, same with platformio esp32 platform - no difference.

### Sketch

```cpp
#include <Arduino.h>
#include <esp_hf_client_api.h>
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void bt_hf_client_cb(esp_hf_client_cb_event_t event, esp_hf_client_cb_param_t *param)
{
}

void setup() {
  // Enable serial
  Serial.begin(115200);

  // Start the A2DP sink
  a2dp_sink.start("BT");  

  // Setup HFP
  esp_hf_client_register_callback(bt_hf_client_cb);
  esp_hf_client_init();
  esp_hf_client_outgoing_data_ready();
}

void loop() {
  // HF Client functions that work just fine on Arduino-ESP32
  esp_hf_client_start_voice_recognition();
  delay(1000);
  esp_hf_client_answer_call();
  delay(1000);
  esp_hf_client_reject_call();
  delay(1000);
}
```

### Debug Message

```plain
Compiling .pio\build\esp32dev\src\main.cpp.o
Linking .pio\build\esp32dev\firmware.elf
c:/users/xxxxxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.cpp.o:(.literal._Z5setupv+0x34): undefined reference to `esp_hf_client_outgoing_data_ready'
c:/users/xxxxxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.cpp.o: in function `setup()':
C:\Users\xxxxxx\Documents\GitHub\xxxxxx/src/main.cpp:149: undefined reference to `esp_hf_client_outgoing_data_ready
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-05-15:
Not experienced with BT. Is the IDF feature in Arduino there. T

jokubasver, 2022-05-15:
> Not experienced with BT. Is the IDF feature in Arduino available? This is the [kconfig](https://github.com/espressif/esp-idf/blob/release/v4.4/components/bt/Kconfig) for and [thats](https://github.com/espressif/esp32-arduino-lib-builder/blob/master/configs/defconfig.esp32) are the settings how Arduino libs are compiled

`CONFIG_BT_HFP_ENABLE=y`
So seems good, especially when other HFP functions work just fine on Arduino. It's just `esp_hf_client_outgoing_data_ready` that fails at the linker

Jason2866, 2022-05-15:
Mhh, from the IDF docu
```
void esp_hf_client_outgoing_data_ready(void)
Trigger the lower-layer to fetch and send audio data. This function is only only used in the case that Voice Over HCI is enabled. After this function is called, lower layer will invoke esp_hf_client_outgoing_data_cb_t to fetch data.
```
Is `Voice Over HCI` enabled?

closed 2025-01-16 by Parsaabasi