espressif/arduino-esp32 #6749: FFAT exists throws error when file does not exist

frankcohen, 2022-05-15:
### Board

Adafruit Feather ESP32-S3

### Device Description

I added these lines to the examples->FFat->FFat_Test at line 182:

    if ( FFat.exists( "/hello.txt" ) )
    {
      Serial.println( "hello.txt exists");
    }
    else
    {
      Serial.println( "hello.txt does not exist");
    }
      
    if ( FFat.exists( "/foo.txt" ) )
    {
      Serial.println( "foo.txt exists");
    }
    else
    {
      Serial.println( "foo.txt does not exist");
    }

I see this in the Arduino IDE Serial Monitor:

...
- read from file:
Hello World!
[  6785][E][vfs_api.cpp:104] open(): /ffat/hello.txt does not exist, no permits for creation
foo.txt does not exist
foo.txt exists
Listing directory: /
  FILE: foo.txt	SIZE: 14
...

I expected file.exists() would return an error.

-Frank

### Hardware Configuration

SD NAND card on the SPI bus with chip select on GPIO 10

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE 1.8.10

### Operating System

MacOS 12.3.1

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

I initialize the Serial object with:
    Serial.begin(115200);
    Serial.setDebugOutput(true);
    delay(5000);

exists() does not seem to return a boolean value when the file does not exist.

### Sketch

```cpp
#include "FS.h"
#include "FFat.h"

// This file should be compiled with 'Partition Scheme' (in Tools menu)
// set to 'Default with ffat' if you have a 4MB ESP32 dev module or
// set to '16M Fat' if you have a 16MB ESP32 dev module.

// You only need to format FFat the first time you run a test
#define FORMAT_FFAT true

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\r\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("- failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println(" - not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("\tSIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void readFile(fs::FS &fs, const char * path){
    Serial.printf("Reading file: %s\r\n", path);

    File file = fs.open(path);
    if(!file || file.isDirectory()){
        Serial.println("- failed to open file for reading");
        return;
    }

    Serial.println("- read from file:");
    while(file.available()){
        Serial.write(file.read());
    }
    file.close();
}

void writeFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Writing file: %s\r\n", path);

    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("- failed to open file for writing");
        return;
    }
    if(file.print(message)){
        Serial.println("- file written");
    } else {
        Serial.println("- write failed");
    }
    file.close();
}

void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\r\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("- failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("- message appended");
    } else {
        Serial.println("- append failed");
    }
    file.close();
}

void renameFile(fs::FS &fs, const char * path1, const char * path2){
    Serial.printf("Renaming file %s to %s\r\n", path1, path2);
    if (fs.rename(path1, path2)) {
        Serial.println("- file renamed");
    } else {
        Serial.println("- rename failed");
    }
}

void deleteFile(fs::FS &fs, const char * path){
    Serial.printf("Deleting file: %s\r\n", path);
    if(fs.remove(path)){
        Serial.println("- file deleted");
    } else {
        Serial.println("- delete failed");
    }
}

void testFileIO(fs::FS &fs, const char * path){
    Serial.printf("Testing file I/O with %s\r\n", path);

    static uint8_t buf[512];
    size_t len = 0;
    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("- failed to open file for writing");
        return;
    }

    size_t i;
    Serial.print("- writing" );
    uint32_t start = millis();
    for(i=0; i<2048; i++){
        if ((i & 0x001F) == 0x001F){
          Serial.print("*");
        }
        file.write(buf, 512);
    }
    Serial.println("");
    uint32_t end = millis() - start;
    Serial.printf(" - %u bytes written in %u ms\r\n", 2048 * 512, end);
    file.close();

    file = fs.open(path);
    start = millis();
    end = start;
    i = 0;
    if(file && !file.isDirectory()){
        len = file.size();
        size_t flen = len;
        start = millis();
        Serial.print("- reading" );
        while(len){
            size_t toRead = len;
            if(toRead > 512){
                toRead = 512;
            }
            file.read(buf, toRead);
            if ((i++ & 0x001F) == 0x001F){
              Serial.print("+");
            }
            len -= toRead;
        }
        Serial.println("");
        end = millis() - start;
        Serial.printf("- %u bytes read in %u ms\r\n", flen, end);
        file.close();
    } else {
        Serial.println("- failed to open file for reading");
    }
}

void setup(){
    Serial.begin(115200);
    Serial.setDebugOutput(true);
    delay(5000);
    
    if (FORMAT_FFAT) FFat.format();
    if(!FFat.begin( 10 )){
        Serial.println("FFat Mount Failed");
        return;
    }

    Serial.printf("Total space: %10u\n", FFat.totalBytes());
    Serial.printf("Free space: %10u\n", FFat.freeBytes());
    listDir(FFat, "/", 0);
    writeFile(FFat, "/hello.txt", "Hello ");
    appendFile(FFat, "/hello.txt", "World!\r\n");
    readFile(FFat, "/hello.txt");
    renameFile(FFat, "/hello.txt", "/foo.txt");
    readFile(FFat, "/foo.txt");

    if ( FFat.exists( "/hello.txt" ) )
    {
      Serial.println( "hello.txt exists");
    }
    else
    {
      Serial.println( "hello.txt does not exist");
    }
      
    if ( FFat.exists( "/foo.txt" ) )
    {
      Serial.println( "foo.txt exists");
    }
    else
    {
      Serial.println( "foo.txt does not exist");
    }

    listDir(FFat, "/", 0);
    
    deleteFile(FFat, "/foo.txt");
    testFileIO(FFat, "/test.txt");
    Serial.printf("Free space: %10u\n", FFat.freeBytes());
    deleteFile(FFat, "/test.txt");
    Serial.println( "Test complete" );
}

void loop(){

}
```

### Debug Message

```plain
***********************************
 - 1048576 bytes written in 13003 ms
- reading++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
- 1048576 bytes read in 127 ms
Free space:    2752512
Deleting file: /test.txt
- file deleted
Test complete
Total space:    3801088
Free space:    3801088
Listing directory: /
Writing file: /hello.txt
- file written
Appending to file: /hello.txt
- message appended
Reading file: /hello.txt
- read from file:
Hello World!
Renaming file /hello.txt to /foo.txt
- file renamed
Reading file: /foo.txt
- read from file:
Hello World!
[  6785][E][vfs_api.cpp:104] open(): /ffat/hello.txt does not exist, no permits for creation
foo.txt does not exist
foo.txt exists
Listing directory: /
  FILE: foo.txt	SIZE: 14
Deleting file: /foo.txt
- file deleted
Testing file I/O with /test.txt
- writing****************************************************************
 - 1048576 bytes written in 12993 ms
- reading++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
- 1048576 bytes read in 127 ms
Free space:    2752512
Deleting file: /test.txt
- file deleted
Test complete
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2022-05-20:
Hi @frankcohen,

tested on ESP32-S3 DevKitC-1 v1.6 with no issues. Everything working as expected :)
Just removed "10" from FFatBegin(10) and in Arduino IDE selected partition scheme: Default 4MB with FFAT.

This is my debug message:
```
Total space:    1458176
Free space:    1458176
Listing directory: /
Writing file: /hello.txt
- file written
Appending to file: /hello.txt
message appended
Reading file: /hello.txt
- read from file:
Hello World!
Renaming file /hello.txt to /foo.txt
- file renamed
Reading file: /foo.txt
- read from file:
Hello World!
[  5809][E][vfs_api.cpp:104] open(): /ffat/hello.txt does not exist, no permits for creation
hello.txt does not exist
foo.txt exists
Listing directory: /
FILE: foo.txt	SIZE: 14
Deleting file: /foo.txt
- file deleted
Testing file I/O with /test.txt
- writing****************************************************************
- 1048576 bytes written in 10607 ms
- reading++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
- 1048576 bytes read in 124 ms
Free space:     409600
Deleting file: /test.txt
- file deleted
Test complete
```

frankcohen, 2022-05-20:
Thanks for testing it Prochy. It might just be my noob-y-ness. For example, the begin(10) is because I was expecting I could use FFat to access an SD card with GPIO 10 for its chip select on SPI. I see now that begin(true/false) changes the format of the mounted drive. I retested using the same partition scheme and get the same results. -Frank

closed 2022-05-23 by P-R-O-C-H-Y

ioviedodev, 2024-08-27:
Still continue in arduino 3.0.4 this issue