espressif/arduino-esp32 #6750: Services are duplicating when 16-bit UUID Number is used for a service

aarbmx6s, 2022-05-15:
### Board

DOIT ESP32 DEVKIT V1

### Device Description

DOIT ESP32 DEVKIT V1

### Hardware Configuration

Nothing attached

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

macOS 12.3.1

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hi there!

I am faced with the situation when service UUID is a 16-bit UUID Number taken from the official specifications document, for example, `0x180D` (`"180D"`), and the LightBlue app shows that peripheral has 2 services:

[image]

Inside the peripheral device in the advertisement section UUIDs of service are displayed twice:

[image]

[I used the next example ](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_server/BLE_server.ino) where I replaced UUIDs on my own.

When service UUID is random (like in the example) everything is OK.

### Sketch

```cpp
/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
    updates by chegewara
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "180D"
#define CHARACTERISTIC_UUID "2A19"

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

podaen, 2022-05-16:
Don't use those, generate your own uuid.

https://www.uuidgenerator.net/

VojtechBartoska, 2022-08-23:
is this still valid @aarbmx6s?

aarbmx6s, 2022-08-24:
@VojtechBartoska I was playing around with the different configurations and found that the application Light Blue on macOS and iOS sometimes showed even very old UUID with old descriptions. I may suppose it is may a problem with cashing devices/services in the  bluetooth of the device.

lyusupov, 2023-05-16:
> is this still valid ?

the issue **still remain in 2.0.9**

[image]
[image]

**16-bit UUIDs are the only affected**. ( 128 bit UUIDs are doing good )

lucasssvaz, 2023-09-25:
I can confirm that this still happens in version 2.0.12. I'll take a look to see if I can find the cause of the problem.

SuGlider, 2023-09-26:
In order to have both 16-bit UUIDs listed only one time:

``` cpp
void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();

  //=======>>>> replacing this part of the code:
  //pAdvertising->addServiceUUID(SERVICE_UUID);
  BLEAdvertisementData BLEAdvData;
  BLEAdvData.setCompleteServices(BLEUUID(SERVICE_UUID));
  pAdvertising->setAdvertisementData(BLEAdvData);
  
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}
```

lyusupov, 2023-09-26:
##  Before

```
      pAdvertising->addServiceUUID(BLEUUID(UART_SERVICE_UUID16)); // 0xFFE0
      pAdvertising->addServiceUUID(BLEUUID(UUID16_SVC_BATTERY));  // 0x180F
      pAdvertising->addServiceUUID(BLEUUID(MIDI_SERVICE_UUID));   // 0x00C7C4...
```

## After

```
      BLEAdvData.setCompleteServices(BLEUUID(UART_SERVICE_UUID16)); // 0xFFE0
      BLEAdvData.setCompleteServices(BLEUUID(UUID16_SVC_BATTERY));  // 0x180F
      BLEAdvData.setCompleteServices(BLEUUID(MIDI_SERVICE_UUID));   // 0x00C7C4...
```

## Results

[image]

SuGlider, 2023-09-26:
@lyusupov 
 For your case I'd suggest this workaround:
 
 ```cpp
  //=======>>>> replacing this part of the code:
  //pAdvertising->addServiceUUID(SERVICE_UUID);
  
  uint8_t advData[] = {
    //  Complete List of two 16-bit UUIDs 0xFFE0 0x180F
    0x05, 0x03, 0xE0, 0xFF, 0x0F, 0x18, // two 16bits UUIDs
    // Complete List of 128bits UUID 03B80E5A-EDE8-4B33-A751-6CE34EC4C700
    0x11, 0x07, 
    0x00, 0xC7, 0xC4, 0x4E, 0xE3, 0x6C, 0x51, 0xA7, 
    0x33, 0x4B, 0xE8, 0xED, 0x5A, 0x0E, 0xB8, 0x03,
  };

  BLEAdvertisementData BLEAdvData;
  BLEAdvData.setFlags(0x06);
  BLEAdvData.addData(std::string((const char*)advData, sizeof(advData)));
  BLEAdvData.setName("SoftRF-5f5cb0-LE");
  pAdvertising->setAdvertisementData(BLEAdvData);
  BLEDevice::setPower(ESP_PWR_LVL_P3);
 ```

SuGlider, 2023-09-26:
@lyusupov - Does it solve your need?

lyusupov, 2023-09-27:
@SuGlider 
Yes, explicit
```
BLEAdvData.setFlags(0x06);
```
gives positive effect necessary.
Thanks for the tip!

SuGlider, 2023-09-27:
@aarbmx6s @lyusupov @lucasssvaz 

Reading the code from Arduino and from IDF examples, I have concluded that, by default, BLE will send Advertising Data with UUID information and also send the Advertising Response with the same information.

BLE 4.2 has about 62 bytes for advertising information.
31 bytes for the main Advertising data block and another 31 for the Advertising Response data block. 
**Most BLE scanning tools will show both data block as one, creating an ilusion that it is duplicated, but actually, it is not.**

In order to "do not duplicate information", it is necessary to add (or replace) this line:
` pAdvertising->setScanResponse(true); // this is the dafault value` by 
` pAdvertising->setScanResponse(false);`

Doing it, the UUID will show up just once in both Advertising Data blocks.
"The duplication" doesn't create any problem for the central BLE device.
The user can also create a customized Advertising Data and/or Advertising Response Data, to add more information, like a more UUIDs or Manufacturer data, for instance, .

Therefore, there is no issue here to be solved.

lucasssvaz, 2023-09-27:
@SuGlider That's great news. Thanks for the investigation!

closed 2023-09-27 by lucasssvaz

trullock, 2024-01-10:
@SuGlider Am I missing something here? `pAdvertising->setScanResponse(false)` makes the UUID not show up at all in the scan response. If you pass `true` it shows up twice as above...