espressif/arduino-esp32 #6751: ESP Core 1.0.6 fails on Arduino IDE

MikeyMoMo, 2022-05-16:
### Board

ESP32

### Device Description

I recently upgraded to ESP32 1.0.6. All went well. Later, I tried to compile. That went well but the upload stopped with the following message:

```
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\Mikey\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/partitions/boot_app0.bin'
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\Mikey\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/partitions/boot_app0.bin'
```

Yes, it came out twice. Note that it is referencing 1.0.5. NOT 1.0.6 and, therefore, the file cannot be found since 1.0.5 is deleted during the installation of 1.0.6.

I removed 1.0.6 and went back to 1.0.5 and can compile and load once again. Please fix this for all of us.

### Hardware Configuration

Does not matter.  Would not finish compile.

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

n/a

### PSRAM enabled

no

### Upload speed

n/a

### Description

I expect the compile to finish!

### Sketch

```cpp
All sketches fail while on 1.0.6
```

### Debug Message

```plain
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\Mikey\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/partitions/boot_app0.bin
```

### Other Steps to Reproduce

esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\Mikey\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/partitions/boot_app0.bin

Note reference to 1.0.5 in a 1.0.6 compile script from ESP.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-05-16:
It sounds like your upgrade did not complete.  Check the [installation instructions](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) to make sure you have the correct URL in boards manager.

VojtechBartoska, 2022-05-16:
Also keep it mind 1.0.6 is an old version which is not longer supported. I recommend to move to v2.0.3 which includes a lot of fixes and improvements comparing to old 1.0.6.

podaen, 2022-05-16:
I think your antivirus is blocking the download.

VojtechBartoska, 2022-05-17:
Any updates @MikeyMoMo?

MikeyMoMo, 2022-06-12:
My IDE never offered to upgrade to 2.x.  When I did that, the issue went away.  Thanks for letting me know about this so I could do the manual update.

closed 2022-06-12 by MikeyMoMo