espressif/arduino-esp32 #6759: WebServer query params return wrong args count

tsctrl, 2022-05-16:
### Board

ESP32 Dev Module

### Device Description

DevKitC

### Hardware Configuration

-

### Version

v2.0.3

### IDE Name

IDF Component, Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

WebServer args() return wrong count if params is without equal= value 

to reproduce use uri/?query as url to handle

refer example sketch as below. is this is valid result? opposite with
asyncwebserver libs ESPAsyncWebServer request.params() count without equal= after ? as +1

### Sketch

```cpp
sketch:
WebServer server(80); 
server.begin()
server.on("", HTTP_GET, handle);
server.on("/", HTTP_GET, handle);

in handle:
url:http://192.168.4.1/?getnetwork
server.args() = 0 (wrong count)

url:http://192.168.4.1/?getnetwork=1&debug=true
server.args() = 2 (correct)

url:http://192.168.4.1/?getnetwork&debug=true
server.args() = 1 (wrong count)
```

### Debug Message

```plain
as above sketch
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

PilnyTomas, 2022-05-19:
Hi @tsctrl, please modify the sketch to be complete - i.e. when I copy-paste it into Arduino IDE it will compile.

tsctrl, 2022-05-21:
edited as per request and will compile

tsctrl, 2022-05-24:
updated with a complete and minimal sketch

PilnyTomas, 2022-05-25:
Can you please explain how did you manage to get working ETHernet on a bare ESP32 DevKitC?
I also don't understand why you mix WiFi and Ethernet.

> **Please provide a complete and minimal sketch that demonstrates the issue.** A modified example from the library can be used too.
> 
> Complete = compiles and starts properly after copy-paste (only change needed in code relates to SSID+PWD)
> Minimal = only code needed to demonstrate the issue - issue demonstration does not require the rest of the system for which it is used.

Please provide an example that is able to reproduce your issue with all other needed steps to reproduce.

SuGlider, 2022-05-25:
The real question of this issue is:

```
refer example sketch as below. is this is valid result? opposite with
ESPAsyncWebServer libs request.params() count without equal= after ? as +1

example with WebServer:

url:http://192.168.4.1/?getnetwork
server.args() = 0 (wrong count)

url:http://192.168.4.1/?getnetwork=1&debug=true
server.args() = 2 (correct)

url:http://192.168.4.1/?getnetwork&debug=true
server.args() = 1 (wrong count)
```

This related to `WebServer Library` in `Parsing.cpp` -- it doesn't matter if it is using ETH or WiFi.

PilnyTomas, 2022-05-26:
@SuGlider, I created my own code and if the argument has an equal sign it works ok every time. I think this problem is due to poor documentation - users don't know what to expect.

SuGlider, 2022-05-26:
The URL may have a Query string which starts after "?" symbol as this general format:
`scheme://username:password@subdomain.domain.tld:port/path/file-name.suffix?query-string#hash`

The syntax of the Query is not specified in the [RFC 3986](https://www.rfc-editor.org/rfc/rfc3986#section-3.4).
But it is recommended (not mandatory) that Query is formed by pairs of `key=value` separated by "&".

The question here is if the `WebServer class` shall consider a Query  expression with no value or with no "=" symbol as valid and meaningful.
It has to do with parsing of the URL and what the server side considers as valid or not.  

For instance: 
`http://192.168.1.1/page?k1&k2=v2&k3=&k4&k5=v5` is a valid URL.
Query string is `k1&k2=v2&k3=&k4&k5=v5` and should split into these **5 arguments**:

- k1 = ""
- k2 = v2
- k3 = ""
- k4 = ""
- k5 = v5

So, I think that we have an issue here, as presented by @tsctrl

tsctrl, 2022-05-26:
thanks @PilnyTomas , @SuGlider 

webserver perquisite a value for a key to count it as a valid args.
is not an issue, but it is convenient to have a key without value as it is meaningful to execute function by just validating the key itself and client did not have to hard follow the key value format.

personally, url?query&example looks and sound better than adding url?query=""&example="" in the browser address bar.

PilnyTomas, 2022-05-27:
Hi, @tsctrl I agree. We might add support for an empty parameter in the future, but I don't want to promise anything right now. We will discuss it internally...

VojtechBartoska, 2022-05-27:
Adding feature request label, we will evaluate this soon.

VojtechBartoska, 2022-06-14:
Other option to be consider is to refactor whole WebServer to use ESP-IDF solution.

VojtechBartoska, 2022-06-15:
Removing this issue from 2.0.4 and adding it under 2.1.0 milestone as WebServer needs more investigation.

PepeTheFroggie, 2023-01-11:
Same issue here. No = means no arg.
On the esp8266 this works perfectly. 
When transfering esp8266 code to esp32 this needs 2 hours debugging :(

cor-of-org, 2023-01-27:
Ow! I just spent the last half hour trying to figure out wtf was going on here. I want to send a simple /url?SomeCommand, which is parsed elsehwere (and used by serial connexion). I figured out I need to send command=value and then conctenate them back together for parsing! lol. Can we please get this working!

PepeTheFroggie, 2023-09-01:
This has still not been resolved. Need help?
Ran into the same problem converting esp8266 code to esp32-c3 !

unimo-chan, 2023-10-02:
Hi, I could probably fix it with the code below.

See WebServer::_parseArguments(String data)  (in Parsing.cpp)

rewrite
> if ((equal_sign_index == -1) || ((equal_sign_index > next_arg_index) && (next_arg_index != -1))) {
to
> if ((equal_sign_index > next_arg_index) && (next_arg_index != -1)) {

And add below code under  " RequestArgument& arg = _currentArgs[iarg]; "
//-----  add code ---- 
if(equal_sign_index == -1){
  arg.value = "";
  ++iarg;
  if (next_arg_index == -1){
    arg.key = urlDecode(data.substring(pos, (int)data.length()));
    break;
  }else{
    arg.key = urlDecode(data.substring(pos,  next_arg_index));
    pos = next_arg_index + 1;
    continue;
  }
}
 //---------

unimo-chan, 2023-10-02:
Hi, I could probably fix it with the code below.
I use 1.0.6.

See WebServer::_parseArguments(String data) (in Parsing.cpp)

rewrite
>
    if ((equal_sign_index == -1) || ((equal_sign_index > next_arg_index) && (next_arg_index != -1))) {
      log_e("arg missing value: %d", iarg);
      if (next_arg_index == -1)
        break;
      pos = next_arg_index + 1;
      continue;
    }
    RequestArgument& arg = _currentArgs[iarg];
>

to

>
    RequestArgument& arg = _currentArgs[iarg];

    if ((equal_sign_index == -1) || ((equal_sign_index > next_arg_index) && (next_arg_index != -1))) {
      log_e("arg missing value: %d", iarg);
      arg.value = "";
      ++iarg;
      if (next_arg_index == -1){
        arg.key = urlDecode(data.substring(pos, (int)data.length()));
        break;
      }
      arg.key = urlDecode(data.substring(pos,  next_arg_index));
      pos = next_arg_index + 1;
      continue;
    }
>

closed 2025-01-16 by Parsaabasi