espressif/arduino-esp32 #6760: IPADDR_NONE conflicting defines

mrengineer7777, 2022-05-16:
### Board

Adafruit Feather ESP32

### Device Description

ESP32-C3-DevKitC-02

### Hardware Configuration

None

### Version

v2.0.3

### IDE Name

VSCODE w/ PIO

### Operating System

Win10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I have run into an issue while creating a new "PingClass" library to replace the deprecated ping.h and esp_ping.h libraries.

This issue is mentioned in issue #6247, #6610.  Even when applying the fix in #6610 to my copy of IPAddress.cpp/h, I still get the error seen below.
 
I have resolved the issue in my class by dropping all references to String and IPAddress, and just using C strings.  However, I believe the definitions still conflict. Definitions:

IPAddress.cpp: `IPAddress INADDR_NONE(0, 0, 0, 0);`
IPAddress.h: `extern IPAddress INADDR_NONE;`
inet.h: `#define INADDR_NONE         IPADDR_NONE`
ip4_addr.h: `#define IPADDR_NONE         ((u32_t)0xffffffffUL)`

I believe `INADDR_NONE` expands in IPAddress.h as: `IPAddress(0,0,0,0)`, which works with the constructor.  But if inet.h is included it may expand as:
`((u32_t)0xffffffffUL)(0, 0, 0, 0)`.

Note also that 0.0.0.0 is a completely different IP than IPADDR_NONE=255.255.255.255.

### Sketch

```cpp
Not available. May repro later.
```

### Debug Message

```plain
In file included from C:/Users/David/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/ip_addr.h:43,
                 from C:/Users/David/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/inet.h:45,
                 from lib/PingClass/PingClass.h:24,
                 from lib/PingClass/PingClass.cpp:1:
C:/Users/David/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/ip4_addr.h:63:37: error: expected ')' before numeric constant
 #define IPADDR_NONE         ((u32_t)0xffffffffUL)
                             ~       ^~~~~~~~~~~~
C:/Users/David/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/inet.h:71:29: note: in expansion of macro 'IPADDR_NONE'
 #define INADDR_NONE         IPADDR_NONE
                             ^~~~~~~~~~~
C:/Users/David/.platformio/packages/framework-arduinoespressif32/cores/esp32/IPAddress.h:95:18: note: in expansion of macro 'INADDR_NONE'
 extern IPAddress INADDR_NONE;
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-05-16:
PR 6659 notes:
`This PR only works when <wifi-provisioning/wifi_config.h> has a modification to include <lwip/ip4_addr.h> instead of <lwip/inet.h>. This will be done directly to the sdk folder in the github structure and it has been fixed in IDF by a separated Merge Request. This will be reflected in the future, for good.`

Hopefully this will be fixed in the IDF.

SuGlider, 2022-05-16:
@mrengineer7777 - 
The fix actually isn't limited to just apply #6659 to  your copy of IPAddress.cpp/h.
It is necessary to change the all header files as listed in the PR:
- libraries/AsyncUDP/src/AsyncUDP.h 
- libraries/WiFi/src/WiFiServer.h
- tools/sdk/esp32c3/include/wifi_provisioning/include/wifi_provisioning/wifi_config.h

SuGlider, 2022-05-16:
> Hopefully this will be fixed in the IDF.

It is already fixed in IDF, but is has not been backported to the 4.4. yet.

SuGlider, 2022-05-16:
You can manually apply all the necessary changes to your local files - it seems you use PIO. 
I guess that PIO will launch an official version with the Core 2.0.3 in the near future and the issue will be solved.

Using Arduino IDE and updating it to 2.0.3 works fine for this issue, because it already has the all those header files changes commited to the main GitHub repository.

mrengineer7777, 2022-05-16:
@SuGlider Thanks for pointing out the modifications to the other files.  I just made those changes and I'm still getting a compile error if I include "IPAddress.h".  I'll do some more digging and see if I can create a simple sketch to repro the issue.

I will note "lwip/inet.h" is still referenced in multiple files, including "netdb.h", which I use to convert an IP string to ip_addr_t.

mrengineer7777, 2022-05-16:
See updated sketch above which reproduces the issue.  Note that it doesn't happen if `#include "Arduino.h" is called first, but the library I'm compiling doesn't include Arduino.h (by design), so it exhibits the same issue.

Also note the IP lookup likely won't work without WiFi being up.  We are just testing for a successful compile here.

SuGlider, 2022-05-16:
The PR#6659 solves the INADDR_NONE for the **ESP32 Arduino framework**.
But if you create a sketch that includess and uses IDF directly into it, then you must take care for which files you include and the order it is done...

There is nothing that we can do to fix any combination of Arduino/IDF users may want to create.
We have already an internal discussion about INADDR_NONE and it won't be removed as it is part of UpStream Arduino.

SuGlider, 2022-05-16:
> See updated sketch above which reproduces the issue. Note that it doesn't happen if `#include "Arduino.h" is called first, but the library I'm compiling doesn't include Arduino.h (by design), so it [...]

This is not a pure Arduino Sketch... it is a mix of  IDF LWIP and Arduino... which only needs IPAddress Class for a log_v().
As I said, I see no real Arduino issue here.

mrengineer7777, 2022-05-17:
> The PR#6659 solves the INADDR_NONE for the **ESP32 Arduino framework**. But if you create a sketch that includes and uses IDF directly into it, then you must take care for which files you include and the order it is done...

Fair point.  I'm creating an Arduino library but it does need some IDF calls.  That makes me wonder how esp_ping.c avoided the URL lookup issue.  Will look into it tomorrow.

>  INADDR_NONE ... won't be removed as it is part of UpStream Arduino.

Accepted.  I will close out this issue.  Thanks for the clarifications!

closed 2022-05-17 by mrengineer7777

Tianbawen, 2024-02-29:
> 请参阅上面更新的草图，其中重现了该问题。请注意，如果首先调用“#include ”Arduino.h“，则不会发生这种情况，但我正在编译的库不包括 Arduino.h（按设计），因此它表现出相同的问题。
> 
> 另请注意，如果没有 WiFi 启动，IP 查找可能无法正常工作。我们只是在这里测试成功的编译。

> 请参阅上面更新的草图，其中重现了该问题。请注意，如果首先调用“#include ”Arduino.h“，则不会发生这种情况，但我正在编译的库不包括 Arduino.h（按设计），因此它表现出相同的问题。
> 
> 另请注意，如果没有 WiFi 启动，IP 查找可能无法正常工作。我们只是在这里测试成功的编译。
It really helps to call the "#" include "Arduino. H" first

Tianbawen, 2024-02-29:
> See updated sketch above which reproduces the issue. Note that it doesn't happen if `#include "Arduino.h" is called first, but the library I'm compiling doesn't include Arduino.h (by design), so it [...]

It really helps to call the #include "Arduino. h" first