espressif/arduino-esp32 #6762: esp32-s3 does not reset after upload

prplz, 2022-05-17:
### Board

esp32s3

### Device Description

esp32s3 devkitc n8r8

### Hardware Configuration

Native USB port plugged into computer, serial convert USB port not plugged in.

### Version

v2.0.3

### IDE Name

Arduino IDE, Platformio

### Operating System

Windows 10, MacOS

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Uploading a sketch over usb otg should reset the chip after upload, but it does not.

### Sketch

```cpp
NA
```

### Debug Message

```plain
NA
```

### Other Steps to Reproduce

- Select USB mode = USB-OTG, USB CDC on boot = enabled, Upload mode = USB OTG CDC.
- Upload sketch
- Board correctly resets into download mode using 1200 baud and uploads the sketch
- Board does not reset and run the sketch as it should, instead staying in download mode. Here you have to press the reset button manually. On S2 the board resets and runs the sketch after upload with no manual intervention.

Tested on s3 devkitc and all UnexpectedMaker s3 boards.

Notes:
- No esptool commands reset the board, on the s2 it does reset
- Esptool doesn't detect when the board was put into bootloader using IO0 as it does on s2 (WARNING: ESP32-S2FNR2 chip was placed into download mode using GPIO0)
- USB mode and upload mode set to hardware CDC resets as expected

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

prplz, 2022-05-17:
cc @UnexpectedMaker

UnexpectedMaker, 2022-05-17:
@me-no-dev I assume this is not the intended behaviour? And the S3 boards should reset after flashing them like the S2 boards do right?

me-no-dev, 2022-05-17:
@UnexpectedMaker not if you are using TinyUSB while booted. It works if the integrated CDC+JTAG is used, but the other USB no.

UnexpectedMaker, 2022-05-17:
> @UnexpectedMaker not if you are using TinyUSB while booted. It works if the integrated CDC+JTAG is used, but the other USB no.

Hmm, but it works correctly with the S2, so why not with the S3? It's really confusing people - they think their boards didn't flash, as no other ESP32 has this issue (original, S2, C3 etc) - so people are complaining about HW faults.

If we use CDC+JTAG, then the board doesn't enumerate with the custom device name, which also confuses people, so it seems like the S3 has the worst of both situations, rather than any improvement in either.

How do we get this behaviour changed?

sukesh-ak, 2022-05-17:
Yes this has become a painful issue. Have tested from Arduino and ESP-IDF with same behaviour.

This also means you have to reconnect serial monitor after every firmware upload for proper update.

VojtechBartoska, 2022-05-17:
This seems worth improving and documenting as well? cc @me-no-dev & @pedrominatel

me-no-dev, 2022-05-17:
> How do we get this behaviour changed?

Surely if we could we would have :) I went through ROM code and everything, but could not figure out what is preventing the board/download mode to reboot the firmware. When booting into download from TinyUSB, we first stop TinyUSB, then init CDC+JTAG and then reboot into download, so in reality it should have been the same as rebooting from CDC+JTAG, but it is not. We might find a way in the future, but this has taken too much resources already as it is.

frankcohen, 2022-05-20:
This is pain, I'm having to use the Boot pin after every upload. And that has the side effect of closing Arduino IDE Serial Monitor randomly.

I'm glad to volunteer my hours with direction. Just tell me what to do.

Background: I'm working on the open-source Reflections wrist watch project, including an ESP32-S3-Mini based main board. I'm building this using the Adafruit Feather ESP32-S3 as a reference and using their Arduino IDE 12.3.1 board definition "Adafruit Feather ESP32-S3 No PSRAM". I'm using ESP32 version 2.0.3, installed through the Boards Manager. Repository for my project is at https://github.com/frankcohen/ReflectionsOS/tree/main/Devices/Hoober

sukesh-ak, 2022-05-21:
@frankcohen 
As a side note you should check out UM boards for S3.
https://unexpectedmaker.com/shop

frankcohen, 2022-05-21:
Thanks, the UM boards look great. I'd be glad for UM to carry my board too. I will open-source the board. -Frank

boberace, 2022-06-04:
Not sure if this is helpful but ... My ESP32-S3-DevKitC-1-N8R2 does reset after upload if I use the UART USB jack but it will not reset after upload when using the OTG USB jack UNLESS I upload right after I erase flash.

My UM Feather S2 does reset after upload. My UM TinyS3 does not reset after upload.

UnexpectedMaker, 2022-06-06:
@me-no-dev Is this issue caused by security advisory AR2022-004 ?

me-no-dev, 2022-06-06:
@UnexpectedMaker yes and no. You can think of this as two separate issues:
1. We can not use the USB-OTG in download mode (so we switch to the integrated CDC+JTAG)
2. When we switch to CDC+JTAG and reboot into Download Mode, chip is not reset after the new firmware is flashed (what we are discussing in this thread)

AR2022-004 talks about issue 1

UnexpectedMaker, 2022-06-06:
> 

So if issue 1 is resolved, then you could use USB-OTG and not have to switch to CDC, bypassing issue 2? So that would also fix this problem on newer silicon?

me-no-dev, 2022-06-06:
@UnexpectedMaker exactly :)

egnor, 2022-07-13:
Can people check my understanding of this (rather irksome) issue? Apologies if I'm covering obvious ground here! I'll edit with any corrections.

### Background

There are multiple ways to connect from a a host (Mac/PC) USB port to an ESP32-S3 for bootloading and serial terminal purposes:
1. Use a physical USB-to-serial converter chip/module, connected to the ESP32-S3's serial TX and RX lines
2. Use the ESP32S3's onboard USB controller, wire the USB port directly to the ESP32-S3.

If using option (1) there is no problem. (For example this is the UART USB jack on the ESP32-S3 DevKit board.) However, that requires an extra part and it's otherwise nice to use the integrated USB controller. So, let's assume option (2). This requires configuring the ESP32-S3 to act as a serial port device. There are again multiple ways to do that:
<ol type="a">
  <li><a href=https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/usb-otg-console.html>Use the ESP32-S3's internal USB-OTG (On The Go) peripheral</a>, which supports acting as a general purpose USB host or device (the bidirectionality is what makes it "On The Go"); in software, set the peripheral up to act as a USB device, and implement the USB CDC ("Communications Device Class", aka serial port) protocol, and thus the device shows up as a serial port to the host.</li>
  <li><a href=https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/usb-serial-jtag-console.html>Use the ESP32-S3's internal USB Serial/JTAG peripheral</a>, which is a separate special purpose USB controller that only supports USB CDC (serial port) and JTAG (debugging port, not important here) protocols.</li>
</ol>

Using the dedicated USB Serial/JTAG controller is simpler for software, since the hardware takes care of most of the details of USB. However, the USB-OTG controller is more general, allowing other device classes (such as mass storage) in addition to being a serial port.

Importantly, the bootloader (used to upload code) and the application (which wants to `Serial.print` to a console) might make different choices about which controller to use. The Arduino platform, for example, tends to use <a href=https://github.com/hathach/tinyusb>TinyUSB</a>, a software USB stack that expects a fully general USB controller, which would imply the USB-OTG controller.

### Issues

So far this is all lovely but now there are some wrinkles.

[As @me-no-dev noted above](https://github.com/espressif/arduino-esp32/issues/6762#issuecomment-1147409509), the bootloader cannot use the USB-OTG controller. This is related to a spot of confusion over one-time-programmable EFUSE bits as described in [security advisory AR2022](https://www.espressif.com/sites/default/files/advisory_downloads/AR2022-004%20Security%20Advisory%20for%20USB_OTG%20%26%20USB_Serial_JTAG%20Download%20Functions%20of%20ESP32-S3%20Series%20Products%20EN.pdf). Basically, for many ESP32-S3 parts, bootloader USB-OTG access is permanently disabled.

So, the bootloader uses the USB Serial/JTAG controller instead (if allowed). This works just fine and allows the flash to be programmed in the normal way.

HOWEVER, if the application uses the USB-OTG controller (as the Arduino platform does by default, using TinyUSB), then the application will not work after the bootloader finishes its work and attempts to run it. Per [@me-no-dev's comment above](https://github.com/espressif/arduino-esp32/issues/6762#issuecomment-1128621518), the reasons for this are not entirely understood, but it's something about switching between the two controllers.

For now, there are two known workarounds:
<ol type="i">
<li>After bootloading, use a full hardware reset (probably a button) to launch the app. This fully resets the board and skips the USB Serial/JTAG controller, avoiding issues with switching. However it's inconvenient, impossible without physical access, and disruptive to port connectivity.</li>
<li>In the application, use the USB Serial/JTAG controller instead of the USB-OTG controller. This limits flexibility (can't be a mass storage device at the same time) and makes the port show up as a generic "USB JTAG/serial debug unit" instead of the user's preferred description.</li>
</ol>

Both workarounds are unfortunate, which is why this bug exists and people are grumpy.

Note, I have personally not yet been able to implement workaround (ii), but haven't gone particularly far down that rabbit hole as of this writing. Per [@me-no-dev's comment above](https://github.com/espressif/arduino-esp32/issues/6762#issuecomment-1128553628) it should be possible to make this work.

egnor, 2022-07-13:
Further discoveries:

It's certainly easy to confirm that the ROM bootloader and also the RAM stub that esptool.py loads are using the USB Serial/JTAG controller. You can tell because it shows up as "Espressif USB JTAG/serial debug unit" with VID:PID 303A:1001. (That should probably get added to Arduino and platformio's list of detected VID:PID values.)

After programming, `esptool.py` (with the default `--after=hard_reset`) toggles RTS (and DTR) lines to reset the chip and run the app (the thing that isn't working). This is the famous "Hard resetting via RTS pin..." step.

On the device side, reset-on-DTR/RTS-wiggle is managed directly in the hardware but the USB Serial/JTAG peripheral. This is described in section 30.3.2 in the technical reference:
> [image]

Here is the further discussion in section 30.4.2:
> [image]

The [hard reset code in esptool.py](https://github.com/espressif/esptool/blob/65f861420515217f7b5d209fce288808dc5b1943/esptool/targets/esp32s3.py#L202) does this, assuming DTR=0 and RTS=0 on entry:

```
    def hard_reset(self):
        if self.uses_usb():
            self._check_if_can_reset()

        print("Hard resetting via RTS pin...")
        self._setRTS(True)  # EN->LOW
        if self.uses_usb():
            # Give the chip some time to come out of reset,
            # to be able to handle further DTR/RTS transitions
            time.sleep(0.2)
            self._setRTS(False)
            time.sleep(0.2)
        else:
            self._setRTS(False)
```

For what it's worth there is an issue in `esptool.py` where `self.uses_usb()` is returning false in this mode. The [implementation of `uses_usb()`](https://github.com/espressif/esptool/blob/65f861420515217f7b5d209fce288808dc5b1943/esptool/targets/esp32s3.py#L161) checks a magic value in the ROM's data segment (`UARTDEV_BUF_NO = 0x3FCEF14C`) to see if it indicates USB loading (`UARTDEV_BUF_NO_USB = 3`). However in my experiments this value is 4, not 3. This is for example why `_check_if_can_reset()` (checking for GPIO0 forced-bootloader mode that prevents resetting?) isn't working as expected on the ESP32-S3. BUT, fixing this to accept 4 to indicate USB mode has no effect, it still doesn't seem to reset, or at least run the app code.

I'm not entirely sure I understand @me-no-dev's comments, but I gather they think the problem is that the app code runs, but immediately crashes and resets back to the bootloader? I can see no evidence of a reset actually happening-- there's no USB reconnect, for example.

Next step is to verify the EFUSE settings and make sure there's nothing pre-burned that could be blocking the reset.

egnor, 2022-07-13:
Ok I've been able to get it to work with an IDF example app that only uses the USB Serial/JTAG controller. Steps:
1. Set up ESP-IDF, per [the installation instructions](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html)
2. In the `esp-idf/examples/system/console/basic`, run `idf.py set-target esp32s3 && idf.py menuconfig`. Enable `ESP_CONSOLE_USB_SERIAL_JTAG` in menuconfig.
3. Then run `idf.py flash monitor`

I see a successful flash, the usual `Hard resetting via RTS pin...`, then `Executing action: monitor`, and then after some chatter the example app's menus! (It thinks my terminal application doesn't support escape sequences but that's [discussed elsewhere](https://www.esp32.com/viewtopic.php?t=25981).) Control-] exits the monitor.

Unfortunately `ESP_CONSOLE_USB_SERIAL_CDC` does not work, so I can't try that:

```
/home/egnor/source/esp-idf/components/esp_system/port/soc/esp32s3/usb_console.c:69:1: error: static assertion failed: "usb_osglue_*_int is not multicore capable"
   69 | _Static_assert(SOC_CPU_CORES_NUM == 1, "usb_osglue_*_int is not multicore capable");
      | ^~~~~~~~~~~~~~
[27/615] Building C object esp-idf/hal...s/__idf_hal.dir/spi_slave_hd_hal.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

I can't get the an Arduino sketch to work regardless, even if I never `Serial.begin()` in the sketch. Maybe in startup code it's still setting up TinyUSB.

frankcohen, 2022-07-14:
Just a note of appreciation to Egnor, thank you. And a question, have you seen JTAG with platform IO and esp32s3 working?

egnor, 2022-07-14:
@frankcohen wrote:
> Just a note of appreciation to Egnor, thank you. And a question, have you seen JTAG with platform IO and esp32s3 working?

I have not tried! I am curious about it but figured I'd open one can of worms at a time. If you do try, or have tried, I'd like to know how it goes.

egnor, 2022-07-14:
Further notes to self-- if I use miniterm, I can toggle DTR and RTS myself (with ^T^D and ^T^R respectively) and attempt to generate that reset sequence, I do get some log output.

After a successful flash or an Arduino app that failed to start after flashing:

```
% python -m serial.tools.miniterm /dev/ttyACM0 115200
--- Miniterm on /dev/ttyACM0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
--- RTS inactive ---
--- DTR inactive ---
--- RTS active ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a76
waiting for download
```

So it did reboot, but the `waiting for download` indicates it ended up back in the bootloader. It's as if the "download flag" was not successfully reset during the RTS=0 DTR=0 point. Notably, if I go through the other sequence to explicitly _set_ the bootloader flag, I get identical output:

```
--- RTS inactive ---
--- DTR active ---
--- RTS active ---
--- DTR inactive ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a76
waiting for download
```

If instead I flash the `esp-idf/examples/system/console/basic` project (configured for `ESP_CONSOLE_USB_SERIAL_JTAG`), then both of the USB/Serial controller's DTR/RTS wiggle sequences work correctly. Going from bootloader to app:

```
waiting for download
--- DTR inactive ---
--- RTS inactive ---
--- RTS active ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400428a0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x113c
load:0x403c9700,len:0xa4c
load:0x403cc700,len:0x2af0
entry 0x403c9898
 ... blah blah blah ...
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
Your terminal application does not support escape sequences.

Line editing and history features are disabled.
On Windows, try using Putty instead.
esp32s3>
```

Going from app back to bootloader:

```
esp32s3> --- RTS inactive ---
--- DTR active ---
--- RTS active ---
--- DTR inactive ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
Saved PC:0x4037c14a
waiting for download
```

Here, when booting into the app, we get `rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)` instead of `rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))`.

Moreover, another observation! The success/failure behavior seems to be defined not by what app we're _loading_ but by what app was running _before_ resetting to the bootloader:

- flash Arduino, reset to run, flash console/basic => loads but does NOT run
- reset to start console/basic, flash console/basic again => loads and DOES run
- flash Arduino => loads and DOES run right away!
- flash Arduino again => loads and does NOT run
- reset to run Arduino, flash Arduino again => loads and does NOT run
- reset to run Arduino, flash console/basic => loads but does NOT run
- reset to start console/basic, flash Arduino => loads and DOES run

So somehow, when resetting back to the bootloader, a running Arduino sketch (presumably with TinyUSB and USB-OTG) leaves some state in the system that prevents the bootloader from jumping to start ANY program it loads. Note that with a running Arduino sketch, it's up to the Arduino USB-OTG / TinyUSB / CDC-Serial implementation to detect a DTR/RTS wiggle and reboot into the bootloader. Maybe it's doing that in some slightly wrong way?

(I hope this is all amusing to any kibitzers...)

frankcohen, 2022-07-14:
I'll let you know. The bigger problem to solve first is getting Platform IO
to recognize the Adafruit Feather ESP32-S3 board definition file. That's
next.

egnor, 2022-07-14:
@frankcohen perhaps https://github.com/platformio/platform-espressif32/pull/851 will be of interest!

egnor, 2022-07-14:
Back to the bootloading issue, I was able to select "Hardware CDC and JTAG" in the Tools menu of the IDE to avoid the bootloader problem with the [workaround (ii) described above](https://github.com/espressif/arduino-esp32/issues/6762#issuecomment-1182821492):
[image]

That was embarrassingly simple, I think I missed it because I don't typically use the IDE. With `arduino-cli`, it's probably something like `--build-property=build.usb_mode=1`; with platformio, it's `-DARDUINO_USB_MODE=1`.

Sadly, with my fix(?) to USB mode detection in esptool.py, this now causes an error:

```
WARNING: ESP32-S3 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
*** [upload] Error 1
```

Fortunately that fix hasn't been accepted yet! But it looks like something's confused about the way it's parsing `GPIO_STRAPPING` and/or `RTC_CNTL_FORCE_DOWNLOAD_BOOT`. It's yaks all the way down...!

egnor, 2022-07-14:
About that: Apparently `GPIO_STRAPPING` gets zeroed out by an internal reset, including a reset triggered by the USB Serial/JTAG peripheral. That triggers the false locked-into-bootloader alert from esptool. (In actual fact if the warning is overridden esptool can reboot into app just fine in that scenario.)

Another method of software reboot-into-bootloader (specifically, setting `RTC_CNTL_FORCE_DOWNLOAD_BOOT`) is detected in esptool to avoid that false positive, but the USB Serial/JTAG peripheral doesn't use that mechanism. USB Serial triggered reboot does have a reason code in `RTC_CNTL_RTC_RESET_STATE_REG`, _but_ esptool.py's initial reset _always_ leaves that reason code in place, even when the system is locked into the bootloader. So that's not a useful discriminator.

With the USB detection broken (as it currently is at head), at least that check is disabled and didn't get in the way... I don't know how to fix it. I guess I'll just leave it.

UnexpectedMaker, 2022-07-14:
@egnor This is great investigative work Daniel! Thanks very much for the awesome effort.

Jason2866, 2022-07-14:
> I'll let you know. The bigger problem to solve first is getting Platform IO to recognize the Adafruit Feather ESP32-S3 board definition file. That's next.
> 
Thats not complicated. Just create a folder `boards` in the root folder of your project.
Add the boards.json manifest there. [Example](https://github.com/arendst/Tasmota/tree/development/boards)

egnor, 2022-07-14:
@Jason2866, are you able to repeatedly re-flash and re-start with equal success, without any button presses? (You only need to go through the cycle, like, twice to answer the question.)

I ask because, in my case at least, it's the firmware running _before_ the bootload attempt that determines success. So if the chip was running something that used the USB Serial/JTAG controller beforehand (or not using any USB at all), then the flash and reboot would in fact work just fine.

I can also try to repro myself with Tasmota! It's possible that Tasmota evades this problem in some other way, which would be a good point of comparison with the default arduino-esp32 environment.

[P.S. I removed a bunch of my previous notes and coalesced them into my summary above, for clarity]

Jason2866, 2022-07-14:
@egnor Will try to find the used device(s) (tomorrow to late now...).
I see no drawback to use the `USB Serial/JTAG controller` for flashing and using as console.

egnor, 2022-07-14:
Indeed, the workaround to use the USB Serial/JTAG controller instead of USB-OTG with a serial driver, is fine for some use cases. (Not for my use case in particular, as I wanted to do other things with the USB port besides console.) It would be my starting recommendation for people bumping into this.

Jason2866, 2022-07-15:
@frankcohen The "official" manifest is for the `Adafruit ESP32-S3 Feather with STEMMA QT / Qwiic - 8MB Flash No PSRAM` is with the knowledge from this thread not the optimal one. For easy flashing a change to use the USBserial/jtag is the way.
I dont get it what benefit of `tinyuf2` should be. Imho flash space waste.
Since many boards manifests are not complete (or wrong) or have unwanted add ons included, we do not use any "official" boards manifest for our project. We define what we need. Big advantage of Platformio, you dont have to use predefined stuff.

frankcohen, 2022-07-15:
Thanks for the tips Jason. I plan to create my own board definition. And
I'll contribute that under an open source license  For right now I'm using
the Adafruit board definition. Frank

Jason2866, 2022-07-15:
@egnor 
``` ioreg -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 27>
  +-o AppleT8103USBXHCI@01000000  <class AppleT8103USBXHCI, id 0x100000308, registered, matched, active, busy 0 (170 ms), $
  | +-o PSSD T7@01200000  <class IOUSBHostDevice, id 0x10000082b, registered, matched, active, busy 0 (157 ms), retain 173$
  +-o AppleT8103USBXHCI@00000000  <class AppleT8103USBXHCI, id 0x1000002ad, registered, matched, active, busy 0 (104789 ms$
    +-o HP USB-C Dock G5@00200000  <class IOUSBHostDevice, id 0x100000716, registered, matched, active, busy 0 (493 ms), r$
    | +-o USB5734@00230000  <class IOUSBHostDevice, id 0x10000076d, registered, matched, active, busy 0 (15 ms), retain 10$
    | +-o USB 10/100/1000 LAN@00240000  <class IOUSBHostDevice, id 0x1000007ca, registered, matched, active, busy 0 (478 m$
    +-o HP USB-C Dock G5@00100000  <class IOUSBHostDevice, id 0x100000747, registered, matched, active, busy 0 (104695 ms)$
      +-o USB Audio@00120000  <class IOUSBHostDevice, id 0x1000007a0, registered, matched, active, busy 0 (417 ms), retain$
      +-o HP USB-C Dock G5@00150000  <class IOUSBHostDevice, id 0x1000007b3, registered, matched, active, busy 0 (402 ms),$
      +-o USB2734@00130000  <class IOUSBHostDevice, id 0x1000283f6, registered, matched, active, busy 0 (692 ms), retain 3$
        +-o Jabra PRO 9470@00131000  <class IOUSBHostDevice, id 0x100028412, registered, matched, active, busy 0 (138 ms),$
        +-o USB 2.0 Hub [Safe]@00133000  <class IOUSBHostDevice, id 0x10002841d, registered, matched, active, busy 0 (345 $
        | +-o Storage Media@00133100  <class IOUSBHostDevice, id 0x100028456, registered, matched, active, busy 0 (77 ms),$
        | +-o IOUSBHostDevice@00133400  <class IOUSBHostDevice, id 0x100028461, registered, matched, active, busy 0 (258 m$
        | +-o USB Receiver@00133200  <class IOUSBHostDevice, id 0x10002846b, registered, matched, active, busy 0 (213 ms),$
        +-o USB JTAG/serial debug unit@00132000  <class IOUSBHostDevice, id 0x10002842b, registered, matched, active, busy$
```

VojtechBartoska, 2022-08-23:
Do you still face this @prplz on the master Branch?

prplz, 2022-08-26:
Still present at master (d22bcb7) with um_tinys3

taf2, 2022-10-01:
Wondering if anyone has um_tinys3 uploads working yet from platformio ?  If so what have you put into your platformio.ini file to get this working? @UnexpectedMaker love your boards but haven't yet been able to get a tinys3 working... hope you keep making the tinypico's

Jason2866, 2022-10-01:
@taf2 Use latest Platformio release. The board is there. So 
```[env:stable]
platform = espressif32 @ 5.2.0
board = um_tinys3
```

taf2, 2022-10-01:
@Jason2866 oh nice this appears to get it to start uploading but still failed for me at about 95% - this is an improvement!

Here's error details:
https://gist.github.com/taf2/cf0cc03d9be63b16cafcd378e9ae90e5

Here's project files:
https://github.com/taf2/tinypico-fridge-alarm

Jason2866, 2022-10-01:
@taf2 You have a problem with your USB port. Use a other USB Cable (shorter / better). If this does not help use a different USB port. This issue has nothing to do with the board nor Platformio.

taf2, 2022-10-01:
Ah you know what i think this is my platformio being out of date.

```
pio --version
PlatformIO Core, version 6.1.4
taf2~/work/arduino/tinypico-fridge-alarm (main) * > which pio
/usr/local/bin/pio
taf2~/work/arduino/tinypico-fridge-alarm (main) * > vi Makefile
taf2~/work/arduino/tinypico-fridge-alarm (main) * > ls -lh /usr/local/Cellar/platformio/
total 0
drwxr-xr-x  8 taf2  admin   256B Oct 20  2021 5.2.2
```
So i've been avoiding brew update (because  painful)  and just running pio upgrade... but it seems like that was not upgrading the actual core platform io files... so this means i had out of date files... 

however same issue after upgrading things... that said the cable does not seem to be the issue because I can use it just fine with the tinypico usbc version... i'll get another usbc cable and try on a different computer

Jason2866, 2022-10-01:
The cable can work with other boards, transfer speed and protocol is different when CDC USB modem is used.
For most Apple users having this problem a USB Hub does solve.

taf2, 2022-10-01:
Yeah, I am not much of a VSC fan... I am maybe too old and just like to use vim when/wherever possible (for better or worse) - this is definitely an older USBC cable got it from i think pixel 1 phone... will try on my newer macbook with usbc to usbc cable next and then a usbc hub.  Really appreciate the help in jumping in on this thread very amazing and very helpful thank you!

Jason2866, 2022-10-01:
Your welcome, please tell me your result.

taf2, 2022-10-01:
same issue with different usbc cable new one from apple store... i'll try a different tinys3 next and see if it's the board...

Jason2866, 2022-10-01:
As i wrote before, if it is not the cable it is a Mac related problem. I highly doubt the board is faulty.

taf2, 2022-10-01:
@Jason2866 yeah i would imagine it's Mac related as well... except i just got everything running on windows environment and have the exact same error with different USBC cable as well... 

i'll go to store tomorrow and buy a few different types of cables... but still unable to get tinys3 working where tinypico always has worked...

taf2, 2022-10-01:
I have 3 tinys3's and just got 1 to work... so definitely seems like 2 out 3 boards are bad... I'll take high resolution image of boards... but 1 board out of 3 worked same usbc cable on windows... i'll test on linux and mac tomorrow.

UnexpectedMaker, 2022-10-01:
> I have 3 tinys3's and just got 1 to work... so definitely seems like 2 out 3 boards are bad... I'll take high resolution image of boards... but 1 board out of 3 worked same usbc cable on windows... i'll test on linux and mac tomorrow.

Sorry, this cant be board specific - 2 bad boards out of 3. There is nothing that could be bad at the board level. It's native USB and internal Flash.  

What OS are you using?

taf2, 2022-10-01:
@UnexpectedMaker big fan so to clarify this is just looking for advice here... figured video to share for context is better then trying to explain via words... can provide as much details as needed here's unlisted yt videos:

1. https://youtu.be/uu4dDb_4XeY
2. https://youtu.be/ERkNJNQTYvk

HD revisions hopefully ready soon.  I have one more tinys3 board i can run through the same tests and share...

UnexpectedMaker, 2022-10-01:
But what you are doing wont work.....   
The boards that are not flashed, do not have the CDC running, so they have no USB serial device present to flash. You need to pout them into download mode first to flash them.

The board that is working, has the CDC running, so presents a USB serial device to be re-flashed.

Can you get on my discord server, for faster support iteration?

UnexpectedMaker, 2022-10-01:
On your Mac video, that 3rd board is not in download mode - it's looking for the wrong USB serial port - you can see that on the fort it's trying to flash. Put it into download mode, then try to flash it.

The 96%'er - use esptool to erase the flash, then try to flash it again.

UnexpectedMaker, 2022-10-01:
> But what you are doing wont work.....
> The boards that are not flashed, do not have the CDC running, so they have no USB serial device present to flash. You need > to put them into download mode first to flash them.
>
> The board that is working, has the CDC running, so presents a USB serial device to be re-flashed.

Ok, ignore the above as I can now see in the Mac video, 2 boards are enumerating in download mode, but your 3rd board (that failed in both) isn't, so you have to manually put it into download mode to flash it the first time.

Jason2866, 2022-10-02:
@UnexpectedMaker  With a fully erased flash it is no problem to flash via CDC.
There is no need for any driver on the device. Esptool.py puts the device in flashing mode (without the need of pressing a button) and flashes. Maybe this behaviour is not consistent with all OS. But for MacOS it does work this way. I have different S3 CDC devices and it works always. The S3 issue is that it is staying in bootloader mode and needs a manual reset to start.

Jason2866, 2022-10-02:
> Just a note of appreciation to Egnor, thank you. And a question, have you seen JTAG with platform IO and esp32s3 working?

OpenOCD via inbuilt USB/jtag does work.

UnexpectedMaker, 2022-10-02:
> @UnexpectedMaker With a fully erased flash it is no problem to flash via CDC.
> There is no need for any driver on the device. Esptool.py puts the device in flashing mode (without the need of pressing [...]

I know all of this...   it's not really relevant to what's happening here because the boards I ship are not erased, but come with CircitPython on them, and they cannot be flashed unless they are put into download mode first. 

So the OP has a working board, a board that fails to flash all of the way (that I asked him to first erase and then try again) and one that he has never erased or flashed that still has CP on it, and that one can't be flashed without putting it manually into download mode first.

Jason2866, 2022-10-02:
I always thought and told too that the boards are most probably fine. The reason is highly sure the macOS issue which i have linked. 

> @taf2 You have a problem with your USB port. Use a other USB Cable (shorter / better). If this does not help use a different USB port. This issue has nothing to do with the board nor Platformio.
> 
> You run probably in this issue [espressif/esptool#712](https://github.com/espressif/esptool/issues/712)
> 
> I do not run in this issue since the M1 Air is connected via a HP G5 USB-C Dockingstation.
>

taf2, 2022-10-02:
@UnexpectedMaker @Jason2866 great news!  

```
esptool.py  erase_flash
pio run -e um_tinys3 -j 2 --target upload
```
Fixed the issue on the 96% board!

Additionally, now that I understand the reset/boot button is needed I was able to get the 3rd board working as well so all 3 boards are working and just pulling out my 4th board and running through a series of reset/boot button presses I've now got all 4 boards working reliably with pio - big thanks for all the great work on this software and hardware.

UnexpectedMaker, 2022-10-02:
@taf2 thats fantastic news! Thanks for being patient enough to get through all of this 👌

josehdx, 2022-12-27:
Dear community.

Wondering if you can clarify the usb host (OTG) feature of some esp32-s3 boards. Im not by any stretch of the imagination and advance user of these kind of boards and i am very confused. I want to use the esp32-s3 usb host feature to receive midi messages from a usb midi controller and transmit them via Bluetooth low energy. If you are not familiarized with midi thats ok. I have seen that some board manufacturers implements the native usb via the GPIO 19 And 20. I saw some other (esp32-s2) boards that have some (deep) dpi switches to change the function of the usb port,  Which I don't have on my adafruit feather s3. What happens if the adafruit boards doesn't have gpio 19 and 20?  I also read that some manufacturers (banana pi) don't need a second port (the uart to usb) as the main s3 chip was improved with some virtual capabilities but i wonder how can the usb c port be set to work as host after flashing the code. So, which board can be use for my project? I don't want to use a secondary ext port. For me the smaller the better. Thanks

UnexpectedMaker, 2022-12-27:
@josehdx why is your Q in this issue - it's not related at all.  Please open a new issue, or better yet, maybe ask in another "non github/issue related" forum? This is a usability question, not an Arduino ESP32 bug.

[100 more replies]

closed 2025-01-16 by Parsaabasi