espressif/arduino-esp32 #6764: Is there a way to encrypt SPIFFS?

hzhh110, 2022-05-17:
### Related area

ESP32

### Hardware specification

ESP32

### Is your feature request related to a problem?

I want to write some encrypted things and store them in SPIFFS and store some data with limited number of functions. Is there any way to encrypt them? Prevent others from reading and modifying

### Describe the solution you'd like

I want to write some encrypted things and store them in SPIFFS and store some data with limited number of functions. Is there any way to encrypt them? Prevent others from reading and modifying

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

igrr, 2022-05-17:
@hzhh110 The only way as of now is to do the encryption at the application level, i.e. encrypt the data before storing it in SPIFFS. The main issue is where to store the encryption key. You would need to enable hardware flash encryption to store the key itself securely, and hardware flash encryption is not supported in arduino-esp32 when used with Arduino IDE.

(You can use arduino-esp32 as an IDF component and enable the [hardware flash encryption feature](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/security/flash-encryption.html), then create an encrypted partition for storing the encryption key. You still need to perform encryption at the application level, since SPIFFS partitions don't support hardware flash encryption.)

VojtechBartoska, 2022-06-14:
@hzhh110 Can I consider this as answered?

VojtechBartoska, 2022-07-14:
Closing as expired, if needed you can reopen the issue.

closed 2022-07-14 by VojtechBartoska