espressif/arduino-esp32 #6765: Investigate UART/CDC in all SoC

SuGlider, 2022-05-17:
@SuGlider I've now tested ESP32 Arduino 2.0.3 with a number of ESP32 boards, using a benchmark that involves uploading a program of about 1400 characters to my uLisp interpreter and then running it. The results are as follows:

| Board | Upload using UART | Upload using USB CDC | Benchmark |
| :----- | -------------------- | ----------------------- | ---------- |
| Adafruit Feather ESP32 | OK | Not available | 8.0 s |
| ESP32-S2-Soala-1 | OK | Not available | 5.7 s |
| Adafruit Feather ESP32-S2 | Not available | OK | 2.7 s |
| Adafruit QT-PY ESP32-S2 | Not available | OK | 2.7 s |
| Unexpected Maker Feather S2 | Not available | OK | 2.7 s |
| ESP32-S3-DevKitM-1 | OK | Serial hangs up | 6.1 s (USB) |
| ESP32-C3-DevKit1 | OK | Not available | 7.8 s |
| Adafruit QT-PY ESP32-C3 | Not available | Serial hangs up | ?? |

Key observations:

* Serial using UART works fine on all boards that support it.
* Serial using USB CDC works fine on ESP32-S2, fails on ESP32-S3 and ESP32-C3.
* Benchmark seems to run much slower if I've connected to the MCU via UART (eg 5.7s) compared to connecting to the same MCU via USB CDC (2.7s).

Any thoughts about why USB CDC is now OK on ESP32-S2 but not ESP32-S3 or ESP32-C3?

Also, any idea where there's such a difference in performance? My benchmark isn't making use of serial, except to print the result at the end.

_Originally posted by @technoblogy in https://github.com/espressif/arduino-esp32/issues/6221#issuecomment-1128950038_

SuGlider, 2022-05-17:
@technoblogy
Let's continue it here.

SuGlider, 2022-07-14:
Thanks @egnor for the reference. I'll take it into consideration for the S3 USB Serial JTAG interface.

SuGlider, 2022-11-20:
@all

Examples about UART controlling are at PR #7412

technoblogy, 2025-01-10:
Why is this closed? I have tested the ESP32-S3 CDC serial with Arduino core 3.1.0 and it still hangs up and so is unusable.