espressif/arduino-esp32 #6767: ESP32-C3 can not connect to WiFi AP (AUTH_EXPIRE)

XFer012, 2022-05-17:
### Board

Lolin C3-Mini

### Device Description

Lolin C3-Mini alone, nothing connected

### Hardware Configuration

Nothing

### Version

v2.0.3

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10 Pro 64 bit

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

ESP32-C3 can not connect to my AccessPoint, while ESP32 "stock" (ESP32 Devkit board) connects without issues

C3 reports continously "[W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE"

### Sketch

```cpp
/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <WiFi.h>

const char* ssid     = "[my SSID]";
const char* password = "[my WPA2 password]";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(2000);
        Serial.print("Not connected. Status="); Serial.println(WiFi.status());
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    delay(5000);
}

int value = 0;

void loop()
{
    delay(5000);
    ++value;

    Serial.print("connecting to ");
    Serial.println(host);

    // Use WiFiClient class to create TCP connections
    WiFiClient client;
    const int httpPort = 80;
    if (!client.connect(host, httpPort)) {
        Serial.println("connection failed");
        return;
    }

    // We now create a URI for the request
    String url = "/input/";
    url += streamId;
    url += "?private_key=";
    url += privateKey;
    url += "&value=";
    url += value;

    Serial.print("Requesting URL: ");
    Serial.println(url);

    // This will send the request to the server
    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n\r\n");
    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            Serial.println(">>> Client Timeout !");
            client.stop();
            return;
        }
    }

    // Read all the lines of the reply from server and print them to Serial
    while(client.available()) {
        String line = client.readStringUntil('\r');
        Serial.print(line);
    }

    Serial.println();
    Serial.println("closing connection");
}
```

### Debug Message

```plain
[W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
```

### Other Steps to Reproduce

I also tried ESP32 Devkit board, same AP, same parameters, same position; works without issues 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

XFer012, 2022-05-17:
Full output from serial monitor:

`Connecting to CasaWiFi
[  1503][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
Not connected. Status=6
[  2528][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  3550][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
Not connected. Status=6
[  4574][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  5600][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
Not connected. Status=6
[  6624][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  7648][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
Not connected. Status=6
[  8672][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  9696][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
Not connected. Status=6
[ 10720][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 11744][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
`
[...]

SuGlider, 2022-05-17:
@PilnyTomas, PTAL - Thanks!

PilnyTomas, 2022-05-19:
Hi @XFer012, could you please get the signal strength of the AP you are trying to connect to?
You can use [WiFiScan example](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiScan/WiFiScan.ino). You can also find it in Arduino IDE under File > Examples > WiFi > WiFiScan

XFer012, 2022-05-19:
Sure, here it's the output from WiFiScan:

scan done
10 networks found
1: [My SSID] (-36)*     <--- this is mine
2: TP-Link_xxxx (-67)*
3: Vodafone-xxxx (-77)*
4: Vodafone-xxxx (-79)*
5: Vodafone-xxxx (-79) 
6: Vodafone-xxxx (-80) 
7: Vodafone-xxxx (-85) 
8: Vodafone-xxxx (-86) 
9: Vodafone-xxxx (-87)*
10: Vodafone-xxxx (-87)*

Jason2866, 2022-05-19:
We have only troubles with the [Lolin C3-Mini](https://github.com/arendst/Tasmota/discussions/15443) and Tasmota. **Every!** other C3 board does work well. After we modified the HF part of the board we got it going more or less good. There is "somewhere" a bug in the HF board design.
Arduino core is NOT the problem!

XFer012, 2022-05-19:
You mean it scans OK, correctly detects RSSI, but cannot connect?

Sys64736, 2022-06-03:
@Jason2866 is right, there is a flaw in the design.  You need to set the transmit power to -8.5db right after wifi.begin then it will connect.  If using the softAP, you need to set it after starting also.
```
     WiFi.begin(ssid_char, password_char); 
     WiFi.setTxPower(WIFI_POWER_8_5dBm);
 ```

Jason2866, 2022-06-03:
The antenna matching is not correct. So the HF signal gets reflected. A really bad workaround is to reduce transmit power so the reflecting is reduced and does not eliminate the (weak) signal completly. A real fix needs changes in the HF matching network.

Sys64736, 2022-06-03:
Yes, it's a bad workaround and although it eliminates the AUTH_EXPIRE response it's deceiving because the board's RX'd signal will be much greater than the TX range.   I bought a bunch of these boards when they first came out and have used many successfully but only within ~10 meters of the AP, after that RX strength is still pretty good but too many dropped packets on the TX end.  I only use them now with SoftAP as it only needs to transmit to a mobile phone which is usually within a few meters of the device.   Hoping they improve the design and release an updated version of the board.

XFer012, 2022-06-03:
> @Jason2866 is right, there is a flaw in the design. You need to set the transmit power to -8.5db right after wifi.begin then it will connect. If using the softAP, you need to set it after starting also.
> 
> ```
>      WiFi.begin(ssid_char, password_char); 
>      WiFi.setTxPower(WIFI_POWER_8_5dBm);
> ```

Thank you, I will try this workaround. Better than nothing!

XFer012, 2022-06-05:
Confirmed working. Thanks!

VojtechBartoska, 2022-06-14:
Closing as solved, if needed, you can reopen the issue.

gauthier12, 2023-06-14:
Hi. 
If someone somewhere search it, same problem with the "banana pi BPI-Centi-S3" and the same workaround works
Thanks @Sys64736

fbernaly, 2024-02-13:
> @Jason2866 is right, there is a flaw in the design. You need to set the transmit power to -8.5db right after wifi.begin then it will connect. If using the softAP, you need to set it after starting also.
> 
> ```
>      WiFi.begin(ssid_char, password_char); 
>      WiFi.setTxPower(WIFI_POWER_8_5dBm);
> ```

That workaround also worked for me!

unmateria, 2024-04-10:
> Thank you, I will try this workaround. Better than nothing!

Thank you! it worked perfectly :)

totial, 2024-09-25:
Had the same problem and after trying MULTIPLE HW changes like adding capacitors, diodes to boost 3.3v to 3.4v, and modifying the smd wifi antenna I decided to try all esp32c3 super mini  boards from a batch of 10 I bought and found out that 4 out of 10 where unable to connect to wifi reporting error AUTH_EXPIRE. Exactly the same sketch (wifi tx power 8.5db) for all of them. My conclusion is the manufacturer doesnt chack for defectives and sells all of them. Dont know what other issues can arise. Hope this helps (to discard the broken ones).

Jason2866, 2024-09-26:
@totial You are right. There are bad designed (defective) C3 out there and sold.
Even well known names had problematic C3 boards design. Just to mention one.
Lolin. The did not really inform the users. Just the "fix" to lower the TX power was published.
The did a redesign of the C3 enhancing rev. number. The new rev. does not need the "fix" anymore.
To be 100% clear. Every C3 board which does need lowering TX power is a defective board!!

Slayingripper, 2024-12-18:
> @Jason2866 is right, there is a flaw in the design. You need to set the transmit power to -8.5db right after wifi.begin then it will connect. If using the softAP, you need to set it after starting also.
> 
> ```
>      WiFi.begin(ssid_char, password_char); 
>      WiFi.setTxPower(WIFI_POWER_8_5dBm);
> ```

This solved my issue! I was going at it for hours and could not for the life of me figure out why it wasn't connecting!

DansDesigns, 2025-02-18:
> [@Jason2866](https://github.com/Jason2866) is right, there is a flaw in the design. You need to set the transmit power to -8.5db right after wifi.begin then it will connect. If using the softAP, you [...]

Thank you so much, this solved a weeks worth of useless messing around in code!

Confirmed working on ESP32-S3 Devkit with ESP32-S3-WROOM-1

Jason2866, 2025-02-18:
@DansDesigns First time i hear about issues with S3 in "classic" devkit form factor.
Good to know there are bad build clones here too.

rubensmacielgomes, 2026-09-06:
Estava com este mesmo problema e encontrei o vilão deste pequeno notável esp32-c3. Depois de muito tempo, resolvi tirar os pinos que soldei na placa para melhorar a conexão na protoboard. Estes pinos cria um ruido ou sujeira na comunicação com o wifi ou bluetooth. Depois de retirados ou cortado o excesso dos pinos, voltou a funcionar perfeitamente a comunicação com wi-fi.

[image]

[image]

Ficou desta maneira e funcionou perfeitamente com o app rainmaker/alexa, sem precisar incluir código para melhorar a conexão com wifi.

[image]