espressif/arduino-esp32 #6773: False error "call of overloaded 'tone(int, int)' is ambiguous" since 2.0.2

ArminJo, 2022-05-19:
### Board

ESP32DevModule

### Device Description

%

### Hardware Configuration

%

### Version

v2.0.2

### IDE Name

Arduino, Sloeber, Github Actions

### Operating System

%

### Flash frequency

%

### PSRAM enabled

no

### Upload speed

%

### Description

Since 2.0.2, the compiler claims that the call tone(TONE_PIN, 1100) is ambiguous, which is totally nonsense 😞 . see sketch below.

### Sketch

```cpp
#include <Arduino.h>
void setup() {
  // put your setup code here, to run once:

}

#define TONE_PIN                27  // D27 25 & 26 are DAC0 and 1
#define TONE_LEDC_CHANNEL        1  // Using channel 1 makes tone() independent of receiving timer -> No need to stop receiving timer.
void tone(uint8_t aPinNumber, unsigned int aFrequency){
    ledcAttachPin(aPinNumber, TONE_LEDC_CHANNEL);
    ledcWriteTone(TONE_LEDC_CHANNEL, aFrequency);
}
void tone(uint8_t aPinNumber, unsigned int aFrequency, unsigned long aDuration){
    ledcAttachPin(aPinNumber, TONE_LEDC_CHANNEL);
    ledcWriteTone(TONE_LEDC_CHANNEL, aFrequency);
    delay(aDuration);
    ledcWriteTone(TONE_LEDC_CHANNEL, 0);
}
void noTone(uint8_t aPinNumber){
    ledcWriteTone(TONE_LEDC_CHANNEL, 0);
}
void loop() {
  // put your main code here, to run repeatedly:
            tone(TONE_PIN, 1100); // what is the compiler doing here???
            delay(50);
            tone(TONE_PIN, 1100, 10);
            delay(50);
}
```

### Debug Message

```plain
E:\WORKSPACE_ARDUINO\ArduinoSketches\ESP32Bug\ESP32Bug.ino: In function 'void loop()':
ESP32Bug:23:32: error: call of overloaded 'tone(int, int)' is ambiguous
             tone(TONE_PIN, 1100);
                                ^
E:\WORKSPACE_ARDUINO\ArduinoSketches\ESP32Bug\ESP32Bug.ino:12:6: note: candidate: 'void tone(uint8_t, unsigned int, long unsigned int)'
 void tone(uint8_t aPinNumber, unsigned int aFrequency, unsigned long aDuration){
      ^~~~
E:\WORKSPACE_ARDUINO\ArduinoSketches\ESP32Bug\ESP32Bug.ino:8:6: note: candidate: 'void tone(uint8_t, unsigned int)'
 void tone(uint8_t aPinNumber, unsigned int aFrequency){
      ^~~~
exit status 1
call of overloaded 'tone(int, int)' is ambiguous
```

### Other Steps to Reproduce

Just compile!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2022-05-19:
Hi @ArminJo,
It may be due to tone function is already implemented in 2.0.3. Check this PR #6402

ArminJo, 2022-05-19:
Thanks!!!
Any idea how I can check core version in c preprocessor?

lbernstone, 2022-05-19:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp_arduino_version.h

ArminJo, 2022-05-19:
Thanks!!! 🥇 
it works now https://github.com/Arduino-IRremote/Arduino-IRremote/blob/81e49468b5f33ead1361015fb1927b4ee4cc701a/examples/SimpleReceiver/PinDefinitionsAndMore.h#L69

closed 2022-05-19 by ArminJo