espressif/arduino-esp32 #6775: esp32 s3 , can't find xtensa-esp32s3-elf-g++

xiexingfeng, 2022-05-19:
### Board

esp32 s3

### Device Description

 - framework-arduinoespressif32 @ 2.0.3+sha.142fceb
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Converting Marlin.ino
'xtensa-esp32s3-elf-g++' \xb2\xbb\xca\xc7\xc4ڲ\xbf\xbb\xf2\xcdⲿ\xc3\xfc\xc1Ҳ\xb2\xbb\xcaǿ\xc9\xd4\xcb\xd0еĳ\xcc\xd0\xf2
\xbb\xf2\xc5\xfa\xb4\xa6\xc0\xed\xceļ\xfe\xa1\xa3
*** Error 1
AssertionError: :
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 184:
    env.SConscript("$BUILD_SCRIPT")

### Hardware Configuration

normal

### Version

v2.0.3

### IDE Name

platformio

### Operating System

windows 10

### Flash frequency

50

### PSRAM enabled

no

### Upload speed

115200

### Description

 - framework-arduinoespressif32 @ 2.0.3+sha.142fceb
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Converting Marlin.ino
'xtensa-esp32s3-elf-g++' \xb2\xbb\xca\xc7\xc4ڲ\xbf\xbb\xf2\xcdⲿ\xc3\xfc\xc1Ҳ\xb2\xbb\xcaǿ\xc9\xd4\xcb\xd0еĳ\xcc\xd0\xf2
\xbb\xf2\xc5\xfa\xb4\xa6\xc0\xed\xceļ\xfe\xa1\xa3
*** Error 1
AssertionError: :
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 184:
    env.SConscript("$BUILD_SCRIPT")

### Sketch

```cpp
- framework-arduinoespressif32 @ 2.0.3+sha.142fceb
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Converting Marlin.ino
'xtensa-esp32s3-elf-g++' \xb2\xbb\xca\xc7\xc4ڲ\xbf\xbb\xf2\xcdⲿ\xc3\xfc\xc1Ҳ\xb2\xbb\xcaǿ\xc9\xd4\xcb\xd0еĳ\xcc\xd0\xf2
\xbb\xf2\xc5\xfa\xb4\xa6\xc0\xed\xceļ\xfe\xa1\xa3
*** Error 1
AssertionError: :
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 184:
    env.SConscript("$BUILD_SCRIPT")
```

### Debug Message

```plain
- framework-arduinoespressif32 @ 2.0.3+sha.142fceb
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Converting Marlin.ino
'xtensa-esp32s3-elf-g++' \xb2\xbb\xca\xc7\xc4ڲ\xbf\xbb\xf2\xcdⲿ\xc3\xfc\xc1Ҳ\xb2\xbb\xcaǿ\xc9\xd4\xcb\xd0еĳ\xcc\xd0\xf2
\xbb\xf2\xc5\xfa\xb4\xa6\xc0\xed\xceļ\xfe\xa1\xa3
*** Error 1
AssertionError: :
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 184:
    env.SConscript("$BUILD_SCRIPT")
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-05-19:
@Jason2866 - Can you help with this PIO question?

Jason2866, 2022-05-20:
Maybe the path is too long (Windows has a limitation)
Placing Platformio stuff in root dir can help. Adding this in section platformio
```
[platformio]
...
core_dir = C:/.platformio
```

Jason2866, 2022-05-26:
Just tried to compile Marlin. No issue in general. Marlin is just not compatible with core 2.0.3 and generates "normal" compile errors. Marlin is still using core 1.0.4.

xiexingfeng, 2022-06-06:
> Maybe the path is too long (Windows has a limitation) Placing Platformio stuff in root dir can help. Adding this in section platformio
> 
> ```
> [platformio]
> ...
> core_dir = C:/.platformio
> ```

i can't the file xtensa-esp32s3-elf-g++ in my windows.

xiexingfeng, 2022-06-06:
i search all my windows files ,i can't find xtensa-esp32s3-elf-g++.

closed 2022-06-06 by xiexingfeng

lbernstone, 2022-06-06:
You should use toolchain-xtensa-esp32s3 for the esp32-s3

Jason2866, 2022-06-06:
Your setup is wrong. 
```
- framework-arduinoespressif32 @ 2.0.3+sha.142fceb
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Converting Marlin.ino
```
Do NOT add any toolchain manually. This is not needed. You have to use a boards.json which reflects your S3 board. In my experience trying in platformio.ini to set all parameters correctly is not so easy (you did not do).

xiexingfeng, 2022-06-10:
> 

> Your setup is wrong.
>
> ```
> - framework-arduinoespressif32 @ 2.0.3+sha.142fceb
>  - tool-esptoolpy @ 1.30300.0 (3.3.0)
>  - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
> Converting Marlin.ino
> ```
>
> Do NOT [...]

i havn't add it. it add by the scripts: arduino-esp32/.github/scripts/install-platformio-esp32.sh

you can check...

#!/bin/bash

export PLATFORMIO_ESP32_PATH="$HOME/.platformio/packages/framework-arduinoespressif32"
PLATFORMIO_ESP32_URL="https://github.com/platformio/platform-espressif32.git"

TOOLCHAIN_VERSION="8.4.0+2021r2-patch3"
ESPTOOLPY_VERSION="~1.30100.0"
ESPRESSIF_ORGANIZATION_NAME="espressif"

echo "Installing Python Wheel ..."
pip install wheel > /dev/null 2>&1

echo "Installing PlatformIO ..."
pip install -U https://github.com/platformio/platformio/archive/master.zip > /dev/null 2>&1

echo "Installing Platform ESP32 ..."
python -m platformio platform install $PLATFORMIO_ESP32_URL  > /dev/null 2>&1

echo "Replacing the package versions ..."
replace_script="import json; import os;"
replace_script+="fp=open(os.path.expanduser('~/.platformio/platforms/espressif32/platform.json'), 'r+');"
replace_script+="data=json.load(fp);"
# Use framework sources from the repository
replace_script+="data['packages']['framework-arduinoespressif32']['version'] = '*';"
replace_script+="del data['packages']['framework-arduinoespressif32']['owner'];"
# Use toolchain packages from the "espressif" organization
replace_script+="data['packages']['toolchain-xtensa-esp32']['owner']='$ESPRESSIF_ORGANIZATION_NAME';"
replace_script+="data['packages']['toolchain-xtensa-esp32s2']['owner']='$ESPRESSIF_ORGANIZATION_NAME';"
replace_script+="data['packages']['toolchain-riscv32-esp']['owner']='$ESPRESSIF_ORGANIZATION_NAME';"
# Update versions to use the upstream
replace_script+="data['packages']['toolchain-xtensa-esp32']['version']='$TOOLCHAIN_VERSION';"
replace_script+="data['packages']['toolchain-xtensa-esp32s2']['version']='$TOOLCHAIN_VERSION';"
replace_script+="data['packages']['toolchain-riscv32-esp']['version']='$TOOLCHAIN_VERSION';"
# Add ESP32-S3 Toolchain
replace_script+="data['packages'].update({'toolchain-xtensa-esp32s3':{'type':'toolchain','optional':True,'owner':'$ESPRESSIF_ORGANIZATION_NAME','version':'$TOOLCHAIN_VERSION'}});"
replace_script+="data['packages']['toolchain-xtensa-esp32'].update({'optional':False});"
# esptool.py may require an upstream version (for now platformio is the owner)
replace_script+="data['packages']['tool-esptoolpy']['version']='$ESPTOOLPY_VERSION';"
# Save results

Jason2866, 2022-06-10:
To do the setup the way Github CI is doing is not the best (recommended way)
Just do `pip install platformio`. Thats all.