espressif/arduino-esp32 #6776: PRIVATE KEY exposed in plain text

tsctrl, 2022-05-19:
### Board

ESP32 Dev Module

### Device Description

DevKitC

### Hardware Configuration

-

### Version

v2.0.3

### IDE Name

IDF Component, Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I can read my own private key that was added to PROGMEM as below:

[0;31m[%6u][E][%s:%u] %s(): [%s] Not connected to Wifi.
[0m
----BEGIN PRIVATE KEY----
!!!SECURITY CERT HERE!!!!
-----END CERTIFICATE-----
-FAIL- ../arduino/cores/esp32/esp32-hal-cpu.c 
[0;31m[%6u][E][%s:%u] %s(): Callback Object Malloc Failed[0m
 [0;31m[%6u][E][%s:%u] %s(): duplicate func=%08X arg=%08X[0m
 [0;31m[%6u][E][%s:%u] %s(): Bad frequency: %u MHz! Options are: 240, 160, 80, %u and %u MHz[0m
 [0;31m[%6u][E][%s:%u] %s(): Can not switch to 240 MHz! Chip CPU frequency rated for 160MHz.[0m
 [0;31m[%6u][E][%s:%u] %s(): CPU clock could not be set to %u MHz
 [0msetCpuFrequencyMhz /arduino/cores/esp32/esp32-hal-gpio.c 
 [0;31m[%6u][E][%s:%u] %s(): Invalid pin selected[0m
 [0;31m[%6u][E][%s:%u] %s(): GPIO config failed[0m
 [0;31m[%6u][E][%s:%u] %s(): GPIO ISR Service Failed To Start[0m

I did not want to add complexity to the code and not to add step to encrypt my plain text prior of deployment.
This below sketch does not effect the fw download or related to the issue. i am just showing how i add the cert and call it.

i am getting the above plain text readable cert from a postman call to the server. the server are in https. but once able to download. i am able to see the firmware content with cert detail in it. similar as opening using the file in texpad. this is security issue. it is probably issue with how i did the logging or how espressif is writing/tracking/logging error in the firmware during or after build.

this concern me on how properly keep security certificate from being printed and exposed! this is our server authorization key. anyone with the firmware can call our server. which can cause big security issue. please suggest simple approach on placing the cert or things to avoid such this issue. example: not using PROGMEM as it will be added to nvs? any suggestion?

Thanks!

### Sketch

```cpp
Here is the way i add my cert in the code as what has been added in lots of espressif example:

const char static root_ca[] PROGMEM = 
"-----BEGIN CERTIFICATE-----\n"
... 
"-----END CERTIFICATE-----\n";

setup(){
WiFiClientSecure *client = new WiFiClientSecure;
HTTPClient http;
}

loop(){
client->setCACert(root_ca);
http.begin(*client, "www.host.firmware.bin")
}
```

### Debug Message

```plain
-
```

### Other Steps to Reproduce

-

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-05-19:
you can read it how? where? by dumping the firmware?

tsctrl, 2022-05-19:
i guess is because i have uart dump feature enabled. so when dumpfile was created somehow the new firmware attached the dump content. the thing i cant understand but how this private key was actually readable in the firmware.

@igrr the sample sketch does not reflect, i just drafting part of it. my program has 3 cert, ca, client and key cert. i have just copy one of it. yes i have all to set as below:

```
    client->setCACert(root_ca);
    client->setCertificate(client_cer);
    client->setPrivateKey(client_key);
```

somehow i need to replicate the issue and i have to replicate the exact error that printing the dump. even i dump the error why the cert also got printed in the firmware? do you have suggestion, probably off the dump feature will fix? i have a custom dump to fs(text file) feature also. this really scared me.

tsctrl, 2022-05-19:
@me-no-dev my case is, i downloaded the firmware from my server and it displayed as plain text in postman. i also can see the key when opening the bin file using text file editor.

tsctrl, 2022-05-19:
to replicate this issue, Use SecureWifiClient to call a https server with 3 cert. then introduce the stack canary on the server.handleClient(); task with uart dump on.

mrengineer7777, 2023-04-13:
@tsctrl In my project I load ROOT_CERT to SPIFFS filesystem.  This allows me to download a replacement cert when this one expires.  If you are really saving a private key to flash memory, then dumping it, it will always be readable.  I suggest encrypting it.

closed 2023-04-13 by mrengineer7777

tsctrl, 2023-04-13:
hi @mrengineer7777, yes i am loading replacement private key from flash. thanks for the suggestion. did you suggest to encrypt ke private key? how could we do that?

mrengineer7777, 2023-04-13:
I don't have a recommendation for a specific encryption. Sorry.