espressif/arduino-esp32 #6783: Implement LED_BUILTIN constant + digitalWrite() overload for RGB LED

PilnyTomas, 2022-05-20:
### Related area

RGB LED, Blink example

### Hardware specification

C3, S3, dev boards with RGB LED

### Is your feature request related to a problem?

No issue found

### Describe the solution you'd like

Implement `LED_BUILTIN` constant for dev boards with RGB LED. Where not applicable print a helpful error message - for example, `This board does not have builtin LED` instead of default `error: 'LED_BUILTIN' was not declared in this scope`
For boards with RGB LED overload function `void digitalWrite(uint8_t pin, uint8_t val)` to activate the RGB LED via the appropriate driver.
Sample use of overloaded function:
```
digitalWrite(LED_BUILTIN, HIGH); // RGB turns full white
digitalWrite(LED_BUILTIN, LOW); // RGB turns off

// Nice to have (1) - ability to control each channel
digitalWrite(RED_LED_BUILTIN, HIGH); // Red channel full brightness
digitalWrite(GREEN_LED_BUILTIN, HIGH); // Green channel full brightness
digitalWrite(BLUE_LED_BUILTIN, HIGH); // Blue channel full brightness
// Similar for LOW -> channel is off

// Nice to have (2) - ability to control brightness
digitalWrite(RED_LED_BUILTIN, 128); // Red channel turns on to 50% brightness
```

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

igrr, 2022-05-20:
Regarding
```c++
// Nice to have (2) - ability to control brightness
digitalWrite(RED_LED_BUILTIN, 128); // Red channel turns on to 50% brightness
```
TBH it looks a bit odd from Arduino API perspective. How about using `analogWrite` for this purpose, instead? For a normal GPIO connected to an LED, analogWrite(LED_BUILTIN, value) can be used to control LED brightness by means of a PWM output.

me-no-dev, 2022-05-20:
This feature is much in line with defining a way for boards to declare what they have on-board. Things like LED, SD/SDMMC, LCD, etc. Also Peripheral Manager :) 
Declaring something like:
```cpp
#define BOARD_HAS_WS2802
#define LED_BUILTIN 18 // maybe we need to add offset so to not prevent using the pin for normal digitalWrite
```

Then:
```cpp
void digitalWrite(uint8_t pin, bool level){
#ifdef BOARD_HAS_WS2802
  if(pin == LED_BUILTIN){
    //use RMT to set all channels on/off
    return;
  }
#endif
  // regular digitalWrite logic
}
```

me-no-dev, 2022-05-20:
that `COLOR_LED_BUILTIN` idea look odd overall... how would you define the colors? If the user knows they have RGB, then they can use another API to set the color IMHO.

igrr, 2022-05-20:
I do agree with @me-no-dev that this should be implemented only to provide compatibility with sketches/libraries which use LED_BUILTIN.

More complex functionality such as setting the color could be handled using a dedicated library (like NeoPixel).

PilnyTomas, 2022-05-20:
Ok, we can keep it minimalistic and provide only the basic compatibility.

PR #6808 (2022-06-24): Implement simple RGB driver via digitalWrite; solving #6783
## Description of Change
This solution enables users to use Blink example with RGB LED on board.
When using function `digitalWrite` with constant `LED_BUILTIN` as pin parameter an RGB LED driver in the background will set white / off.
Added new example BlinkRGB extending the original Blink example with a demonstration of separate RGB channel setup.

## Tests scenarios
Tested on
S3 DevkitC
C3 DevkitC
S2 Kaluga

## Related links
Closes #6783
closed 2022-06-24 by me-no-dev