espressif/arduino-esp32 #6787: Conflicting declaration on variable INADDR_NONE

tsctrl, 2022-05-22:
### Board

ESP32 Dev Module

### Device Description

DevkitC

### Hardware Configuration

-

### Version

v2.0.3

### IDE Name

IDF Component, Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Build Failed. Conflicting declaration on variable INADDR_NONE

error: ambiguous overload for types 'IPAddress' and 'u32_t' components/lwip/lwip/src/include/lwip/inet.h
https://github.com/espressif/arduino-esp32/blob/1a7962ece8a4c6ffa1d64c5a86ed8ee58dde10ba/cores/esp32/IPAddress.h#L95
https://github.com/espressif/esp-lwip/blob/76303df2386902e0d7873be4217f1d9d1b50f982/src/include/lwip/inet.h#L71

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### Other Steps to Reproduce

-

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-05-22:
Duplicated #6760 

@tsctrl 
Please read #6760 and check if your issue is similar.
In general, this conflict depends on the order of file including.
It only happens when mixing IDF and Arduino, for which we can't help, given it goes beyond ESP32 Arduino Framework.

tsctrl, 2022-05-23:
yup same issue, just different library. and my case official Arduino Ethernet library function to check if dns is working.
here [github.com/arduino-libraries/Ethernet](https://github.com/arduino-libraries/Ethernet/blob/4e356a64b811199892c3d64648986cdb7985f632/src/Dns.cpp#L99)

not sure if the fix mention will fix this issue, closing this issue as per solution added in the referenced pr above.

SuGlider, 2022-05-24:
Is the issue in this line?

https://github.com/arduino-libraries/Ethernet/blob/4e356a64b811199892c3d64648986cdb7985f632/src/Dns.cpp#L99