espressif/arduino-esp32 #6789: ESP32 S3 - PSRAM activation make reboot

moreauftheobald, 2022-05-23:
### Board

ESP32-S3-DevKitC-1 N8R8

### Device Description

ESP32-S3-DevKitC-1 N8R8 on bread board.

### Hardware Configuration

GPIO 1 13,12,11, 10, 9, 46, 3, 8, 18, 17, 16, 15, 7, 6, 5, 4 connected to TFT data pin D0 to D15.
GPIO 14 connected to TFT Back light
GPIO 47, 21, 48 connected to SPI SD Card reader (FSPI Port 47 = SCK, 21 = MOSI, 48 = MISO) no CS PIN
GPIO 39, 40, 41 connected to SPI Touch Screen (HSPI Port 41 = SCK, 40 = MOSI, 41 = MISO) no CS PIN
GPIO 37, 38 Connected to I2C devices (3) (Twowire 0 port 38 = SCL, 37 = SDA)
GPIO 45, 35 Connected to Serial1 (45 = RX 35 = TX)
GPIO 36 connected to Buzzer (vie PWM);
GPIO 0 connected to Pull Down Button (open at startup)
GPIO 19, 20 Connected to LED
GPIO 1 Connected to voltage divider and used a analog input.

Reset Pin connected to pulldown button (open a startup) and  Reset pin of the screen.

Screen is powered up by 3.3v pin of board all aother device are powerd up by 5V pin.

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Linux 

### Flash frequency

QIO 80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

IDE params are : 

USB MODE: HARDWARE CDC and JTAG
USB CDC on boot : Dasabled
USB Firmware MSC on boot : disabled
USB DFU on boot : disabled
Upload Mode : USB-OTG CDC (Tiny USB);
CPU Frequency 240Mhz,
Flash Size : 8mb
Partition Scheme : Huge app
Core Debug Level : Nothing
Arduino run on Core 1
Events run on core 1

When Setting PSRAM to OPIPSRAM the ESP32 S3 reboot whitout any error message. When Setting PSRAM to DISABLE, everything run well no problem.

i think about power issue, but when reading voltage on 5V pin i get:
- 4.7 v when Software not running around 3.8 volt when running and a leat 3.6 V. even if PSRAM is On or OFF.

i think about Chip problem but i got 2 chip and get the same result with both.

### Sketch

```cpp
void setup() {
  //données test avant implémentation du GPS.
  donnes.latitude.value = 49.329411;
  donnes.longitude.value = 6.060297349;
  
  //Serial.setDebugOutput(true);
  //while (!Serial);
  pinMode(BATT, INPUT);
  pinMode(LED1, OUTPUT);
  pinMode(LED2, OUTPUT);
  digitalWrite(LED1, HIGH);
  digitalWrite(LED2, HIGH);
  //Starting TFT
  tft.init();
  tft.setRotation(ROTATION);
  tft.fillScreen(TFT_BLACK);
  tft.setTextSize(1);
  tft.setTextColor(TFT_WHITE);
  tft.println("Starting...");
  psramInit();
  tft.print("PSRAM init -> free PSRAM : ");
  tft.print(ESP.getFreePsram() / 1024.0, 2); 
  tft.println("KB");

  tft.print("Restoring Settings: ");
  //Starting Settings and restore stored Value
  settings.begin("vario", false);
  restore_settings();
  tft.println("done");

  //Starting Touch Screen
  tft.print("Starting Touch Screen: ");
  tp.begin();
  tft.println("done");

  //Starting SD Card
  tft.print("Starting SD Card Reader: ");
  vspi.begin(SDSCK, SDMISO, SDMOSI);
  if (!SD.begin(SDCSPIN, vspi, 80000000)) {
    tft.print("Card Mount Failed");
  } else {
    tft.print("SD Card reader started");
    uint8_t cardType = SD.cardType();
    if (cardType == CARD_NONE) {
      tft.print(", No SD card attached");
    } else {
      tft.print(", SD card found");
    }
  }
  tft.println();

  //Starting I2C Bus
  tft.print("Starting I2C Bus: ");
  if ( !I2C_0.begin(I2CSDA, I2CSCL)) {
    tft.println("Failled");
  } else {
    tft.println("Succeded");

    //Starting accelerometer
    tft.print("  --> Starting IMU: ");
    if (!bno.begin()) {
      tft.println("no IMU detected");
    } else {
      bno.setExtCrystalUse(true);
      tft.println("Succeded");
    }

    //Strating Presure Sensor
    tft.print("  --> Starting Barrometer: ");
    if (!bmp.begin_I2C(BMP3XX_DEFAULT_ADDRESS, &I2C_0)) {   // hardware I2C mode, can pass in address & alt Wire
      tft.println("no Barrometer detected");
    } else {
      bmp.setTemperatureOversampling(BMP3_OVERSAMPLING_8X);
      bmp.setPressureOversampling(BMP3_OVERSAMPLING_4X);
      bmp.setIIRFilterCoeff(BMP3_IIR_FILTER_COEFF_3);
      bmp.setOutputDataRate(BMP3_ODR_50_HZ);
      tft.println("Succeded");
    }

    //Strating Pitot Sensor
    tft.print("  --> Starting Speedmeter: ");
    if (!sens.begin()) {   // hardware I2C mode, can pass in address & alt Wire
      tft.println("no Speedmeter detected");
    } else {
      tft.println("Succeded");
    }
  }

  //Starting Serial1
  tft.print("Starting GPS: ");
  Serial1.begin(9600, SERIAL_8N1, RXD2, TXD2);
  while (!Serial1){
    tft.print("..");
  }
  tft.print("GPS Found");
   
  //Starting GPS
  GPS.sendCommand(PMTK_SET_NMEA_OUTPUT_RMCGGA);
  GPS.sendCommand(PMTK_SET_NMEA_UPDATE_5HZ);
  GPS.sendCommand(PMTK_API_SET_FIX_CTL_5HZ);
  //GPS.sendCommand(PGCMD_ANTENNA);
  //GPS.sendCommand(PMTK_ENABLE_SBAS);
  //GPS.sendCommand("$PMTK251,57600*2C");
  //Serial1.flush();
  //Serial1.end();
  //Serial1.begin(57600, SERIAL_8N1, RXD2, TXD2);
  delay(1100);
  String sent;
  while(Serial1.available()){
    char c = Serial1.read();
    sent.concat(c);
  }
  tft.println(sent);
  if((sent.indexOf("GGA") > 0) && (sent.indexOf("RMC") > 0)) {
    tft.println(" Data OK");
  } else {
    tft.println(" Data Corrupted");
  }
 

  tft.print("Testing Wifi Connection: ");
  wifi_test = test_wifi();
  tft.println("done");
  
  //Call the settings menu
  tft.println("Entering Setup.....");
  delay(2000);
  tft.fillScreen(TFT_BLACK);
  
  menu_settings();

  tft.fillScreen(TFT_WHITE);
}
```

### Debug Message

```plain
Not any
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-05-23:
I can not replicate this. Can you try something simpler?
[image]

me-no-dev, 2022-05-23:
In module variants that have embedded OSPI PSRAM, i.e., that embed ESP32-S3R8, pins IO35, IO36, and IO37 connect to the OSPI PSRAM and are not available for other uses.

closed 2022-05-23 by me-no-dev

moreauftheobald, 2022-05-23:
Ok thanks. i was thinking about something like this, but did not found anything over the net about it ;)

me-no-dev, 2022-05-23:
it's in the module datasheet :)

moreauftheobald, 2022-05-23:
Ok, i have changed my wiring excluding use of pin 35 36 and 37 and everything is working perfectly.

Sorry disturbing with this fools problem.

me-no-dev, 2022-05-23:
nah :) glad it worked. It is actually expected to get such issues now, till people get used to the S3 variants. We had similar problem with pins 16 and 17 on ESP32 with PSRAM