espressif/arduino-esp32 #6790: LED Intensity support is missed in the example CameraWebServer

DmytroKorniienko, 2022-05-23:
### Board

ESP32-CAM

### Device Description

CAMERA_MODEL_AI_THINKER ESP32-CAM

### Hardware Configuration

GPIO_NUM_4 // GPIO_NUM_4 = 4

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

win10

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

[image]

LED Intensity support is missed, moreover brownout detector was triggered raised periodically for CAMERA_MODEL_AI_THINKER ESP32-CAM

### Sketch

```cpp
https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer
```

### Debug Message

```plain
Brownout detector was triggered
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-05-23:
Brownout trigger is not a software issue. It is a function in the ESP32 chip which triggers when the needed power supply voltage is under 3.3V

DmytroKorniienko, 2022-05-23:
@Jason2866 Agree, but this issue is typical for a CAMERA_MODEL_AI_THINKER ESP32-CAM board, even if power supply is good enough. So my suggestion - disable this detection for this particular board.

Jason2866, 2022-05-23:
I have this board too. It does work well with a good power supply. Not a single brownout.
Disable brownout detection is not a good idea. Power problems creates weird issues.

DmytroKorniienko, 2022-05-23:
@Jason2866 If you will use intensive WiFi exchange of 1600*1200 streams with a enabled led you might got this too. Are you ever tried measure power consumption at different cases with multimeter & oscilloscope for this board? Onboard AMS1117 weak and not stable.

Jason2866, 2022-05-24:
I use highest possible resolution. Powering the device with a extern 3.3V power supply.
When lowering the resolution the onboard AMS1117 is capable to deliver.
So you just said the same it is a hardware issue. It is no solution to change the brownout detection.

SuGlider, 2022-05-24:
@DmytroKorniienko 
Regarding disabling brownout, I think it won't happen for the ESP Arduino Framework. Check #675
Doing it may lead to unpredictable new issues.

But you can do it in your own personal project when using Arduino as Component in an IDF project.
Or you can also create your own precompiled libraries using Arduino Lib Builder.
In both cases, brownout can be disabled using IDF Menuconfig 
(make menuconfig -> Component Config -> ESP32-specific -> Hardware brownout detect & reset)

#### Other Brownout discussions at 
https://github.com/nkolban/esp32-snippets/issues/168
https://www.esp32.com/viewtopic.php?t=2462
https://github.com/espressif/arduino-esp32/issues/863

#### SW way to disable it - not tested
**use at your own risk**  https://github.com/espressif/arduino-esp32/issues/863#issuecomment-671055308

DmytroKorniienko, 2022-05-24:
> SW way to disable it - not tested

Why you think so? I made a lot of my own tests and can provide screenshot with more than month uptime.

[image]

You can find this project in my repo and check by yourself :).

SuGlider, 2022-05-25:
>Why you think so? I made a lot of my own tests and can provide screenshot with more than month uptime.

Ok, I didn't test it myself - just that. 
But if you say so, I believe you :-)

rodyeo, 2022-07-27:
The ESP32-CAM Flight Light LED Illuminator solution which everyone is awaiting is now resolved by the blog author and programmer Bye Renzo

https://www.mischianti.org/forums/topic/esp32-cam-upgrade-camerawebserver-with-flash-on-esp32-framework-2-0-4/

Thank you ;)

closed 2022-12-08 by SuGlider

PUPPY98, 2023-09-06:
Does anybody know the details of the Flash led . like name?

PR #7533 (2022-12-08): Update camera example for support LED Intensity
## Description of Change
Reworked CameraWebServer example to add LED intensity support which use Arduino LEDC API.
In the `camera_pins.h` I have added definition of `LED_GPIO_NUM` on boards that I was able to find the LED pin in data sheet.
in the `app_httpd.cpp` I have added definition of `CONFIG_LED_ILLUMINATOR_ENABLED 1` to enable LED Intensity.

If the `LED_GPIO_NUM` pin is not defined, the LED intensity will be disabled.
If `CONFIG_LED_ILLUMINATOR_ENABLED 0`, the LED intensity will be disabled even if `LED_GPIO_NUM` is defined.

## Tests sce [... cut]