espressif/arduino-esp32 #6794: attachInterrupt() called in a constructor of a stack allocated class causes exception

sweetlilmre, 2022-05-24:
### Board

ESP32 Dev Module

### Device Description

Geekworm Easy ESP32-B1 Kit and other ESP32 devices.

### Hardware Configuration

Nothing attached to the module

### Version

v2.0.3

### IDE Name

PlatformIO

### Operating System

Windows 10 x64

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

This is more of a note of behaviour change in the GPIO handling between versions than necessarily a bug.
Pre 2.0.3 as far as I can tell, a class allocated on the stack in the global scope, with a constructor that called `attachInterrupt()` worked OK.

(see sketch)

Admittedly this may be a naive implementation, but it worked. From 2.0.3 (possibly 2.0.2) this causes the following exception:

```
E (278) gpio: esp_ipc_call_blocking failed (0x103)
[     6][E][esp32-hal-gpio.c:156] __attachInterruptFunctionalArg(): GPIO ISR Service Failed To Start
E (281) gpio: gpio_install_isr_service(449): GPIO isr service already installed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
```

This can be remedied by moving the `attachInterrupt()` call to an `Init()` method which is called from `setup()`.
I'm assuming this has something to do with initialization order and the move to IDF 4.4.

Note: To properly replicate the issue `attachInterrupt()` should be called more than once (in my code I attach to 2 pins for an encoder) otherwise the serial output is corrupted:

```
E (128) gpio: esp_ipc_call_blocking failed (0x�␘4Jj
[     3][E][esp32-hal-gpio.c:156] __attachInterruptFunctionalArg(): GPIO ISR Service Failed To Start
```

As I'm probably not going to be the last person to implement something in this manner, is there somewhere in the documentation where a note could be added to highlight this?
I'd be happy to contribute the text if someone could point me to where this would be appropriate to add.

### Sketch

```cpp
#include <Arduino.h>

class MyClass
{
  public:
      MyClass();
};

static void StaticCallbackFunc() {}

MyClass::MyClass()
{
    pinMode(25, INPUT_PULLUP);
    attachInterrupt(digitalPinToInterrupt(25), StaticCallbackFunc, CHANGE);
    pinMode(33, INPUT_PULLUP);
    attachInterrupt(digitalPinToInterrupt(33), StaticCallbackFunc, CHANGE);
}

MyClass myClassInstance;

void setup() {}
void loop() {}
```

### Debug Message

```plain
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
E (278) gpio: esp_ipc_call_blocking failed (0x103)
[     6][E][esp32-hal-gpio.c:156] __attachInterruptFunctionalArg(): GPIO ISR Service Failed To Start
E (281) gpio: gpio_install_isr_service(449): GPIO isr service already installed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40127877  PS      : 0x00060733  A0      : 0x800d9c0f  A1      : 0x3ffe3b10
A2      : 0x00000000  A3      : 0x3ffc2638  A4      : 0x3ffc2638  A5      : 0x00060723
A6      : 0x00060720  A7      : 0x00000001  A8      : 0x800d9778  A9      : 0x3ffe3ad0
A10     : 0x3ffbdcf4  A11     : 0x00000002  A12     : 0x00000021  A13     : 0x3ffe3b10
A14     : 0x007bdce0  A15     : 0x003fffff  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x40088bad  LEND    : 0x40088bbd  LCOUNT  : 0xfffffff7

Backtrace:0x40127874:0x3ffe3b100x400d9c0c:0x3ffe3b30 0x400d78a0:0x3ffe3b60 0x400d78ed:0x3ffe3b90 0x400d272e:0x3ffe3bb0 0x400d277a:0x3ffe3bd0 0x400d5778:0x3ffe3bf0 0x400e323b:0x3ffe3c10 0x40083592:0x3ffe3c40 0x400791fa:0x3ffe3c90  |<-CORRUPTED
  #0  0x40127874:0x3ffe3b100 in esp_intr_get_cpu at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/intr_alloc.c:703
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

bertmelis, 2022-05-24:
Put the constructor body in a class method 'setup' and call this method inside Arduino's 'setup()'.

Your class instance is a global variable and therefore not to be treated as a stack variable.

SuGlider, 2022-05-25:
@P-R-O-C-H-Y PTAL, thanks.

closed 2022-05-25 by me-no-dev

sweetlilmre, 2022-05-25:
@me-no-dev I'll admit it wasn't a clever thing to do, but it's an interesting new constraint on the system.

Maybe if I get a proper ESP32 debugger I will step through to find out what has changed in the init code as an exercise.

@bertmelis exactly, this is the fix I put in the issue text "This can be remedied by moving the `attachInterrupt()` call to an `Init()`..."