espressif/arduino-esp32 #6795: ESP32: WiFi stops responding if serial monitoring program closes port

avillacis, 2022-05-24:
### Board

YUBOX Node (ESP32)

### Device Description

YUBOX Node board with ESP32, no PSRAM, cp210x chip for serial-to-USB bridge, power supply via USB bridge 

### Hardware Configuration

LED on GPIO 4

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Linux (Fedora 36 x86_64)

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

On ESP32 (not ESP32-S2), when I power the device through the serial port bridge and keep a serial monitor program with the serial device open, the wifi works correctly until I shut down the serial monitor program or otherwise close the serial port. At that point, the WiFi of the ESP32 board stops responding.

To reproduce:
1. Attach a LED to a known GPIO port to check the board is alive. The YUBOX Node has an integrated LED at GPIO 4. Change sketch shown below to match if required.
2. Compile and flash the sketch shown below, with WiFi SSID and password changed to match local setup.
3. Open a serial monitor program at 115200 baud so the assigned WiFI IP can be read.
4. Start a ping to the indicated IP, verify that the IP answers the ping, and leave it running.
5. Verify that the LED is regularly turning on and off, indicating that the board is alive.
6. Shut down the serial monitor program, or otherwise close the serial port. The board should be left powered.

Actual result: the ping started in step 4 stops responding as soon as the port is closed. The LED keeps flashing on and off.
Expected result: the ping should keep running normally.

### Sketch

```cpp
#include <Arduino.h>
#include <WiFi.h>

#define WIFI_SSID "..."
#define WIFI_PASSWORD "..."

#define GPIO_LED 4

void on_wifievent(WiFiEvent_t event) {
    Serial.printf("[WiFi-event] event: %d\r\n", event);
    switch(event) {
    case SYSTEM_EVENT_STA_GOT_IP:
        Serial.println("WiFi connected");
        Serial.println("IP address: ");
        Serial.println(WiFi.localIP());
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        Serial.println("WiFi lost connection");
        break;
    }
}

void setup()
{
    Serial.begin(115200);
    delay(2000);

    pinMode(GPIO_LED, OUTPUT);
    digitalWrite(GPIO_LED, LOW);

    Serial.println("Connecting to Wi-Fi...");
    WiFi.onEvent(on_wifievent);
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
}

//uint8_t readbuf[64];

void loop()
{
    static uint32_t t_led = 0;
    static bool ledstate = false;

    if (millis() - t_led > 250) {
        t_led = millis();
        ledstate = !ledstate;
        digitalWrite(GPIO_LED, ledstate ? HIGH : LOW);
    }
}
```

### Debug Message

```plain
$ miniterm.py /dev/ttyUSB0 115200                                                                                                                                         --- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Connecting to Wi-Fi...
[WiFi-event] event: 0
[WiFi-event] event: 2
[WiFi-event] event: 4
[WiFi-event] event: 7
WiFi connected
IP address: 
192.168.200.31
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-05-25:
@avillacis 
I tested this sketch with an ESP32-Devkit-C. 
I close, open, close again, open it again the Serial Monitor Window from the Arduino IDE (Windows 11) .... many times.
Ping keeps working and replying back... no issue found.

I can't imagine why you have such issue.... maybe the board you are using has a defect?
Could you please try this sketch with another kind of ESP32 board?

me-no-dev, 2022-05-25:
I'm surprised it compiles. Events have changed: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.h

Anyway, there is no reason for the sketch to halt. Probably miniterm is switching the RST/DTR lines and causing the chip to halt/reboot/download

avillacis, 2022-05-26:
Confirmed to be a hardware issue.